ARMmbed/mbed-os #6979: ppp_lwip_disconnect hangs on acquiring ppp_close_sem semaphore when cellular connection fails with NSAPI_ERROR_CONNECTION_LOST

everhardt, 2018-05-22:
### Description
When using the OnboardCellularInterface to connect, it might be that the connection fails with reason NSAPI_ERROR_CONNECTION_LOST. If afterwards, disconnect() is called on the same interface, the thread will hang forever on https://github.com/ARMmbed/mbed-os/blob/master/features/FEATURE_LWIP/lwip-interface/ppp_lwip.cpp#L302.

* Target: [custom target for STM32F205RG](https://github.com/withthegrid/mbed-os/tree/wtg-mbed-os/targets/TARGET_STM/TARGET_STM32F2/TARGET_STM32F205RG)
* Toolchain: GCC_ARM,  7.2.1 20170904
* Tools: 1.5.0
* Mbed OS SHA: we use a fork (https://github.com/withthegrid/mbed-os/commit/0fcddaac2156599df88725d904acbd4f7e39d004), which is regularly updated from the master branch (last on 2104d8a)

I've fixed this locally by only waiting for the semaphore if `ppp_active` is true, see https://github.com/withthegrid/mbed-os/commit/df65e7d9352d810d65123f9bb83e2e569e4ba07e. I'm not sure this quickfix hasn't any unexpected side-effects, which is why I'm opening an issue instead of a pull request.

Reproduction steps:
* initialize a OnboardCellularInterface
* ensure the antenna is connected and call `connect();`
* remove the antenna during connection, in such a way that NSAPI_ERROR_CONNECTION_LOST is returned
* call `disconnect();`

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

0xc0170, 2018-05-22:
@ARMmbed/mbed-os-wan Please review

AriParkkila, 2018-05-23:
Yes, it looks like a good fix to me.

TeemuKultala, 2018-05-29:
ARM internal ref: https://jira.arm.com/browse/IOTCELL-1028

jarvte, 2018-06-07:
@everhardt please test this fix: https://github.com/ARMmbed/mbed-os/pull/7160

everhardt, 2018-06-08:
@jarvte, thank you, that seems to work!

jarvte, 2018-06-12:
@everhardt fix is now in master. Will be in release 5.9.1.

closed 2018-10-02 by adbridge