ARMmbed/mbed-os #6982: Remove target_config.json as behaviour changes when compiling tests

MarceloSalazar, 2018-05-22:
### Description

Mbed OS contains a target_config.json file here:
https://github.com/ARMmbed/mbed-os/blob/master/tools/test_configs/target_configs.json

When compiling tests for any of the platforms listed in this file (UBLOX_EVK_ODIN_W2, K64, REALTEK_RTL8195AM, NUCLEO_F429ZI), the build system goes and reads the target_config.json and ignores the mbed_app.json in my project.

This is an annoying and inconsistent behavior when comparing it with other platforms.

Steps to reproduce:
```
mbed import https://github.com/ARMmbed/mbed-stress-test
cd mbed-stress-test
mbed test --compile -m NUCLEO_F429ZI -t GCC_ARM
```

Workaround:
```
mbed test --compile -m NUCLEO_F429ZI -t GCC_ARM --app-config=mbed_app.json
```

Note, app-config is not needed when using other platforms.

Mbed CLI 1.5.1

### Issue request type
 [ ] Question
 [ ] Enhancement 
 [X] Bug

0xc0170, 2018-05-22:
@ARMmbed/mbed-os-tools Please review

theotherjimmy, 2018-05-22:
@MarceloSalazar The target_config.json is an imporntant feature in our testing infrastructure. We will not remove it.

However, we can give priority to mbed_app.json in the root of your project.

MarceloSalazar, 2018-05-23:
@theotherjimmy I think your proposal to prioritize mbed_app.json from the local directory is simple and good enough. Thanks!