ARMmbed/mbed-os #6986: LoRaWAN Stack Compiler Warnings in Release Build

mattbrown015, 2018-05-23:
### Description
When building using the release profile the following warning appear several times:

    Compile [ 15.8%]: LoRaWANInterface.cpp
    [Warning] LoRaMac.h@450,32: unused variable 'status' [-Wunused-variable]
    [Warning] LoRaMac.h@451,34: unused variable 'status' [-Wunused-variable]

These only appear once:

    Compile [ 17.1%]: LoRaMac.cpp
    ...
    [Warning] LoRaMac.cpp@843,22: unused variable 'status' [-Wunused-variable]
    [Warning] LoRaMac.cpp@949,26: unused variable 'status' [-Wunused-variable]

Only a small point I know but, any chance you could insert `MBED_UNUSED` for these 4 local varibles reduce the number of warnings.

https://github.com/ARMmbed/mbed-os/#3bcc076c0c08a4f411bfbeb9fcf173bb1343cd8c

### Issue request type

[ ] Question  
[X] Enhancement  
[ ] Bug

hasnainvirk, 2018-05-24:
:) yeah, I have been building with debug or developer profile where MBED_ASSERT is available that's why warning was not shown. If you are running with a release profile you will get this warning. I will see to it right away. Thanks

0xc0170, 2018-05-30:
@hasnainvirk Has this been fixed ?

hasnainvirk, 2018-05-30:
It should be in #6910

closed 2018-05-30 by 0xc0170

maxgerhardt, 2018-08-16:
This is not fixed in the most recent version of the stack and mbed-os version, can you recheck, @hasnainvirk?

```
features/lorawan/lorastack/mac/LoRaMac.h: In member function 'void LoRaMac::lock()':
features/lorawan/lorastack/mac/LoRaMac.h:476:18: warning: unused variable 'status' [-Wunused-variable]
osStatus status = _mutex.lock();
^~~~~~
features/lorawan/lorastack/mac/LoRaMac.h: In member function 'void LoRaMac::unlock()':
lorawan-physec/features/lorawan/lorastack/mac/LoRaMac.h:481:18: warning: unused variable 'status' [-Wunused-variable]
osStatus status = _mutex.unlock();
^~~~~~
``` 

Isn't the underlying problem that the `MBED_ASSERT` macro doesn't use the `expr`, thus leading to the warning?

```cpp
#ifdef NDEBUG
#define MBED_ASSERT(expr) ((void)0)
```

hasnainvirk, 2018-08-16:
@maxgerhardt You are right. This came back from the dead :) Will send a patch. Thanks for catching it.

hasnainvirk, 2018-08-16:
@maxgerhardt https://github.com/ARMmbed/mbed-os/pull/7807 Here is the fix mate.