ARMmbed/mbed-os #6998: Changing SPI Clock Pin On-the-Fly

AGlass0fMilk, 2018-05-23:
### Description
Platform: Nordic nRF52832, custom board

To minimize external hardware I am trying to software mux a SPI driver to communicate with two chips. One has a slave select (flash) and one does not (display). Sending data to the flash may cause strange behavior to appear on the display. I am trying to change the SPI Clock pin used On-the-Fly so only one chip gets the clock at a time.

@marcuschangarm with the recent nRF5x update, do you see any reasonably simple way to accomplish this while using official mbed SPI drivers? I am already using the three I2C/SPI/UART peripherals in the nRF52832 but I feel like there must be a clean way of changing driver pins at runtime.

Also - is there a way to allow deep sleep when the SPI peripheral is instantiated but inactive?

### Issue request type

[X] Question  
[ ] Enhancement  
[ ] Bug

0xc0170, 2018-05-24:
@ARMmbed/mbed-os-core Please review

AGlass0fMilk, 2018-05-24:
> Then the pin selection is automatically handled by the Mbed OS SPI driver.
I thought that may be the case so I tried instantiating two SPI instances that only differed in the CLK pin (and one had a slave select as well). The SPI-driven display did not operate. When I commented out the 2nd SPI driver instantiation the SPI-driven display started working again.

> On the NRF52, you can define your own pin map to ensure that both SPI objects gets assigned to the same hardware instance.

How might one go about doing this?

marcuschangarm, 2018-05-24:
>  I tried instantiating two SPI instances

Can you show us the before and after code please.

> On the NRF52, you can define your own pin map 

It is described in this section: https://github.com/ARMmbed/mbed-os/tree/master/targets/TARGET_NORDIC/TARGET_NRF5x#mbed-hal-implementation-details

AGlass0fMilk, 2018-05-24:
Here's a stripped-down version that reproduces the issue I'm seeing:

```
// PinDefinitions.h contains the definitions used for instantiating SPI drivers
// @marcuschangarm you can run this code if you define these pins yourself
#include "PinDefinitions.h"
#include "drivers/DigitalOut.h"
#include "drivers/SPI.h"
#include "mbed_wait_api.h"

/*!< SPI bus instance connected to display driver */
mbed::SPI display_spi(SPI_MOSI_PIN, NC, SPI_CLOCK_PIN);

/*!< SPI bus instance connected to external flash */
// ISSUE: If this line isn't commented out, the SPI transfer on the display_spi will not work correctly!
// To fix the display_spi, comment this line out
mbed::SPI flash_spi(SPI_MOSI_PIN, SPI_MISO_PIN, SPI_CLOCK_ALT_PIN, FLASHCHIPSELECT_PIN);

/*!< Latches the power on (custom-board specific) */
mbed::DigitalOut power_enable_pin(POWERENABLE_PIN, 1);

int main(void)
{

	// Latch the device power on
	power_enable_pin = 1;

	int i = 0;
	int j = 10;

	while (true)
	{
		/** Application code goes here */
		display_spi.write(i++);
		wait_ms(500);
		flash_spi.write(j++);
		wait_ms(500);
	}
}
```
The test code above uses two counters, i and j. The counter for the display spi starts at 0 and increments every display SPI transfer. The counter for the flash spi starts at 10 and increments every flash SPI transfer. Since the display doesn't reply with any data, the MISO line for that SPI instance is NC.

**NOTE:** In the logic analyzer data below, any transaction of the format `MOSI: X` pertains to the display SPI (since there is no response). Any transaction of the format `MOSI: X; MISO: X` pertains to the flash SPI.

In all cases, the clock pin operated correctly for both the display SPI and the flash SPI.

I captured the behavior of the muxed SPI instances in several cases using a logic analyzer:

## First Case - Both Instantiated, Display SPI Instantiated First
In this case, the **display SPI is instantiated before the flash SPI** (as in the code chunk above). The behavior indicates the flash SPI overrides or otherwise interferes with the proper operation of the display MOSI pin. That is, the flash SPI (instantiated last) counts out correctly starting from 10 but the display SPI always outputs 0.

[image]
_Transaction data for Case 1_

[image]
_Captured SPI transaction for Display SPI - note the clock functions correctly but MOSI does not_

[image]
_Captured SPI transaction for Flash SPI - note it functions properly_

## Second Case - Both Instantiated, Flash SPI Instantiated First
In this case, the display SPI functions properly but the flash SPI does not.

[image]
_Transaction data for Case 2_

[image]
_Captured SPI transaction for Display SPI - display SPI functions properly now that the instantiation order has been swapped_

[image]
_Captured SPI transaction for Flash SPI - MOSI no longer functioning (CLK still okay)_

## Third Case - Display SPI Only
In this case, the flash SPI has been commented out and the display SPI functions properly again
[image]

## Fourth Case - Flash SPI Only
In this case, the display SPI has been commented out and the flash SPI functions properly again
[image]

marcuschangarm, 2018-05-25:
It looks like a hardware limitation: two different SPI hardware instances can't use the same pins. 

The first object is assigned to hardware instance 0. The second object is assigned to instance 1 because the pins are not identical after you added an alternate clock pin.

You can force the two objects to use the same hardware instance by adding a custom pin map file:

map.c:
```
#include "pinmap_ex.h"

// { MOSI, MISO, CLK, Instance }
const PinMapSPI PinMap_SPI[] = {
    { SPI_MOSI_PIN, NC,           SPI_CLOCK_PIN,     0  },
    { SPI_MOSI_PIN, SPI_MISO_PIN, SPI_CLOCK_ALT_PIN, 0  },
    { NC,           NC,           NC,                NC }
};
```

You can read more about the custom map file here: https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NORDIC/TARGET_NRF5x/README.md#mbed-hal-implementation-details

AGlass0fMilk, 2018-05-25:
I'll try that out! Thanks for making the new driver system for the nRF5X so flexible. It really helps more advanced mbed developers fully utilize their platform 👍

AGlass0fMilk, 2018-05-29:
Seems to be working, thanks again @marcuschangarm

closed 2018-05-29 by AGlass0fMilk