ARMmbed/mbed-os #7013: The web export to uVision5 build fail.

mickeyln, 2018-05-25:
### Description

I export my program from to uVision5. Target is NUCLEO-F446RE. 
It always build fail and shows can not find configs/nanostack_full.cfg.
It can build after I copy the configs folder to mbed-os\features\nanostack\sal-stack-nanostack\source

### Issue request type

[ ] Question  
[ ] Enhancement  
[x] Bug

0xc0170, 2018-05-25:
@mickeyln How to reproduce? Which example and you use Mbed OS 5.8.5?

mickeyln, 2018-05-25:
Yes, I use Mbed OS 5.8.5.
Just import the example "mbed-os-example-blinky" to web compiler and export to uVision5 off-line compiler.

mister-walter, 2018-05-26:
I also have this issue when exporting for NUCLEO-F767ZI using the Make-GCC-ARM toolchain exporter.

0xc0170, 2018-05-30:
@ARMmbed/mbed-os-tools @ARMmbed/team-st-mcd  Please review

mickeyln, 2018-06-12:
It seems worked at mbed version 5.8.6

closed 2018-06-12 by mickeyln