ARMmbed/mbed-os #7017: Cellular and 27.007 mux

RobMeades, 2018-05-25:
### Description
A roadmap question for the new cellular interface: is it planned to add 27.007 mux support and, if so, which Mbed release is that likely to arrive in?

### Issue request type
[X] Question  
[ ] Enhancement  
[ ] Bug

TeemuKultala, 2018-05-29:
This is planned to be included in Mbed OS 5.11

AnttiKauppila, 2018-05-31:
Hi Rob,
Our MUX will be based on 27.010 supporting basic option and convergence layer type 1.
Will this be sufficient for your use cases?
We are checking the possibility to push this into a feature branch so you can start evaluating it before 5.11. We try to make this happen in following weeks.

RobMeades, 2018-06-05:
Hi Antti: I've asked the guy who asked me about this if he had resource to take a look at your earlier delivery but he said he hadn't.  Do let me know if you push something though and we'll see if we can find someone to have a poke at it.