ARMmbed/mbed-os #7026: Migration of https://github.com/ARMmbed/mbed-cli/issues/677

cmonr, 2018-05-26:
### Description

Migrating from https://github.com/ARMmbed/mbed-cli/issues/677

@amq is having issues using an mbed-os tool with Python 3.

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

amq, 2018-05-26:
I've used [5.8.3](https://github.com/ARMmbed/mbed-os/commit/c05d72c3c005fbb7e92c3994c32bda45218ae7fe), maybe mbed-cli >= 1.7.0 requires a newer version?

cmonr, 2018-05-26:
@amq Ah, yes. Give that a try. The Python 3 changes are currently staged for 5.9-rc1.

You should be able to go into the mbed-os directory, and do a `git checkout master` to pull the changes.

Oddly enough, I think I did find an issue, but haven't been able to reproduce it since...

cmonr, 2018-05-26:
Also, if you could provide steps using a project that's a bit more public (it mbed-os-example-blinky) it would help eliminate with further debugging.

Thanks!

0xc0170, 2018-05-30:
@amq Can you share how to reproduce this issue locally?

devangel77b, 2018-06-18:
I can reproduce this issue by cloning Blinky and compiling after installing mbed-cli 1.7.3 with Python 3.5.2
When I pip3 uninstall mbed-cli and revert to pip2 install mbed-cli it works fine. 

The end of the Python dump gives this: 
TypeError: the JSON object must be str, not 'bytes'
[mbed] ERROR: "/usr/bin/python3" returned error code 1.
[mbed] ERROR: Command "/usr/bin/python3 -u /home/evangeli/Desktop/mbed-os-example-blinky/mbed-os/tools/make.py -t GCC_ARM -m LPC1768 --source . --build ./BUILD/LPC1768/GCC_ARM" in "/home/evangeli/Desktop/mbed-os-example-blinky"
---

cmonr, 2018-07-17:
@devangel77b @amq Could y'all post the outputs of the following commands?
```
python --version
pip list --verbose
```

amq, 2018-07-17:
@cmonr 
mbed-cli: 1.7.5
python: 3.5.3

```
Traceback (most recent call last):
  File "/drone/src/github.com/redacted/redacted/mbed-os/tools/make.py", line 41, in <module>
    from tools.tests import TESTS, Test, TEST_MAP
  File "/drone/src/github.com/redacted/redacted/mbed-os/tools/tests.py", line 18, in <module>
    from tools.data.support import DEFAULT_SUPPORT, CORTEX_ARM_SUPPORT
  File "/drone/src/github.com/redacted/redacted/mbed-os/tools/data/support.py", line 17, in <module>
    from tools.targets import TARGETS
  File "/drone/src/github.com/redacted/redacted/mbed-os/tools/targets/__init__.py", line 555, in <module>
    update_target_data()
  File "/drone/src/github.com/redacted/redacted/mbed-os/tools/targets/__init__.py", line 544, in update_target_data
    in Target.get_json_target_data().items()
  File "/drone/src/github.com/redacted/redacted/mbed-os/tools/targets/__init__.py", line 70, in wrapper
    CACHES[(func.__name__, args)] = func(*args, **kwargs)
  File "/drone/src/github.com/redacted/redacted/mbed-os/tools/targets/__init__.py", line 142, in get_json_target_data
    Target.__targets_json_location_default)
  File "/drone/src/github.com/redacted/redacted/mbed-os/tools/utils.py", line 351, in json_file_to_dict
    object_pairs_hook=OrderedDict)
  File "/usr/lib/python3.5/json/__init__.py", line 312, in loads
    s.__class__.__name__))
TypeError: the JSON object must be str, not 'bytes'
```

andrewleech, 2018-09-19:
I've hit this same issue on debian stretch with python3 from the apt repos (python3_3.5.3-1_amd64.deb)
It also occurs in WSL with Python-3.5.2

It does not occur with the same project on Windows 10 (python 3.6.4 amd64 ).

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-441*

closed 2018-10-10 by cmonr

thi10, 2018-10-12:
Hello, I had the same problem.
Im running on UBUNTU 18.04
Ive tried with dozens of python versions (installed with pip2 and pip3). I also checked the GCC_ARM path and version...

Im using now:
Python 3.6.6
mbed-cli 1.82

I also installed all packages, 

`binutils build-essential bzip2 cpp cpp-6 dbus dh-python dirmngr dpkg-dev
  fakeroot file g++ g++-6 gcc gcc-6 gir1.2-glib-2.0 git-man gnupg gnupg-agent
  gnupg-l10n javascript-common krb5-locales less libalgorithm-diff-perl
  libalgorithm-diff-xs-perl libalgorithm-merge-perl libapparmor1 libasan3
  libassuan0 libatomic1 libbsd0 libc-ares2 libc-dev-bin libc6-dev libcc1-0
  libcilkrts5 libcurl3-gnutls libdbus-1-3 libdbus-glib-1-2 libdpkg-perl
  libedit2 liberror-perl libexpat1 libexpat1-dev libfakeroot libffi6
  libfile-fcntllock-perl libgcc-6-dev libgdbm3 libgirepository-1.0-1
  libglib2.0-0 libglib2.0-data libgmp10 libgnutls30 libgomp1 libgpm2
  libgssapi-krb5-2 libhogweed4 libicu57 libidn11 libidn2-0 libisl15 libitm1
  libjs-excanvas libk5crypto3 libkeyutils1 libkrb5-3 libkrb5support0 libksba8
  libldap-2.4-2 libldap-common liblocale-gettext-perl liblsan0 libmagic-mgc
  libmagic1 libmpc3 libmpdec2 libmpfr4 libmpx2 libncurses5 libnettle6
  libnghttp2-14 libnpth0 libp11-kit0 libperl5.24 libpopt0 libpsl5
  libpython-stdlib libpython2.7-minimal libpython2.7-stdlib libpython3-dev
  libpython3-stdlib libpython3.5 libpython3.5-dev libpython3.5-minimal
  libpython3.5-stdlib libquadmath0 libreadline7 librtmp1 libsasl2-2
  libsasl2-modules libsasl2-modules-db libsqlite3-0 libssh2-1 libssl1.0.2
  libssl1.1 libstdc++-6-dev libtasn1-6 libtsan0 libubsan0 libunistring0
  libx11-6 libx11-data libxau6 libxcb1 libxdmcp6 libxext6 libxml2 libxmuu1
  linux-libc-dev make manpages manpages-dev mercurial-common mime-support
  netbase openssh-client openssl patch perl perl-modules-5.24 pinentry-curses
  publicsuffix python python-minimal python-pip-whl python2.7
  python2.7-minimal python3 python3-cffi-backend python3-crypto
  python3-cryptography python3-dbus python3-dev python3-gi python3-idna
  python3-keyring python3-keyrings.alt python3-minimal python3-pkg-resources
  python3-pyasn1 python3-secretstorage python3-setuptools python3-six
  python3-wheel python3-xdg python3.5 python3.5-dev python3.5-minimal
  readline-common rename rsync sgml-base shared-mime-info ucf xauth
  xdg-user-dirs xml-core xz-utils`

cmonr, 2019-01-29:
@thi10 If this is still a problem, please open a new issue, since your problem is not directly related to the original issue.

PR #8177 (2018-10-10): tools/utils: Fix issue with loading json files as ascii on python3 linux
### Description

This resolves #7026, a failure to run mbed-cli tool on linux python 3.5 due to the error:
```
Traceback (most recent call last):
  File "/project/mbed-os/tools/make.py", line 41, in <module>
    from tools.tests import TESTS, Test, TEST_MAP
  File "/project/mbed-os/tools/tests.py", line 18, in <module>
    from tools.data.support import DEFAULT_SUPPORT, CORTEX_ARM_SUPPORT
  File "/project/mbed-os/tools/data/support.py", line 17, in <module>
    from tools.targets import TARGETS
  File "/project/mbed-os/to [... cut]