ARMmbed/mbed-os #7038: Mutex, Lock Guards and C++11 Question

mattbrown015, 2018-05-28:
### Description

I was thinking about using `Mutex` to protect some global state and musing how best to use it.

I didn't see any obvious reference to lock guards in the `Mutex` documentation but I did find `TESTS\mbedmicro-rtos-mbed\threads\LockGuard.h`.

At the present my best plan is to do something for myself similar to `LockGuard.h`.

I was wondering if there was a reason why a lock guard hasn't been implemented along with `Mutex`?

Another option would be create a profile and switch to C++11.  I don't understand the ramifications of using C++11.

### Issue request type

[X] Question  
[ ] Enhancement  
[ ] Bug

0xc0170, 2018-05-29:
@ARMmbed/mbed-os-core Please review

pan-, 2018-06-04:
@mattbrown015 Have a look at [`ScopedLock`](https://github.com/ARMmbed/mbed-os/blob/master/platform/ScopedLock.h)