ARMmbed/mbed-os #7041: NUCLEO_F746ZG export to uVision5 defaults to STM32F746NG

RaEyE, 2018-05-29:
### Description

When exporting a mbedOS based project to be used offline with uVision5 the default MCU used by
Keil uVision is not, as expected, the STM32F746ZGTx but the STM32F746ZGYx.

Therefore some Project-Settings might be off.
e.g. Insted of using the FloatingPoint Hardware provided by the STM32F746ZGTx MCUs the Code
Generation is set up to us Library Calls.

Another issue might be that a scatterfile, according to the comment section' of said scatterfile, for a STM32F746NG MCU is beeing used. Until now I can't make out issues regarding this, but who knows what changes in future applications.

A simple change of the Device used by Keil uVision is enough to solve this problem,
but it still needs to be done manually and simply might be forgotten.

### To reproduce
1) Simply export a random project for the target "NUCLEO_F746ZG"
    e.g.: `..\mbed-os\tools\project.py -m NUCLEO_F746ZG -i uvision5 --source . --source ..\mbed-os`
    (mbed-os folder and project-folder are separated for this example)

2) Have a look at the resulting Keil uVision5 Project Settings    
    [image]

    As you can see the Device (MCU) referenced to by Keil uV5 is not as expected the STM32F746ZGTx,
    but the STM32F746ZGYx. The Target settings the show, a FloatingPoint Hardware is not supported and
    therefore FPoperations are done in software.
    
    [image]
    
    
### Summary
**mbedOS-Target**: NUCLEO_F746ZG
**mbedOS-Version**: mbed-os-5.8.3 @ libarary-version 162
**mbedOS-SHA**: ae6c7c60f91c89cbf755a2f3c8ec9c66635849fd
(actualy it's not specific to this release, it's just now I got around to mention this minor issue)

**Keil UV5-Versio**: 5.25.2.0
**Keil Software Pack**: Keil.STM32F7xx_DFP.2.9.0

### Issue request type

[ ] Question  
[ ] Enhancement  
[x] Bug

0xc0170, 2018-05-29:
@ARMmbed/team-st-mcd Please review

bcostm, 2018-05-29:
Thanks to report this. I searched for "F746ZG" in all mbed-os files but I didn't find anything that could explain this for uVision exporter. I have no idea how the "Yx" device is selected ?

jeromecoutant, 2018-05-29:
Hi

In the Keil pdsc file, it seems that there are 2 "variants" for device STM32F746ZG:

        <device Dname="STM32F746ZG">
          <variant Dvariant="STM32F746ZGYx"> <feature type="CSP" n="143"/> </variant>
          <variant Dvariant="STM32F746ZGTx"> <feature type="QFP" n="144"/> </variant>

Maybe this is the reason which explains that the first one is selected by default ?

Is there a way to specify the second variant during export ? (need ARM experts help...)

Thx

bcostm, 2018-05-30:
Thanks for the info. Yes, it looks that the first line is selected. For SW4STM32 exporter we specify the full device in the mbed-os\tools\export\sw4stm32\__init__.py file. It would be better to write this device ID in the targets.json.

jeromecoutant, 2018-05-30:
@theotherjimmy 

Do you think this issue will be solved if we replace in targets.json file:
        "device_name": "STM32F746ZG",
by
        "device_name": "STM32F746ZGTx",
?

Thx

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-91*

jeromecoutant, 2019-06-24:
@0xc0170 
I think this issue can be closed now with the new arm pack manager.
Thx

closed 2019-06-24 by 0xc0170