ARMmbed/mbed-os #7043: Cordio H4 transport driver issue.

lrusinowicz, 2018-05-29:
### Description

While working on Mbed OS port to PSoC6 chip I was trying to use Cordio BLE stack with Cordio H4 UART transport (that Cypress libs also support) so I would have:

- Mbed OS and Cordio BLE running on M4 core (PSoC6 is is dual-core chip, with Cortex M0+ and M4F cores) connected to UART1

- Cypress BLE controller running on M0 core and connected to UART2

- both UARTs coss-connected outside of the chip.

I was using GCC_ARM toolchain and debug profile.
This bombed out, landing in mbed error when response from the controller was received, because H4 transport was calling Serial::writeble() in interrupt context (from on_controller_irq()), which then tried to obtain a lock on a mutex (in SerialBase::writable()), which is illegal in this context.
I've worked around this issue by adding a task to the H4 transport driver, which performs the actual serial communication and is only released in interrupt context. This actually works, but I believe a better way would be to use asynchronous serial API.
Unfortunately, the project is not (yet) publicly released, and I'm not intending on using this driver in the release version of the project, so this is a dead end for me.
Ultimately, I'm going to use a customized transport driver that will not require out-of-the-chip connections. :)

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

MarceloSalazar, 2018-05-29:
Internal reference: IOTPAN-59

0xc0170, 2018-05-29:
@ARMmbed/mbed-os-pan Please review

donatieng, 2018-05-31:
Thanks for identifying this @lrusinowicz - fix is on the way

donatieng, 2018-05-31:
#7075 should fix it!

PR #7075 (2018-06-03): BLE: Replace Serial with RawSerial in Cordio H4 Transport Driver
### Description

This PR should fix #7043. This replaces the use of the Serial class with RawSerial as the driver needs access to the serial port in interrupt context.

### Pull request type

    [X] Fix
    [ ] Refactor
    [ ] New target
    [ ] Feature
    [ ] Breaking change
closed 2018-06-03 by cmonr