ARMmbed/mbed-os #7046: CMSIS-RTOS error: Stack overflow for test :features-nvstore-tests-nvstore-functionality on LPC546XX and LPC54114

studavekar, 2018-05-29:
### Description
- Type: Bug
- Priority: Major 

---------------------------------------------------------------
## Bug

**Target**
LPC546XX and LPC54114

**Toolchain:**
GCC_ARM | ARM | IAR

**mbed-os sha:**
88d863a Merge pull request #6984 from stevew817/'efm32_irq_bugfix'

**Steps to reproduce**

Use command run :  mbed  test -m LPC54114  -t GCC_ARM -n features-nvstore-tests-nvstore-functionality -v

[ ] Question  
[ ] Enhancement  
[ X] Bug  

** log snippet 

```
(env) arm@austin-ci-linux-002:~/repo/mbed-os$ mbed test -m LPC54114 -c -v --run -n features-nvstore-tests-nvstore-functionality
[mbed] Working path "/home/arm/repo/mbed-os" (program)
[mbed] Exec "mbedgt --test-spec /home/arm/repo/mbed-os/BUILD/tests/LPC54114/gcc_arm/test_spec.json -n features-nvstore-tests-nvstore-functionality -V" in /home/arm/repo/mbed-os
mbedgt: greentea test automation tool ver. 1.3.3
mbedgt: test specification file '/home/arm/repo/mbed-os/BUILD/tests/LPC54114/gcc_arm/test_spec.json' (specified with --test-spec option)
mbedgt: using '/home/arm/repo/mbed-os/BUILD/tests/LPC54114/gcc_arm/test_spec.json' from current directory!
mbedgt: detecting connected mbed-enabled devices...
[1527609963.07][mbedls.platform_database]Duplicate platform ids found: daplink.5000, ignoring the definitions from /home/arm/.local/share/mbedls/platforms.json
[1527609963.07][mbedls.lstools_base]Could not read htm on from usb id 10540000024cac8c00000000000000000000000097969905. Falling back to usb id
mbedgt: detected 1 device
        +---------------+----------------------+--------------+----------------------------------------------------+--------------------------------------------------+
        | platform_name | platform_name_unique | serial_port  | mount_point                                        | target_id                                        |
        +---------------+----------------------+--------------+----------------------------------------------------+--------------------------------------------------+
        | LPC54114      | LPC54114[0]          | /dev/ttyACM0 | /mnt/pci-0000_00_14_0-usb-0_6_1_5_1_0-scsi-0_0_0_0 | 10540000024cac8c00000000000000000000000097969905 |
        +---------------+----------------------+--------------+----------------------------------------------------+--------------------------------------------------+
mbedgt: processing target 'LPC54114' toolchain 'GCC_ARM' compatible platforms... (note: switch set to --parallel 1)
        +---------------+----------------------+-------------------+----------------------------------------------------+--------------------------------------------------+
        | platform_name | platform_name_unique | serial_port       | mount_point                                        | target_id                                        |
        +---------------+----------------------+-------------------+----------------------------------------------------+--------------------------------------------------+
        | LPC54114      | LPC54114[0]          | /dev/ttyACM0:9600 | /mnt/pci-0000_00_14_0-usb-0_6_1_5_1_0-scsi-0_0_0_0 | 10540000024cac8c00000000000000000000000097969905 |
        +---------------+----------------------+-------------------+----------------------------------------------------+--------------------------------------------------+
mbedgt: test case filter (specified with -n option)
        features-nvstore-tests-nvstore-functionality
        test filtered in 'features-nvstore-tests-nvstore-functionality'
mbedgt: running 1 test for platform 'LPC54114' and toolchain 'GCC_ARM'
        use 1 instance of execution threads for testing
mbedgt: checking for 'host_tests' directory above image directory structure
        'host_tests' directory not found: two directory levels above image path checked
mbedgt: selecting test case observer...
        calling mbedhtrun: mbedhtrun -m LPC54114 -p /dev/ttyACM0:9600 -f "BUILD/tests/LPC54114/gcc_arm/features/nvstore/TESTS/nvstore/functionality/functionality.bin" -d /mnt/pci-0000_00_14_0-usb-0_6_1_5_1_0-scsi-0_0_0_0 -c default -t 10540000024cac8c00000000000000000000000097969905 -r default -C 4 --sync 5 -P 60
mbedgt: mbed-host-test-runner: started
[1527609963.45][HTST][INF] host test executor ver. 1.3.1
[1527609963.45][HTST][INF] copy image onto target...
[1527609963.45][COPY][INF] Waiting up to 60 sec for '10540000024cac8c00000000000000000000000097969905' mount point (current is '/mnt/pci-0000_00_14_0-usb-0_6_1_5_1_0-scsi-0_0_0_0')...
[1527609963.45][mbedls.platform_database]Duplicate platform ids found: daplink.5000, ignoring the definitions from /home/arm/.local/share/mbedls/platforms.json
[1527609963.45][mbedls.lstools_base]Could not read htm on from usb id 10540000024cac8c00000000000000000000000097969905. Falling back to usb id
[1527609969.80][mbedls.platform_database]Duplicate platform ids found: daplink.5000, ignoring the definitions from /home/arm/.local/share/mbedls/platforms.json
[1527609969.81][mbedls.lstools_base]MBED with target id '10540000024cac8c00000000000000000000000097969905' is connected, but not mounted. Use the '-u' flag to include it in the list.
[1527609970.32][mbedls.platform_database]Duplicate platform ids found: daplink.5000, ignoring the definitions from /home/arm/.local/share/mbedls/platforms.json
[1527609970.33][mbedls.lstools_base]MBED with target id '10540000024cac8c00000000000000000000000097969905' is connected, but not mounted. Use the '-u' flag to include it in the list.
[1527609970.83][mbedls.platform_database]Duplicate platform ids found: daplink.5000, ignoring the definitions from /home/arm/.local/share/mbedls/platforms.json
[1527609970.83][mbedls.lstools_base]MBED with target id '10540000024cac8c00000000000000000000000097969905' is connected, but not mounted. Use the '-u' flag to include it in the list.
[1527609971.34][mbedls.platform_database]Duplicate platform ids found: daplink.5000, ignoring the definitions from /home/arm/.local/share/mbedls/platforms.json
[... 83 lines cut ...]
mbedgt: all tests finished!
mbedgt: shuffle seed: 0.7764211262
mbedgt: test suite report:
+------------------+---------------+----------------------------------------------+---------+--------------------+-------------+
| target           | platform_name | test suite                                   | result  | elapsed_time (sec) | copy_method |
+------------------+---------------+----------------------------------------------+---------+--------------------+-------------+
| LPC54114-GCC_ARM | LPC54114      | features-nvstore-tests-nvstore-functionality | TIMEOUT | 131.11             | default     |
+------------------+---------------+----------------------------------------------+---------+--------------------+-------------+
mbedgt: test suite results: 1 TIMEOUT
mbedgt: test case report:
+------------------+---------------+----------------------------------------------+-------------------------------+--------+--------+--------+--------------------+
| target           | platform_name | test suite                                   | test case                     | passed | failed | result | elapsed_time (sec) |
+------------------+---------------+----------------------------------------------+-------------------------------+--------+--------+--------+--------------------+
| LPC54114-GCC_ARM | LPC54114      | features-nvstore-tests-nvstore-functionality | NVStore: Basic functionality  | 1      | 0      | OK     | 0.51               |
| LPC54114-GCC_ARM | LPC54114      | features-nvstore-tests-nvstore-functionality | NVStore: Multiple thread test | 0      | 0      | ERROR  | 0.0                |
| LPC54114-GCC_ARM | LPC54114      | features-nvstore-tests-nvstore-functionality | NVStore: Race test            | 1      | 0      | OK     | 0.8                |
+------------------+---------------+----------------------------------------------+-------------------------------+--------+--------+--------+--------------------+
mbedgt: test case results: 2 OK / 1 ERROR
mbedgt: completed in 131.30 sec
mbedgt: exited with code 1
[mbed] ERROR: "mbedgt" returned error code 1.
[mbed] ERROR: Command "mbedgt --test-spec /home/arm/repo/mbed-os/BUILD/tests/LPC54114/gcc_arm/test_spec.json -n features-nvstore-tests-nvstore-functionality -V" in "/home/arm/repo/mbed-os"
---
```
cc : @maclobdell @bentcooke

cmonr, 2018-05-29:
@ARMmbed/mbed-os-storage Mind taking a quick look at this as well, since an NVStore example is involved?

studavekar, 2018-05-29:
@davidsaada from the platform page : https://os.mbed.com/platforms/LPCXpresso54114/  it has 256KB flash, 192KB SRAM

davidsaada, 2018-05-29:
Thanks @studavekar. Should be quite sufficient then (comparing to other boards that have less RAM and work fine). I'll try looking into it tomorrow.

davidsaada, 2018-06-05:
Created PRs [#7127](https://github.com/ARMmbed/mbed-os/pull/7127) and [#7128](https://github.com/ARMmbed/mbed-os/pull/7128) to fix this issue. First one is mandatory, second one is optional (but recommended).

cmonr, 2018-08-03:
Closing since both PRs are merged!

closed 2018-08-03 by cmonr