ARMmbed/mbed-os #7047: TARGET_NRF52: Serial.getc() stops working after async Serial.read()

seppestas, 2018-05-29:
### Description
After reading data from a serial object using the async `Serial.read()` function and handling the data in the callback, calling `Serial.getc()` on the same object causes it to block forever, even if data is received.

Note: I only tested this issue on the new TARGET_NRF5x/TARGET_NRF52 target of mbed-os master (fe4337793).

E.g:

```cpp
#include "mbed.h"

Serial motor_controller(NINA_B1_GPIO_24, NINA_B1_GPIO_25, 9600);
event_callback_t serialEventCb;
#define RX_BUFFER_SIZE 20
uint8_t rx_buf[RX_BUFFER_SIZE];
volatile int extra_data = 0;

void serialCb(int events) {
    if (events & SERIAL_EVENT_RX_COMPLETE) {
        unsigned char i;
        int last_found_i = -1;
        for (i = 0; i < RX_BUFFER_SIZE; i++) {
            if (rx_buf[i] == '\n') {
                rx_buf[i] = 0;
                // parse_command((const char*)&rx_buf[last_found_i+1]);
                last_found_i = i;
            }
        }

        if (rx_buf[i] != 0 && last_found_i >= 0) {
            extra_data = last_found_i+1;
        }
    }
}

int main(void)
{
    serialEventCb = serialCb;
    motor_controller.read(rx_buf, RX_BUFFER_SIZE, serialEventCb, SERIAL_EVENT_RX_ALL);
    while (true) {
        if (extra_data != 0) {
            // motor_controller.getc() blocks
            printf("%c", motor_controller.getc());
        }
    }
}
```

Strangely calling `getc()` between calling `Serial.read()` and handeling the received data does not cause getc to block, seeming to cause the read command to be canceled.

E.g
```cpp
#include "mbed.h"

Serial motor_controller(NINA_B1_GPIO_24, NINA_B1_GPIO_25, 9600);
event_callback_t serialEventCb;
#define RX_BUFFER_SIZE 20
uint8_t rx_buf[RX_BUFFER_SIZE];

void serialCb(int events) {
    // ...
}

int main(void)
{
    serialEventCb = serialCb;
    motor_controller.read(rx_buf, RX_BUFFER_SIZE, serialEventCb, SERIAL_EVENT_RX_ALL);
    while (true) {
        // motor_controller.getc() works fine but serialCb is never called
        printf("%c", motor_controller.getc());
    }
}
```

### Issue request type
[ ] Question  
[ ] Enhancement  
[X] Bug

0xc0170, 2018-05-30:
@ARMmbed/mbed-os-core Please review

marcuschangarm, 2018-06-18:
The NRF52 hardware doesn't support synchronous UART so all the synchronous Mbed HAL calls are emulated in software. Whenever you call the asynchronous API you bypass the software emulation and get direct access to the hardware, which is why you can't mix the two modes.

donatieng, 2018-07-25:
I don't think concurrent use of sync/async APIs is a supported use case. @ARMmbed/mbed-os-core should this be documented more clearly?

0xc0170, 2018-07-25:
> It looks like combining sync and async serial is broken in both directions.

As responses indicate, concurrent access is not supported and should be documented.

seppestas, 2018-07-25:
Similar problems exist on the EFM32 target, see #4105.

Having the ability to use both the sync and async API is pretty use-full IMO so I would prefer this just being supported. If not, it should indeed be better documented and preferably even cause compile time (if possible) or run time errors instead of causing undefined behaviour.

The reason I wanted to use the sync API on the NRF52 is that the (emulated) character matching feature was removed on the new NRF52 target. I was planning to make a separate issue for this (and maybe fix it at one point), but for now I resorted to using sync getc()'s in combination with async reads to keep power consumption low (see my example).

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTHAL-73*

adbridge, 2020-10-13:
We've updated our automation, I will fix the requirements .

seppestas, 2020-10-13:
Hi @ciarmcom and @adbridge 

I believe this issue is more a documentation issue than a technical problem. IMO it should be clarified in the docs async (UART) operations should not be mixed with sync operations.

I found this issue on the NRF52832 target (ublox NINA-B1 module) but I expect it to exists on other targets as well.

It looks like the Serial API has been replaced by a BufferedSerial and UnbufferedSerial API. I'm not sure if it's still applicable so I'll close this issue.

PS. You might want to update the dead link in https://os.mbed.com/handbook/Serial. The mbed-os 2 handbook is sadly still the first thing to show up in Google searches.

closed 2020-10-13 by seppestas