ARMmbed/mbed-os #7048: Reducing Feature set

jflynn129, 2018-05-29:
### Description
Is there some way to reduce the features in a particular instance of Mbed OS?  For example, I would like to build the OS without the following features:

-  cellular
-  FEATURE_LWIP
-  nanostack
-  storage
-  FEATURE_BLE
-  lorawan
-  TESTS
-  filesystem
-  mbedtls
-  nvstore

I was thinking there may  be a way to use the mbed_app.json file to disabled/remove some of these features and thus reduce the size of the mbed OS memory footprint by eliminating features I am not using. Is this possible and if so, can you point me to the instructions for doing this.

I thought I could always simply place these sub-directories in the **.mbedignore** file but I didn't know  how that would affect the OS operation, can you comment.

### Issue request type
[X] Question  
[ ] Enhancement  
[ ] Bug

0xc0170, 2018-05-30:
The footprint should not be affected much - what you dont use should not be there. I would however understand if you want to reduce the build time.

We use .mbedignore for that purpose as you noticed, for instance here are some instructions how to reduce the footprint https://os.mbed.com/blog/entry/Reducing-memory-usage-by-tuning-RTOS-con/ (the mbedignore provides similar what you are after).

@ARMmbed/mbed-os-tools Please review

jflynn129, 2018-05-30:
Thank you for the reply/answer.

closed 2018-05-30 by jflynn129