ARMmbed/mbed-os #7064: Multiple SPIs on the same channel

OscarGarciaF, 2018-05-31:
[X] Question  
[ ] Enhancement  
[ ] Bug  

I'm trying to implement a datalogger on a SD card (using the standard sd-driver library) while communicating over the internet with a ethernet shield. Both of these devices uses SPI, their libraries have their own SPI instance and they are both connected on the same channel.

I know the two CS pins play a role in this but i cannot get them to work together on any combination of the CS pins value.

Should there only be a single SPI class instance for both of these devices or using two instances should work? (both of the instances are declared with the proper CS pin)
Is there a problem with using different frequencies on both of them?

OscarGarciaF, 2018-05-31:
For more info check the issue on the  sd-driver repo:  https://github.com/ARMmbed/sd-driver/issues/100#

deepikabhavnani, 2018-05-31:
CC @ithinuel 
Are you aware of any limitation/issues with SPI when shared by two instances with different CS?

ithinuel, 2018-05-31:
What seems fine is to share a single SPI channel between threads as long as they only use the synchronous part of the API.
I am not familiar enough yet with mbed's API to tell whether the async API is thread safe or not.

OscarGarciaF, 2018-05-31:
Insufficent stack size of the thread made it hang. I guess since one hanged it made the SD spi fail too.
I allocated more memory and it works now, thanks for the help

closed 2018-05-31 by OscarGarciaF