ARMmbed/mbed-os #7069: Cryptocell 310 doesn't work on NRF52840_DK

pan-, 2018-05-31:
### Description

When I do compile with GCC for the NRF52840, the function `trng_get_bytes` present in the crypto cell features never returns when it calls [`LLF_RND_GetTrngSource`](https://github.com/ARMmbed/mbed-os/blob/master/features/cryptocell/FEATURE_CRYPTOCELL310/trng.c#L68). 

With the help of the debugger I'm able to see that the application is actually stuck in a loop in the function `LLF_RND_StartTrngHW()`.

Also it looks like there's a leftover a merge operation in [GCC's binary folder](https://github.com/ARMmbed/mbed-os/tree/master/features/cryptocell/FEATURE_CRYPTOCELL310/binaries/TOOLCHAIN_GCC_ARM)

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

RonEld, 2018-05-31:
@pan- thank you for pointing to the leftover. It's probably a bug in the rebasing I have done.

RonEld, 2018-05-31:
@pan- could you mention how you reproduce this issue?
Have you called `mbedtls_platform_setup()` before the calls to Mbed TLS and `mbedtls_platform_teardown()` at the end?

pan-, 2018-06-01:
@RonEld Thanks for pointing out the functions to use unfortunately the implementation will not scale well with libraries. 

In my case; ble some crypto operations may be done by the controller attached to the board or by mbedtls so we can't ask blindly users to perform setup and teardown for use as we don't want to pay for code that we don't use. 

As a consequence the setup and teardown operations must be made in the implementation of the BLE abstraction when necessary. In the meantime application code or another library may also setup and teardown others `mbedtls_platform_context` when necessary. 

Unfortunately platform setup and teardown do not operate on context passed in parameter: they do operate on global context ([1](https://github.com/ARMmbed/mbed-os/blob/2f86c152f0b7c3c4944360591f3bed2cd76a97d8/features/cryptocell/FEATURE_CRYPTOCELL310/platform_alt.c#L29-L30), [2](https://github.com/ARMmbed/mbed-os/blob/2f86c152f0b7c3c4944360591f3bed2cd76a97d8/features/cryptocell/FEATURE_CRYPTOCELL310/TARGET_MCU_NRF52840/cc_platform_nrf52840.c#L26)).

This can cause issue if one module of the application has its platform context running while another application module terminate its own context.

RonEld, 2018-06-04:
@pan- I have added a reference counter for these functions, in #7099 that should now fit your needs.
Now BLE whould be ale to setup \ teardown the platform without interrupting application module.

Will this fit your needs?

marcuschangarm, 2018-07-13:
Perhaps we should disable Cryptocell support on the NRF52840 until we have a more permanent solution? At the moment it will only lead to problems under certain circumstances.