ARMmbed/mbed-os #7077: make .mbedignore more like .gitignore

bmcdonnell-ionx, 2018-05-31:
### Description
Change suggestions to make `.mbedignore` more like `.gitignore`:
1. Make leading slash anchor to the pattern to the current directory. i.e. `/mydir/` matches `./mydir/`, but not `./other/mydir/`; `mydir/` matches either/both.
2. `!` prefix negates a pattern.

### Issue request type
[ ] Question  
[X] Enhancement  
[ ] Bug

RaEyE, 2018-06-02:
I'd also would like to suggest to implemt the possibility to refer to subdirectories.
This would be convenient for projects not relying on the standard mbedOS projectstructure includeing the mbed-os folder directly as part of the project.

Currently we do use projects, used with Keil as offline toolchain, structured as followed:
```
projects/
├── mbed-os/
├── project01/
│   ├── submodule/
│   │   ├── stm32xx_emac_REPLACE.c
│   │   ├── .mbedignore
│   │   └── mbed_lib.json
│   └── main.cpp
```

Lets say the folder submodule contains files of the original mbed-os folder supposed to be excluded from an export, so they can be replaced by individual, project specific files.
The current `.mbedignore` does not allow for this.

cmonr, 2018-06-04:
@theotherjimmy

theotherjimmy, 2018-06-04:
@RaEyE Given that you have 2 mbed-os's in that project, could you clarify which one you would expect to be ignored?

theotherjimmy, 2018-06-04:
@bmcdonnell-ionx We can extend the patterns to include negations, if you would find that useful. I'm not so sure about 1; that has the potential to break current projects.

RaEyE, 2018-06-05:
@theotherjimmy Usually there aren't multiple mbed-os folders. If there are they usually are marked by a nameaddtion. e.g. `mbed-os_v157`.

The actual differentiation between mbed-os folder is done during the export for an offline-toolchain.
Give the folder-structure mentioned above, an export would be initiated by calling the mbed `project.py` script as follows:

`..\mbed-os\tools\project.py -m NUCLEO_F746ZG -i uvision5 --source . --source ..\mbed-os`

If referening the project to a specific mbed-os library-version this would usually be done by cloning/checking out the specific commit and exporting the project again by calling `project.py`.

theotherjimmy, 2018-06-05:
@RaEyE So could you give an example of an `.mbedignore` usage that you expect to work in one way, from `.gitignore` experience, but works another way?

RaEyE, 2018-06-05:
An example would be this. (Again under the assumption mentioned folderstructure is still in use).

A `.mbedignore` placed at `projects/project01/submodule/` containing the instruction to ignore all files containing `stm32xx_emac.c` or `stm32xx_emac.c` in their name/path.

Due to the `.mbedignore` system adding the actual subpath (e.g. `projects/project01/submodule/`) in front of the lines stored in `.mbedignore`, following example results are different to what I'd like them to be. Or at least the resulting effect.

```
*stm32xx_emac.c
*/stm32xx_emac.c
```

Result:
```
projects/project01/submodule/*stm32xx_emac.c
projects/project01/submodule/*/stm32xx_emac.c
```

This leads to the effect, that files from the mbed-os folder containing `stm32xx_emac.c` in their name. are still included after an export.

What I'd like the system to do, actualy is, on recognition of a leading `*` in front of a path, is to NOT add the subpath in front of that line. So, although the `.mbedignore` system is supposed to (my interpretation), only ignore files / folder further down it's own location, I'd like a possibility to globaly ignore files.

Therefore beeing able to ignore files from a mbed-os folder not beeing part of the actual project-path.

To prevent naming issues (although a bit more soffisticated) an ignore like this could be used to specifically ignore files or folders that are somewhere deeper down the path of mbed-os.

`*/mbed-os/*/stm32xx_emac.c`

I hope this TL;dr behemoth of a simple example is somewhat understandable and helps you figuring out supporting the mechanism I'm thinking of.

theotherjimmy, 2018-06-05:
>  the `.mbedignore` system is supposed to (my interpretation), only ignore files / folder further down it's own location

This is true, so

> I'd like a possibility to globaly ignore files.

Is not going to happen, without having a directory above the directory include an `.mbedignore` file.

Instead, you might be able to avoid duplicating mbed-os.

RaEyE, 2018-06-05:
> Instead, you might be able to avoid duplicating mbed-os.

That's the point, I'm not.
The was, is, and will be only one mbed-os folder.

theotherjimmy, 2018-06-05:
My mistake. I saw the `mbed-os.lib` and assumed you were using mbed-cli to get depnednecies. My mistake. You could work around this with a symlink. Otherwise we don't know which source directory is the "project root" and what the path to the other dir should be named in the `.mbedignore` file.

RaEyE, 2018-06-05:
Yeah I guess, that's the actual problem.
For now, if I wan't to exclude files directly from beeing imported right of mbed-os to the project, I remove them afterwards. Not ideal, I know, but for only a few files I at least works. (As long as I don't forget to do so :) ).

theotherjimmy, 2018-06-05:
It's hard to follow git's example on this one, as by definition git always has a "project root" and all paths make sense from that.

RaEyE, 2018-06-05:
Following the usual way of exporting project for an offline toolchain by mbed-cli or exporting them from the online-compiler this makes sense, yes, perfectly fine.

But for an environment completely set up to be used with a offline toolchain from the beginning to the end and multiple projects, I find it unnecessary to keep a mbed-os in each project folder. Therefore the attempt to move mbed-os outside the actual project folder one layer higher.

Obviously this leads / may lead to problems here and there, this beeing one. Although more of a minor nuissance than a real problem.

Since the mbed rootpath is defined by `.mbed` at the rootfolder of the project, wouldn't it also be possible to note down a reference on the location of the mbed-os folder relative to the position of the projectfolder to where a project has been exportet to.

For example, if a project export is initiated as mentioned above:
`..\mbed-os\tools\project.py -m NUCLEO_F746ZG -i uvision5 --source . --source ..\mbed-os`

the arguments passed to `project.py` do show, relative to the project-folder, the med-os folder is positioned at `..\mbed-os`. Might it be thinkable (not necessarily possible...) to extend the `.mbedignore` system in such a wait that instead of adding the subpath in front of lines found in `.mbedignore`, the path to the mbed-os folder is added? (I don't know if the ignoresystem could even handle relative paths from projectroot directed to mbed-os folder).

A possible `.mbedignore` line implementing this extension might startwith a token implying the switch of the path added in front.

e.g. `[MBED_LIB_PATH]*\stm32xx_emac.c`

theotherjimmy, 2018-06-05:
I had momentarily forgotten about the `.mbed` file. Good catch. I'm not sure about the proposed `.mbedignore` syntax, as the tools don't know that you're describing mbed-os in that `--source` parameter.

bmcdonnell-ionx, 2018-06-11:
@theotherjimmy,

> We can extend the patterns to include negations, if you would find that useful.

I think so.

> I'm not so sure about 1; that has the potential to break current projects.

How so?

theotherjimmy, 2018-06-11:
I think there may be `.mbedignore` files that currently exists that contain a leading `/` and don't expect them to be an anchor.

bmcdonnell-ionx, 2018-06-11:
Where?

AFAICT, that makes the leading `/` effectively meaningless. Right?

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-347*

SenRamakri, 2018-12-04:
@bmcdonnell-ionx - Thanks for your suggestion, unfortunately we are also looking into other methods to accomplish code-selection for individual projects. So, we are not looking into enhancing .mbedignore behavior for now as it also conflicts with some other tools improvements being planned. So I'm closing issue we cannot address this for now.

closed 2018-12-04 by SenRamakri