ARMmbed/mbed-os #7083: NUCLEO F446RE triggers interrupt directly even without SysTick and IRQs disabled

janjongboom, 2018-06-01:
### Description

I'm trying to wait for an interrupt from the RTC, but `__WFI` immediately returns, even with the SysTick disabled and calling `__disable_irq` right before. This is my code:

```cpp
#include "mbed.h"

static DigitalOut led(LED1);

int main() {
    wait_ms(1000);

    led = 1;
    wait_ms(200);
    led = 0;
    wait_ms(200);
    led = 1;

    // Disable IRQs
    __disable_irq();

    // Stop HAL tick to avoid to exit sleep in 1ms
    HAL_SuspendTick();

    __WFI();

    led = 0;
}
```

WFI immediately returns and the LED is turned off. I have no idea where this interrupt would come from. This blocks me from turning this board into sleep (because it waits for an interrupt, and this one fires directly).

### Issue request type

[ ] Question  
[ ] Enhancement  
[x] Bug

LMESTM, 2018-06-01:
@janjongboom - I guess you tried it first, but can you confirm that the same happen if you call sleep() ? 
Also I'm wondering whether the mbed ticker wouldn't fire anyway ...  Tickless support is planned for 5.9 release. 

```
int main() {
    wait_ms(1000);

    led = 1;
    wait_ms(200);
    led = 0;
    wait_ms(200);
    led = 1;

    sleep();

    led = 0;
}
```

janjongboom, 2018-06-01:
@LMESTM Yes, I found the issue because my device did not want to sleep, so I dissected what the sleep API was doing.

janjongboom, 2018-06-02:
So on Mbed OS master (rev 8c26e463ae9e160e89fec1bd95af8c8d92d9d8d1) the problem is still there. I have enabled tickless mode, and even call `OS_Tick_Disable();` before the `__WFI()` and the problem still persists.

LMESTM, 2018-06-04:
@janjongboom One more question : is this STM32 specific ? 
I don't know how mbed scheduling work but from my experiments, this is the timer interrupt that wakes-up the MCU from WFI. I'm not clear why we can't prevent it with a call to __disable_irq

janjongboom, 2018-06-04:
@LMESTM Yes this is STM32 specific... As far as I know the timer interrupt needs to be set up by the HAL too (but I'm not a core OS engineer).

LMESTM, 2018-06-05:
@janjongboom  It's surprising because tickless has been tested fine during last workshop so I was expecting this is the cause.  I will need some time to instrument, test further and dig into that ... will keep you posted

janjongboom, 2018-06-05:
@LMESTM So I've been testing this on NUCLEO-F446RE and NUCLEO-F411RE on Mbed OS 5.9 in tickless and non-tickless mode... The F411RE actually works fine in tickless mode (not in non-tickless mode, which I think is a bug), but the F446RE jumps straight out of sleep in both modes. Here's my complete test program: https://github.com/janjongboom/stm32-sleep

Wondering what's different between these two targets.

LMESTM, 2018-06-12:
@janjongboom 
Hi, I actually worked on DISCO_L475VG_IOT01A because I think (or thought before your last post) that there is not much board-related issue here.  

I used variations of the code copied below to investigate. My key findings are:
* the wake-up reasons are always related to timers. Either timers in case of Sleep or Low Power Timers in case of deep sleep / tickless mode. 
* mbed-os will always schedules the next ticks. either Every 1ms in ticked mode, or after the next even which is planned in tickless or in the maximum allowed time by HW in case of Tickless (should be around 1sec from my test on my board). So in any case, I understand that this is mbed-os design that will always schedule wake-ups. If you want to disable those wake-ups I think you need to sync with mbed-os core team.
* on STM32, you can force the mbed-os tick to be switched off by using the calls I have added in the code to     __HAL_TIM_DISABLE_IT(&TimMasterHandle, TIM_IT_CC1);    ... then the Timers will not wake-up the system anymore. I have added instead a low power timer to wake-up the system in the example code.

In all case, what I udnerstand is that mbed-os core is in charge of calling to sleep or deep-sleep, not the user thread. This is also expected to have regular wake-ups, for maintaining the system time and overall scheduling. If you want infinite sleep, then I think there is a need for a specific OS service. 

@0xc0170  - Martin, any feedback on the above ? 

```
#include "mbed.h"

static DigitalOut led(LED1);
static RawSerial serial(USBTX, USBRX);
static InterruptIn wakeup(USER_BUTTON);
extern TIM_HandleTypeDef TimMasterHandle;

void trigger() {
     serial.printf("triggered!\n");
}
	
int main() {
    serial.printf("TEST starts\r\n");
    LowPowerTimer lp_timer;
	uint32_t before_sleep, after_sleep;
	
	lp_timer.start();
    wait_ms(1000);

    led = 1;
    wait_ms(200);
    led = 0;
    wait_ms(200);
    led = 1;
 
	wakeup.rise(&trigger);
	
    while(1) {
	
	before_sleep = lp_timer.read_us();
    __HAL_TIM_DISABLE_IT(&TimMasterHandle, TIM_IT_CC1);
	hal_sleep();
    __HAL_TIM_ENABLE_IT(&TimMasterHandle, TIM_IT_CC1);
	after_sleep = lp_timer.read_us();
	
	for (int i = NonMaskableInt_IRQn; i < FPU_IRQn + 1; i++) {
        if(HAL_NVIC_GetPendingIRQ((IRQn_Type) i)) {
            serial.printf("Pending IRQ n=%d\r\n", i);
            HAL_NVIC_ClearPendingIRQ((IRQn_Type) i);
            HAL_NVIC_DisableIRQ((IRQn_Type) i);
        }
        if(HAL_NVIC_GetActive((IRQn_Type) i)) {
            serial.printf("Active IRQ n=%d\r\n", i);
            HAL_NVIC_DisableIRQ((IRQn_Type) i);
            HAL_NVIC_ClearPendingIRQ((IRQn_Type) i);
        }
    }
	
    led = !led;
    serial.printf("Slept for %d us\r\n", after_sleep - before_sleep);
    wait_ms(5000);
    }

    serial.printf("TEST ends\r\n");

}
```

0xc0170, 2018-06-12:
@ARMmbed/mbed-os-core  Please review

LMESTM, 2018-06-13:
@janjongboom  

I still need to make more tests on F446RE to check what could be specific - no idea so far. Will look at that.

Concerning HAL_SuspendTick, it is a STM32 specific API that was removed recently if you rebase to the latest mbed-os master. this was not meant to disable the mbed-os ticker that it will not prevent it to wake-up the system. 

With latest master, I have activated "MBED_TICKLESS" and "MBED_CPU_STATS_ENABLED"  and tested the below code.  The test shows that system spends 99% in sleep mode. (But susprissnigly not in deep sleep ... but this may be related to #7154 you've entered)

```
#include "mbed.h"

static DigitalOut led(LED1);
static RawSerial serial(USBTX, USBRX);
static InterruptIn wakeup(USER_BUTTON);
extern TIM_HandleTypeDef TimMasterHandle;
static mbed_stats_cpu_t stats;

void trigger() {
     serial.printf("triggered!\n");
	 mbed_stats_cpu_get(&stats);
	 serial.printf("Sleep %d \%\n", (stats.sleep_time * 100) / stats.uptime);
	 serial.printf("DeepSleep %d \%\n", (stats.deep_sleep_time * 100) / stats.uptime);
}

int main() {
    serial.printf("TEST starts\r\n");
	
    wait_ms(1000);
    led = 1;
    wait_ms(200);
    led = 0;
    wait_ms(200);
    led = 1;
 
	wakeup.rise(&trigger);
	
    while(1) {
      led = !led;
      wait_ms(1000);
    }
    serial.printf("TEST ends\r\n");
}
```

janjongboom, 2018-06-13:
@LMESTM You need to use `Thread::wait` otherwise the device cannot go into deepsleep, see #7154 (I believe this behavior is very weird).

LMESTM, 2018-06-13:
@janjongboom  ok - so if most of it works fine now, why do you try to call sleep modes from the main thread instead of relying on mbed-os ?  I think his somehow makes things more complex to analyze. Should I rather understand that tickless doesn't work on F446RE ?

LMESTM, 2018-06-18:
> Should I rather understand that tickless doesn't work on F446RE ?

I just tested on F446RE and cpu stats indicates that deep sleep works ok

janjongboom, 2018-06-19:
@LMESTM I was calling the sleep functions manually because I couldn't get my device to go into sleep mode, and was trying to break down the issue. If you see proper sleep behavior on latest master then that's good enough for me.

closed 2018-06-19 by janjongboom

LMESTM, 2018-06-19:
@janjongboom Ok thanks for feedback. Indeed in mbed-os the system will regularly wake-up for tickers / timers management. Note that I asked for a RTC alarm service to be able to sleep for very long time without wake-up ..