ARMmbed/mbed-os #7085: custom_targets.json target_overwrites ignored

drahnr, 2018-06-01:
### Description

GCC_ARM
mbed-cli 1.7.2
mbed-os 5.7.2

E
```
{
    "CUSTOM_TARGET_FUN": {
        "default_toolchain": "GCC_ARM",
        "inherits": [
            "NRF52_DK"
        ],
        "macros_add": [
            "MBED_TICKLESS",
            "CONFIG_NFCT_PINS_AS_GPIOS",
            "SOFTDEVICE_NONE",
            "NDEBUG=1",
            "OS_TASKCNT=3",
            "OS_IDLESTKSIZE=32",
            "OS_STKSIZE=4096",
            "OS_TIMERS=3",
            "OS_FIFOSZ=4",
            "OS_MUTEXCNT=3",
            "MBED_FAULT_HANDLER_DISABLED=1"
        ],
        "device_has_add": [
            "FLASH"
        ],
        // none of these are taking effect
        "macros_remove": [
            "S132",
            "SOFTDEVICE_PRESENT",
            "CONFIG_GPIO_AS_PINRESET",
            "BLE_STACK_SUPPORT_REQD"
        ],
        "release_versions": [
            "5"
        ],
        "device_name": "nRF52832_xxAA",
        "features_remove": [
            "BLE"
        ],
        "MERGE_BOOTLOADER": false,
        // the softdevice is still attempted to be merged
        "MERGE_SOFTDEVICE": false,
        "bootloader_supported": true,
        "target_overrides": {
            "*": {
                "platform.stdio-flush-at-exit": false
            },
           // these are not taking effect and are not defined in mbed_config.h generated header file
            "nvstore": {
                "enabled": true,
                "max_keys": 32,
                "area_1_address": "0x000F00000",
                "area_1_size": "16384"
            }
        },
        "stack_size": 8192
    }
}
```

### Issue request type
[ ] Question  
[ ] Enhancement  
[X] Bug

cmonr, 2018-06-01:
@drahnr What's the actual bug?

Also, what OS is this on?

drahnr, 2018-06-04:
I just checked out 5.9.rc1 and the issue is still there:

```
        "target.macros_remove": [
            "CONFIG_GPIO_AS_PINRESET",
            "BLE_STACK_SUPPORT_REQD",
            "SOFTDEVICE_PRESENT",
            "S132"

        ],
        "target.extra_labels_add" : [
            "SOFTDEVICE_NONE"
        ],
        "target.extra_labels_remove" : [
            "SOFTDEVICE_S132_FULL",
            "SOFTDEVICE_COMMON"
        ],
        "target.features_remove": [
            "BLE"
        ],
```
has to be placed in `mbed_app.json`  - `custom_targets.json` does NOT work.

drahnr, 2018-06-06:
OS is Linux, Fedora 28 to be specific, using python 2.7.x

theotherjimmy, 2018-06-28:
@drahnr That's not how target_overrides works. Each key in target_overrides is the name of a target, and the keys within that target are  in`<library>.<parameter>` format.

Further, a target cannot override a library's configuration.

drahnr, 2018-06-28:
Yes, build failures with descriptive errors would be helpful.

> Further, a target cannot override a library's configuration.

This would imply that I can not overwrite config parameters defined in `mbed_lib.json` by overwrites from `custom_targets.json`, correct?
Despite having read the `advanced configuration` section for the third time, I am not 100% sure on what overrules what, especially with `custom_targets.json` .

theotherjimmy, 2018-06-29:
That's correct. @drahnr You will be better served by reading the up to date documentation. https://github.com/ARMmbed/mbed-os-5-docs/blob/new_engine/docs/reference/configuration/configuration.md

cmonr, 2018-07-03:
Publiched documentation can also be found here: https://os.mbed.com/docs/v5.9/reference/configuration.html

SenRamakri, 2018-10-03:
@drahnr - We hope the documentation pointed out provided the info you wanted. At this point, we dont have more action items on this we are closing this issue. If you still think there are pending issues/questions please re-open the issue with the details.

closed 2018-10-03 by SenRamakri

drahnr, 2018-10-03:
@SenRamakri yes, it did indeed. thank you!