ARMmbed/mbed-os #7091: Truncate with littlefs

OscarGarciaF, 2018-06-01:
How may i truncate with the littlefs?

I see that lfs.cpp has the method defined but LittleFileSystem.cpp does not have a file_truncate method
how may i accomplish this?

Also may i know what's the maximum size of a file in the littlefs?
@geky

geky, 2018-06-04:
Hi @OscarGarciaF, you will need a `file_truncate` function that calls into littlefs like this:
``` cpp
int LittleFileSystem::file_truncate(fs_file_t file, off_t length)
{
    lfs_file_t *f = (lfs_file_t *)file;
    _mutex.lock();
    LFS_INFO("file_truncate(%p)", file);
    int err = lfs_file_truncate(&_lfs, f, length);
    LFS_INFO("file_truncate -> %d", lfs_toerror(err));
    _mutex.unlock();
    return lfs_toerror(err);
}
```

Sorry about all the hoops you've been having to jump through, truncate was added very recently and you've probably noticed it's not integrated into Mbed OS yet.

Files in littlefs limited by a 32-bit file size. So ~4GB.

dannybenor, 2018-06-26:
Please do not merge

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTSTOR-445*

closed 2018-12-04 by ciarmcom