ARMmbed/mbed-os #7093: LoRaWAN stack: get_channel_plan() work incorrectly

vsupacha, 2018-06-03:
### Description
LoRaWAN stack of mbed os 5.8.5 exported to Keil
DISCO_L072CZ_LRWAN1 board
AS923 frequency band
Connect to loriot.io server using the RisingHF gateway with packet forwarder V1 protocol

### Issue request type
Executing get_channel_plan() in CONNECTED and TX_DONE events return an incorrect list of channel parameters.
```
lorawan_channelplan_t channel_plan;
loramac_channel_t channels[8];
retcode = lorawan.get_channel_plan(channel_plan);
if (retcode == LORAWAN_STATUS_OK ) {
  printf("No. channels: %d\n", channel_plan.nb_channels);
  for (int i = 0; i < channel_plan.nb_channels; i++) {
    printf("ID (%d): %d-%d, RX1: %d\n", channel_plan.channels[i].id, channel_plan.channels[i].ch_param.band, channel_plan.channels[i].ch_param.frequency, channel_plan.channels[i].ch_param.rx1_frequency);
  }
} else {
  printf("retcode: %d\n", retcode);
}
```
Expected result:
Number of channels: 8
frequency: 923.2MHz to 924.6MHz

Got:
No. channels: 7
ID (0): 0-923200000, RX1: 0
ID (1): 0-923400000, RX1: 0
ID (2): 0-923000000, RX1: 0 <-- incorrect
ID (3): 0-923600000, RX1: 0 <-- shifted
ID (4): 0-923800000, RX1: 0 <-- shifted 
ID (5): 0-924000000, RX1: 0 <-- shifted
ID (6): 0-924200000, RX1: 0 <-- shifted

Communication is ok as reported by gateway logs
JUL: {"rxpk":[{"tmst":1081747612,"chan":2,"rfch":0,"freq":923.600000,"stat":1,"modu":"LORA","datr":"SF10BW125","codr":"4/5","lsnr":13.3,"rssi":-22,"size":17,"data":"gLO8eEiADwAPcKGun0KAQzI="}]}
JDL: {"txpk":{"tmst":1082747612,"freq":923.6,"rfch":0,"powe":14,"modu":"LORA","ipol":true,"datr":"SF10BW125","size":12,"data":"YLO8eEigFwBflurS","imme":false,"codr":"4/5"}}

[ ] Question  
[ ] Enhancement  
[x] Bug

vsupacha, 2018-06-03:
Using Keil debugger, I found that the problem is from LoRaPHY::add_channel() method. Line 1344 calls memcpy() to transfer channel setting to internal array variables phy_params.channels.channel_list[]. This method is invoked during OTAA confirmation stage. Therefore, it may be the problem from the loriot.io server, but I can not access other network server to confirm now.
[image]

AnttiKauppila, 2018-06-04:
@kivaisan, @hasnainvirk Can you check and verify this behaviour?

AnttiKauppila, 2018-06-04:
Hi @vsupacha The first 2 values are defaults but clearly the 3rd item looks wrong.
Do you have a logs for "freq":923.0(00000)? Just would like to get more information for this. This looks like an indexing error at first glance so more logs might open this up.

vsupacha, 2018-06-05:
By enabling trace, I can confirm that the channel index 2 does have the effect on the operation of LoRaWAN stack.
1. after the transmit is scheduled and next channel index is 2 (923.0MHz as configured in channel plan), the transmit will fail and fallback to default channel (923.2MHz)
2. on the gateway side, no acknowledge is done.

This is the log from mbed device.
(10:10:57.312) ␣Message␣Sent␣to␣Network␣Server␣
(10:10:57.312) No.␣channels:␣7
(10:10:57.312) ID␣(0):␣923200000,␣RX1:␣0
(10:10:57.312) ID␣(1):␣923400000,␣RX1:␣0
(10:10:57.312) ID␣(2):␣923000000,␣RX1:␣0
(10:10:57.312) ID␣(3):␣923600000,␣RX1:␣0
(10:10:57.312) ID␣(4):␣923800000,␣RX1:␣0
(10:10:57.350) ID␣(5):␣924000000,␣RX1:␣0
(10:10:57.350) ID␣(6):␣924200000,␣RX1:␣0
(10:10:57.350) <CR><ESC>[2K<ESC>[39m[INFO][LMAC]:␣RTS␣=␣4␣bytes,␣PEND␣=␣0,␣Port:␣15<ESC>[0m
(10:10:57.350) <CR><ESC>[2K<ESC>[90m[DBG␣][LMAC]:␣Frame␣prepared␣to␣send␣at␣port␣15<ESC>[0m
(10:10:57.350) <CR><ESC>[2K<ESC>[90m[DBG␣][LMAC]:␣Next␣Transmission␣in␣31341␣ms<ESC>[0m
(10:10:57.350) 
(10:10:57.350) ␣Message␣in␣queue␣
**(10:11:28.824) <CR><ESC>[2K<ESC>[90m[DBG␣][LMAC]:␣Next␣Channel␣Idx=2,␣DR=2<ESC>[0m
(10:11:29.157) <CR><ESC>[2K<ESC>[90m[DBG␣][LSTK]:␣Transmission␣completed<ESC>[0m
(10:11:34.171) <CR><ESC>[2K<ESC>[90m[DBG␣][LMAC]:␣Next␣Transmission␣in␣27683␣ms<ESC>**[0m
(10:12:01.986) <CR><ESC>[2K<ESC>[90m[DBG␣][LMAC]:␣Next␣Channel␣Idx=0,␣DR=2<ESC>[0m
(10:12:02.321) <CR><ESC>[2K<ESC>[90m[DBG␣][LSTK]:␣Transmission␣completed<ESC>[0m
(10:12:03.618) <CR><ESC>[2K<ESC>[90m[DBG␣][LSTK]:␣Ack=OK,␣NbTrials=0<ESC>[0m
(10:12:04.048) 
(10:12:04.048) ␣Message␣Sent␣to␣Network␣Server␣
(10:12:04.048) No.␣channels:␣7
(10:12:04.048) ID␣(0):␣923200000,␣RX1:␣0
(10:12:04.048) ID␣(1):␣923400000,␣RX1:␣0
(10:12:04.048) ID␣(2):␣923000000,␣RX1:␣0
(10:12:04.048) ID␣(3):␣923600000,␣RX1:␣0
(10:12:04.048) ID␣(4):␣923800000,␣RX1:␣0
(10:12:04.048) ID␣(5):␣924000000,␣RX1:␣0
(10:12:04.048) ID␣(6):␣924200000,␣RX1:␣0
(10:12:04.048) <CR><ESC>[2K<ESC>[39m[INFO][LMAC]:␣RTS␣=␣4␣bytes,␣PEND␣=␣0,␣Port:␣15<ESC>[0m
(10:12:04.048) <CR><ESC>[2K<ESC>[90m[DBG␣][LMAC]:␣Frame␣prepared␣to␣send␣at␣port␣15<ESC>[0m
(10:12:04.048) <CR><ESC>[2K<ESC>[90m[DBG␣][LMAC]:␣Next␣Transmission␣in␣31341␣ms<ESC>[0m
(10:12:04.048) 
(10:12:04.048) ␣Message␣in␣queue␣
(10:12:35.164) <CR><ESC>[2K<ESC>[90m[DBG␣][LMAC]:␣Next␣Channel␣Idx=1,␣DR=2<ESC>[0m
(10:12:35.497) <CR><ESC>[2K<ESC>[90m[DBG␣][LSTK]:␣Transmission␣completed<ESC>[0m
(10:12:36.799) <CR><ESC>[2K<ESC>[90m[DBG␣][LSTK]:␣Ack=OK,␣NbTrials=0<ESC>[0m

This is the log from gateway.
LRX: 60A38FC848A01200DBF528D8
**JUL: {"rxpk":[{"tmst":721342652,"chan":6,"rfch":1,"freq":924.400000,"stat":1,"modu":"LORA","datr":"SF10BW125","codr":"4/5","lsnr":-0.5,"rssi":-115,"size":12,"data":"YKOPyEigEgDb9SjY"}]}**
LRX: 80A38FC8488014000F35865007A2F9F3ED
JUL: {"rxpk":[{"tmst":753222724,"chan":0,"rfch":0,"freq":923.200000,"stat":1,"modu":"LORA","datr":"SF10BW125","codr":"4/5","lsnr":12.5,"rssi":-37,"size":17,"data":"gKOPyEiAFAAPNYZQB6L58+0="}]}
JDL: {"txpk":{"tmst":754222724,"freq":923.2,"rfch":0,"powe":14,"modu":"LORA","ipol":true,"datr":"SF10BW125","size":12,"data":"YKOPyEigEwC8n7MJ","imme":false,"codr":"4/5"}}
LTX: 60A38FC848A01300BC9FB309
LTS: 753490896 754222724 731828
LRX: 000404000000007ABE1111EF89CADB00AAE8CAA36E533E
JUL: {"rxpk":[{"tmst":791781868,"chan":0,"rfch":0,"freq":923.200000,"stat":1,"modu":"LORA","datr":"SF10BW125","codr":"4/5","lsnr":13.3,"rssi":-32,"size":23,"data":"AAQEAAAAAHq+ERHvicrbAKroyqNuUz4="}]}
JDL: {"txpk":{"tmst":796781868,"freq":923.2,"rfch":0,"powe":14,"modu":"LORA","ipol":true,"datr":"SF10BW125","size":33,"data":"IO0wDPJEVWWXRMG4d7uC/xP7kNVoDs94PiF+9SVS/gCb","imme":false,"codr":"4/5"}}
LTX: 20ED300CF24455659744C1B877BB82FF13FB90D5680ECF783E217EF52552FE009B
LTS: 792058825 796781868 4723043
LRX: 80215720498001000FE9E11FA1075527F8
JUL: {"rxpk":[{"tmst":829027204,"chan":1,"rfch":0,"freq":923.400000,"stat":1,"modu":"LORA","datr":"SF10BW125","codr":"4/5","lsnr":12.5,"rssi":-30,"size":17,"data":"gCFXIEmAAQAP6eEfoQdVJ/g="}]}
JDL: {"txpk":{"tmst":830027204,"freq":923.4,"rfch":0,"powe":14,"modu":"LORA","ipol":true,"datr":"SF10BW125","size":12,"data":"YCFXIEmgAAAxegcV","imme":false,"codr":"4/5"}}
LTX: 6021572049A00000317A0715
LTS: 829318551 830027204 708653

vsupacha, 2018-06-05:
I have added get_channel_plan() code just after lorawan.connect() line. It is proved that the updating is done after OTAA joins success.

(10:24:05.641) <CR><ESC>[2K<ESC>[90m[DBG␣][LSTK]:␣Initializing␣MAC␣layer<ESC>[0m
(10:24:05.683) 
(10:24:05.683) ␣LoRaWAN␣stack␣initialization␣OK␣
(10:24:05.695) 
(10:24:05.695) ␣set_confirmed_msg_retries␣OK␣
(10:24:05.695) 
(10:24:05.695) ␣enable_adaptive_datarate␣OK␣
(10:24:05.695) <CR><ESC>[2K<ESC>[90m[DBG␣][LSTK]:␣Initiating␣OTAA<ESC>[0m
(10:24:05.708) <CR><ESC>[2K<ESC>[90m[DBG␣][LSTK]:␣Sending␣Join␣Request␣...<ESC>[0m
(10:24:05.733) <CR><ESC>[2K<ESC>[90m[DBG␣][LMAC]:␣Frame␣prepared␣to␣send␣at␣port␣0<ESC>[0m
(10:24:05.753) <CR><ESC>[2K<ESC>[90m[DBG␣][LMAC]:␣Next␣Channel␣Idx=0,␣DR=2<ESC>[0m
**(10:24:05.753) No.␣channels:␣2
(10:24:05.753) ID␣(0):␣923200000,␣RX1:␣0
(10:24:05.783) ID␣(1):␣923400000,␣RX1:␣0**
(10:24:05.783) 
(10:24:05.783) ␣Connecting␣
(10:24:06.129) <CR><ESC>[2K<ESC>[90m[DBG␣][LSTK]:␣Transmission␣completed<ESC>[0m
(10:24:12.258) <CR><ESC>[2K<ESC>[90m[DBG␣][LMAC]:␣Frame␣prepared␣to␣send␣at␣port␣0<ESC>[0m
(10:24:12.268) <CR><ESC>[2K<ESC>[90m[DBG␣][LMAC]:␣Next␣Transmission␣in␣30625␣ms<ESC>[0m
(10:24:43.052) <CR><ESC>[2K<ESC>[90m[DBG␣][LMAC]:␣Next␣Channel␣Idx=0,␣DR=2<ESC>[0m
(10:24:43.426) <CR><ESC>[2K<ESC>[90m[DBG␣][LSTK]:␣Transmission␣completed<ESC>[0m
(10:24:48.885) <CR><ESC>[2K<ESC>[90m[DBG␣][LSTK]:␣OTAA␣Connection␣OK!<ESC>[0m
(10:24:48.895) 
(10:24:48.895) ␣Connected␣
**(10:24:48.895) No.␣channels:␣7
(10:24:48.895) ID␣(0):␣923200000,␣RX1:␣0
(10:24:48.935) ID␣(1):␣923400000,␣RX1:␣0
(10:24:48.935) ID␣(2):␣923000000,␣RX1:␣0
(10:24:48.935) ID␣(3):␣923600000,␣RX1:␣0
(10:24:48.935) ID␣(4):␣923800000,␣RX1:␣0
(10:24:48.935) ID␣(5):␣924000000,␣RX1:␣0
(10:24:48.935) ID␣(6):␣924200000,␣RX1:␣0**
(10:24:48.935) <CR><ESC>[2K<ESC>[39m[INFO][LMAC]:␣RTS␣=␣4␣bytes,␣PEND␣=␣0,␣Port:␣15<ESC>[0m
(10:24:48.935) <CR><ESC>[2K<ESC>[90m[DBG␣][LMAC]:␣Frame␣prepared␣to␣send␣at␣port␣15<ESC>[0m
(10:24:48.935) <CR><ESC>[2K<ESC>[90m[DBG␣][LMAC]:␣Next␣Transmission␣in␣31257␣ms<ESC>[0m

hasnainvirk, 2018-06-05:
@vsupacha There can be 5 channels configured in AS923 using a CFList. Two channels are default channels (always enabled.). So you have 7 channels to play with. Default channels always occur in the beginning of the channel list. 

The channel 92300000 (ID=2) is received in the CFList, so most probably the bug lies with Loriot. I am unable to locate a discrepancy in our side. Please ping me if you find anything else.

vsupacha, 2018-06-05:
@hasnainvirk I also think that it is the CF list problem from the Loriot server. So I sent the bug report to their server already. 

At present, I would like to close this issue first. BTW, the assertion of channel plan information would be good, because band and RX1 frequency information are absent as well.

closed 2018-06-05 by vsupacha