ARMmbed/mbed-os #7101: mbed cli export error on Python3

soleilplanet, 2018-06-04:
### Description
During MbedOS 5.9 OOB, when the default python is Python 3 (3.6.3) with mbed-cli 1.7.2, mbed cli will give a 'bad escape' error in the script file when performing 'mbed export'
Note that this is not seen with mbed-cli 1.7.1, but with 1.7.1, there is another issue of not recognizing izip, which is not seen in 1.7.2.
Also note that the 'izip not being recognized' problem is also seen with mbed-cli 1.7.1/Python 2.7.x, but with mbed-cli 1.7.2 and Python 2.7.x, everything works fine.

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

cmonr, 2018-06-04:
@soleilplanet Thanks for the issue report.

The izip issue was an artifact of an incomplete string replace. Will look into this today.

cmonr, 2018-06-04:
@soleilplanet If you could test this PR to confirm things have been resolved, that would be great. I'm in the process of generating the final test result log.

https://github.com/ARMmbed/mbed-os/pull/7111

closed 2018-06-07 by cmonr