ARMmbed/mbed-os #7112: GCC Windows export does not build as of #6577

cmonr, 2018-06-05:
### Description

When attempting to build an exported GCC project, the Makefile generated is now so large that the link step fails to link all files because the command is too long to be run successfully (all files are linked in one command).

Mbed CLI and Pip are at their latest versions. Issue is independent of Python versions and the target specified.

### Steps to reproduce
```
git clone https://github.com/ARMmbed/mbed-os-example-blinky blinky
cd blinky
git clone https://github.com/ARMmbed/mbed-os

# Will pass
cd mbed-os
git checkout 53aa1b0
cd ..

mbed export --ide GCC_ARM --m K22F --clean -vv
make

# Will fail
cd mbed-os
git checkout 5038135
cd ..

mbed export --ide GCC_ARM --m K22F --clean -vv
make
```

### Files
Attached are the generated makefiles for `mbed-os-5.8.0`, and the two Mbed OS SHAs in question.
[Makefile-5.8.5.txt](https://github.com/ARMmbed/mbed-os/files/2070656/Makefile-5.8.5.txt)
[Makefile-53aa1b0.txt](https://github.com/ARMmbed/mbed-os/files/2070657/Makefile-53aa1b0.txt)
[Makefile-5038135.txt](https://github.com/ARMmbed/mbed-os/files/2070655/Makefile-5038135.txt)

### Issue request type

[ ] Question  
[ ] Enhancement  
[x] Bug

SeppoTakalo, 2018-06-05:
Duplicate of:
* https://github.com/ARMmbed/mbed-os/issues/6335
* https://github.com/ARMmbed/mbed-os/issues/6861

cmonr, 2018-06-05:
Closing due to duplicate issue.

closed 2018-06-05 by cmonr