ARMmbed/mbed-os #7118: ARMC6 - Out of heap memory

mirelachirica, 2018-06-05:
### Description
Application, built with ARMC6 compiler for MTB_ADV_WISE_1570 target, runs into "SIGRTMEM: Out of heap memory". Same problem is not seen if application is built with GCC_ARM, IAR or ARM verison 5.

---------------------------------------------------------------
## Bug

**Target**
MTB_ADV_WISE_1570

**Toolchain:**
ARMC6

**Toolchain version:**

**mbed-cli version:** 
1.6.0

**mbed-os sha:**
5d8570b Merge pull request #7030 from kjbracey-arm/emac_ppp_fix

**Expected behavior**

**Actual behavior**
SIGRTMEM: Out of heap memory

**Steps to reproduce**
1. Found for mbed-os-example-cellular but same problem seen with any app, even empty main()
2. mbed compile -m MTB_ADV_WISE_1570 -t ARMC6
3. flash the .hex to the target
4 reset the target -> "SIGRTMEM: Out of heap memory" printed to terminal

kegilbert, 2018-06-26:
Hi  @mirelachirica!

I'm trying to track down one of those boards so I can take a look myself. Thanks for the great issue write up, just had one question in the meantime to help get me started.

When you mentioned 
> seen with any app, even empty main()

does that mean you grabbed something like the blinky example and wiped out the main file to just be:
```C++
#include <mbed.h>

int main(void) {
    printf("Hello!\r\n");
}
```

or you cleared out the `main()` function leaving the globally defined constants in place from the cellular example? Just wanted to double check.

kegilbert, 2018-07-24:
There hasn't been any activity on this issue, so I will be closing it for now. Hopefully your issue was resolved. If not, please feel free to reopen this issue and I will look back into it!

closed 2018-07-24 by kegilbert