ARMmbed/mbed-os #7125: NRF52832 .noinit section is fixed size and only 0xDC of size

drahnr, 2018-06-05:
### Description

    - What target does this relate to? NRF52832
    - What toolchain (name + version) are you using? GCC
    - What tools (name + version - is it mbed-cli, online compiler or IDE) are you using? 1.7,2
    - What is the SHA of Mbed OS (git log -n1 --oneline)? 5.9.1 rc1 6817dc43f9f5216cbe077059fc561fa89a766dc9

The NRF52 as of at least 5.9-rc1 has very small .noinit section defined.

This conflicts with projects using .noinit for various reasons leaving one without any means of configuring it's size.

What's the motivation behind that? 

Why is it placed in a separate section  RAM_NVIC? Is this only related to the vector table to be held in this section?

### Issue request type

[X] Question  
[ ] Enhancement  
[ ] Bug

drahnr, 2018-06-06:
Please see https://github.com/ARMmbed/mbed-os/pull/7145

cmonr, 2018-06-22:
Closing since the referenced PR is now merged :)

closed 2018-06-22 by cmonr