ARMmbed/mbed-os #7126: How to use custom linker script with mbed-cli?

drahnr, 2018-06-05:
mbed-os 5.9.1-rc1
mbed-cli 1.7.2

Trying to use a custom linker file always fails:
Just placing it under my custom target dir still yields the issue as described in https://github.com/ARMmbed/mbed-cli/issues/687 and https://github.com/ARMmbed/mbed-os/pull/6914

Also trying to explicitly set the linker file via the cmd line opt `-l` fails the same way:
```
mbed compile --target MY_TARGET --app-config ./config/mbed_app.json --profile ./config/debug.json -N artifact -DLAUNCH_APP=18 -l ./src/TARGET_MY_TARGET/MY_TARGET.ld
```

The underlying question is more, on how to properly use a custom linker file by means of defineing a custom target and exploiting the implicit directory structure, such that the derived MCU linker file does not collide with the custom linker script?

drahnr, 2018-06-06:
But this does not really fix the usecase of a custom linker file? Does it?

theotherjimmy, 2018-06-06:
@drahnr it does fix the error, the custom linker scirpt thing is supposed to work, aside from that bug.

drahnr, 2018-06-06:
@theotherjimmy thanks for clarifying, tracking that PR, thanks!

theotherjimmy, 2018-06-06:
@drahnr, @0xc0170 is mistaken, the PR he mentoined, #7124, has nothing to do with linker scripts. The PR that fixed it is #6914

mattbrown015, 2018-07-18:
@theotherjimmy sorry to start going over old ground but the comments above confused me a bit...

Just to be clear, placing a linker script in custom target folder is the correct thing to(?).

I created `TARGET_MYTARGET\my-script.ld` and it appeared to be found and used.

The warning I get is *by design* and the issues discussed above are because, for a while, an **error** occurred.

    Warning: Multiple linker scripts detected: .\mbed-os\targets\TARGET_STM\TARGET_STM32L4\TARGET_STM32L432xC\device\TOOLCHAIN_GCC_ARM\STM32L432XX.ld -> .\TARGET_MYTARGET\my-script.ld

Mbed CLI 1.7.5
Mbed OS [f4382d9](https://github.com/ARMmbed/mbed-os/commit/f4382d95b886482a707cf1249a16fd6d260f16b6)

mattbrown015, 2018-07-19:
> as we can probably arbitrarily pick a linker script

@theotherjimmy words like 'arbitrary' are a bit of a problem in the context of a build system :-}

Are you suggesting it may use the custom target linker script **or** it may use the base target linker script, depending on which one it found first?

If this is the case it sounds like a problem to me.  Hopefully you don't mean this and it knows the custom target is more important but the warning is for other situations.

drahnr, 2018-07-19:
Event the warning is a bit of a red hering and not clear _which_ linker script is used at first glance.

Maybe printing all found linker scripts and then marking the one that is used would be clearer.

theotherjimmy, 2018-07-19:
@drahnr @mattbrown015 It's the one that is found first; The linker script before the `->` is used. I used the word "arbitrary" intentionally :).

@mattbrown015 I have no way to determine which is "more important" at that time. 

@drahnr We can improve that warning. What would you suggest?

mattbrown015, 2018-07-19:
> It's the one that is found first; The linker script before the -> is used. I used the word "arbitrary" intentionally :).

Hhmm, OK, so putting a linker script in my custom target directory didn't do what I thought.

To ensure the custom target linker script is used I need to use `.mbedignore` to ignore any base target linker scripts.  This is fine, I think, unless trying to support targets when it could get complicated.

@drahnr's original question was:

> how to properly use a custom linker file by means of defineing a custom target

I now think the answer is 'carefully'! :-}

As it happens I'm not bothered any more because I achieved what I wanted with `"target.restrict_size"`. :-)

theotherjimmy, 2018-07-19:
@mattbrown015 Yeah! Who needs custom targets when you can configure ROM the way you like.

On a less silly note: others reading this issue should use a `.mbedignore` file to ignore the linker script that they no longer need.