ARMmbed/mbed-os #7130: online IDE fail to build BLE example during v5.9 OOB campaign

soleilplanet, 2018-06-05:
### Description
During v5.9 OOB campaign, when trying to build mbed-os-example-ble/BLE_SM with online IDE on target nrf52_dk, it failed with error message:
"Error: Attempt to override undefined parameter 'platform.stdio-baud-rate' in 'application[*]'

### Issue request type
[ ] Question  
[ ] Enhancement  
[X] Bug

theotherjimmy, 2018-06-05:
@soleilplanet The parameter in question: https://github.com/ARMmbed/mbed-os/blob/master/platform/mbed_lib.json#L19-L22

Should be available to override.

ashok-rao, 2018-06-05:
@donatieng ..any thoughts on this?

donatieng, 2018-06-05:
@soleilplanet @theotherjimmy Just tested here, works well - could you please retest @soleilplanet ?

donatieng, 2018-06-05:
Are you on the beta website @soleilplanet ?

theotherjimmy, 2018-06-05:
@soleilplanet I also cannot reproduce it on the beta website or on live os.mbed.com.

soleilplanet, 2018-06-08:
hi, I just deleted the project and re-do import again, this time it works.

0xc0170, 2018-06-08:
Thanks for the update. Closing as resolved

closed 2018-06-08 by 0xc0170