ARMmbed/mbed-os #7134: TCPsocket connect() error after close().

flipflop22, 2018-06-06:
### Description

I have been making a code with TCPsocket API. After I create a WIFI interface, socket.open() --> socket.connect() is working. But after I socket.close(), then I try to re-open socket.open(), then connect() always been error
  -target : UBLOX_EVK_ODIN_W2
  -toolchain: GCC ARM Embedded Toolchain
  -mbed-CLI
  -SHA : d18f86a

 Here is the main function()

    
	int count = 0;
    printf("WiFi example\n\n");
    count = scan_demo(&wifi);
    if (count == 0) {
        printf("No WIFI APNs found - can't continue further.\n");
        return -1;

    }
    printf("\nConnecting to %s...\n", MBED_CONF_APP_WIFI_SSID);
    int ret = wifi.connect(MBED_CONF_APP_WIFI_SSID, MBED_CONF_APP_WIFI_PASSWORD, NSAPI_SECURITY_WPA_WPA2);
    if (ret != 0) {
        printf("\nConnection error\n");
        return -1;
    }
    printf("Success\n\n");
    printf("MAC: %s\n", wifi.get_mac_address());
    printf("IP: %s\n", wifi.get_ip_address());
    printf("Netmask: %s\n", wifi.get_netmask());
    printf("Gateway: %s\n", wifi.get_gateway());
    printf("RSSI: %d\n\n", wifi.get_rssi());
	while(0==sending(&wifi)){wait(0.5)};
    wifi.disconnect();
    printf("\nDone\n");

open(), close(), and connect is in function sending()

    TCPSocket socket;
	SocketAddress serverAddr;
    nsapi_error_t response;
	
	if(!serverAddr.set_ip_address("192.168.2.220"))printf("error setting address");
	serverAddr.set_port(5555);
	
    printf("Open Socket\n");
    // Open a socket on the network interface, and create a TCP connection
    if(0>socket.open(net)) printf("Error initialize port");

    response = socket.connect(serverAddr);
    if(0 != response) {
        printf("Error connecting: %d\n", response);
        socket.close();
        return -1;
    }
    printf("Connected to 192.168.2.220 port 5555 %d\n", response);
    // Send a simple buffer
    printf("Sending buffer to 192.168.2.220 port 5555\n");
    char sbuffer[] = "ABCDEFGHIJKLMNO";
    nsapi_size_t size = strlen(sbuffer);
	response = socket.send(sbuffer, size);
	if (response < 0) {
            printf("Error sending data: %d\n", response);
            socket.close();
            return -2;}
    printf("Sent... %d\n", response);
    response = socket.close();
	if (response < 0) {
            printf("Error sending data: %d\n", response);
            return -3;}
	return 0;

### Issue request type

[x] Question  
[ ] Enhancement  
[ ] Bug

SeppoTakalo, 2018-06-06:
I cannot reproduce the problem described. See the log after found issues.

**Issues in the example:**
`while(0==sending(&wifi)){wait(0.5)};` is missing `;` after wait

`if(!serverAddr.set_ip_address("192.168.2.220"))printf("error setting address");` does not return, if there is error.

`if(0>socket.open(net)) printf("Error initialize port");` does not return in case of error.

**Log of running the provided example**
I modified the echo server to use our public one, and port to Discard port (9). As seen, the example continues, and does not stop.

```

Welcome to minicom 2.7

OPTIONS: 
Compiled on Mar 17 2016, 21:54:43.
Port /dev/tty.usbmodem23, 14:18:59

Press Meta-Z for help on special keys

$$$12360221012467153E60E41EWiFi example

Scan:
Network: 4tousnet secured: WPA2 BSSID: C:D6:BD:3c:56:79 RSSI: -63 Ch: 1
Network: DNA-Mokkula-2G-GJ73fE secured: WPA2 BSSID: B0:C1:9E:64:29:ca RSSI: -61 Ch: 1
Network: Dev-AccessNG secured: WPA2 BSSID: 84:D4:7E:e6:bc:82 RSSI: -64 Ch: 1
Network: Dev-AccessNG secured: WPA2 BSSID: 70:3A:E:55:65:c2 RSSI: -77 Ch: 1
Network: panoulu secured: None BSSID: 0:2A:10:d8:77:d0 RSSI: -72 Ch: 1
Network: OUKA-WLAN secured: WPA2 BSSID: 0:2A:10:d8:77:d1 RSSI: -74 Ch: 1
Network: Handsome secured: WPA2 BSSID: F0:5C:19:94:60:80 RSSI: -72 Ch: 1
Network: Evolta secured: WPA2 BSSID: F0:5C:19:94:60:82 RSSI: -71 Ch: 1
Network: mbed secured: WPA2 BSSID: C4:7:2F:c:48:5d RSSI: -39 Ch: 4
Network:  secured: WPA2 BSSID: C6:7:2F:c:48:5e RSSI: -43 Ch: 4
Network: Elisa_Mobi_817F secured: WPA2 BSSID: 8C:25:5:e2:81:7f RSSI: -46 Ch: 5
Network: AccessNG secured: WPA2 BSSID: 70:3A:E:55:69:60 RSSI: -66 Ch: 6
Network: Guest-AccessNG secured: None BSSID: 70:3A:E:55:69:61 RSSI: -65 Ch: 6
Network: Dev-AccessNG secured: WPA2 BSSID: 70:3A:E:55:69:62 RSSI: -65 Ch: 6
Network: AccessNG secured: WPA2 BSSID: 70:3A:E:55:63:e0 RSSI: -70 Ch: 11
15 networks available.

Connecting to mbed...
Success

MAC: d4:ca:6e:70:71:f2
IP: 192.168.100.14
Netmask: 255.255.255.0
Gateway: 192.168.100.1
RSSI: -39

Open Socket
Connected to 52.215.34.155 port 9 0
Sending buffer to 192.168.2.220 port 9
Sent... 15
Open Socket
Connected to 52.215.34.155 port 9 0
Sending buffer to 192.168.2.220 port 9
Sent... 15
Open Socket
Connected to 52.215.34.155 port 9 0
Sending buffer to 192.168.2.220 port 9
Sent... 15
Open Socket
Connected to 52.215.34.155 port 9 0
Sending buffer to 192.168.2.220 port 9
Sent... 15
Open Socket
Connected to 52.215.34.155 port 9 0
Sending buffer to 192.168.2.220 port 9
Sent... 15
Open Socket
Connected to 52.215.34.155 port 9 0
Sending buffer to 192.168.2.220 port 9
Sent... 15
Open Socket
Connected to 52.215.34.155 port 9 0
Sending buffer to 192.168.2.220 port 9
Sent... 15
Open Socket
Connected to 52.215.34.155 port 9 0
Sending buffer to 192.168.2.220 port 9
Sent... 15
Open Socket
Connected to 52.215.34.155 port 9 0
Sending buffer to 192.168.2.220 port 9
Sent... 15
Open Socket
Connected to 52.215.34.155 port 9 0
Sending buffer to 192.168.2.220 port 9
Sent... 15
Open Socket
Connected to 52.215.34.155 port 9 0
Sending buffer to 192.168.2.220 port 9
Sent... 15
Open Socket
Connected to 52.215.34.155 port 9 0
Sending buffer to 192.168.2.220 port 9
Sent... 15
Open Socket
Connected to 52.215.34.155 port 9 0
Sending buffer to 192.168.2.220 port 9
Sent... 15
Open Socket
Connected to 52.215.34.155 port 9 0
Sending buffer to 192.168.2.220 port 9
Sent... 15
Open Socket
Connected to 52.215.34.155 port 9 0
Sending buffer to 192.168.2.220 port 9
Sent... 15
Open Socket
Connected to 52.215.34.155 port 9 0
Sending buffer to 192.168.2.220 port 9
Sent... 15
Open Socket
Connected to 52.215.34.155 port 9 0
Sending buffer to 192.168.2.220 port 9
Sent... 15
Open Socket
Connected to 52.215.34.155 port 9 0
Sending buffer to 192.168.2.220 port 9
Sent... 15
Open Socket
Connected to 52.215.34.155 port 9 0
Sending buffer to 192.168.2.220 port 9
Sent... 15
Open Socket
Connected to 52.215.34.155 port 9 0
Sending buffer to 192.168.2.220 port 9
Sent... 15
Open Socket
Connected to 52.215.34.155 port 9 0
Sending buffer to 192.168.2.220 port 9
Sent... 15
Open Socket
Connected to 52.215.34.155 port 9 0
Sending buffer to 192.168.2.220 port 9
Sent... 15
Open Socket
Connected to 52.215.34.155 port 9 0
Sending buffer to 192.168.2.220 port 9
Sent... 15
Open Socket
Connected to 52.215.34.155 port 9 0
Sending buffer to 192.168.2.220 port 9
Sent... 15
Open Socket
Connected to 52.215.34.155 port 9 0
Sending buffer to 192.168.2.220 port 9
Sent... 15
Open Socket
Connected to 52.215.34.155 port 9 0
Sending buffer to 192.168.2.220 port 9
Sent... 15
Open Socket
Connected to 52.215.34.155 port 9 0
Sending buffer to 192.168.2.220 port 9
Sent... 15
Open Socket
Connected to 52.215.34.155 port 9 0
Sending buffer to 192.168.2.220 port 9
Sent... 15
Open Socket
... continues forever ...
```

**Conclusion:**

The problem described does not happen. Most probable problem is the server blocking further connections after one disconnecting.

SeppoTakalo, 2018-06-06:
Tested with Ublox EVK ODIN W2 and Mbed OS 5.8.6 using GCC

flipflop22, 2018-06-06:
Hi Thanks for confirming.

Indeed, it looks like the server is problematic. It has been error after it try to accept incoming connection. The problem is with its socket abstraction layer (sal) stack.

Sorry for being out of topic. But does Arm Mbed still supports Mbed OS 3. The problem I have is that the in Mbed OS 5, ODIN-W2 does not have Access Point in C++. And if I am using Mbed OS 3, the sal stack is problematic.

SeppoTakalo, 2018-06-06:
What do you mean with Access Point?

Mbed OS have never supported access points, but we have C++ class named `WiFiAccessPoint` which is result of `WifiInterface::scan()` and holds information about found access points.

Mbed OS 3 is not anymore supported. All focus is now in Mbed OS 5.

When you refer to error, do you mean that the server is also Mbed OS application? Documentation has an example of TCPServerSocket https://os.mbed.com/docs/v5.8/reference/tcpserver.html if you need to check. Also note that after `accept()` call you get a new TCPSocket which you need to `close()` before using it for another `accept()` call.

flipflop22, 2018-06-06:
Ublox ODIN-W2 supports Access Point which is included in their stack in C.
There is an example using this in Mbed OS3, refer to:
https://github.com/u-blox/ublox-odin-w2-drivers-binary-mbed-3/tree/master/test/wifi-ap

So I create a TCPserver class replacing the UDPTimeClient in the example above using yotta module sockets (TCPstream and TCPlisten).  The Access Point is working, but there is problem with the socket stack, since it returns an error when using socket_accept. Here is the TCPserver class I created.

`class TCPServer {
public:
    /**
     * The TCPServer Constructor
     * Initializes the server socket
     */
    TCPServer():
        _server(SOCKET_STACK_LWIP_IPV4), _stream(NULL),
        _disconnect_pending(false)
        {
            _server.setOnError(TCPStream::ErrorHandler_t(this, &TCPServer::onError), 1);
        }
    /**
     * Start the server socket up and start listening
     * @param[in] port the port to listen on
     * @return SOCKET_ERROR_NONE on success, or an error code on failure
     */
    void start(const uint16_t port)
    {
        do {
            socket_error_t err = _server.open(SOCKET_AF_INET4);
            if (_server.error_check(err)) break;			
            err = _server.bind("0.0.0.0",port);
            if (_server.error_check(err)){printf("MBED: Failed to bind \r\n"); break;}
            err = _server.start_listening(TCPListener::IncomingHandler_t(this, &TCPServer::onIncoming)); 
            if (_server.error_check(err)) break;
        } while (0);
		printf("TCP echo server started at port %d\r\n",port);
    }`

However in Mbed OS 5 now there is TCPserver. I will close this issue, since the issue has been addressed.

closed 2018-06-06 by flipflop22