ARMmbed/mbed-os #7137: UBLOX_EVK_ODIN_W2 - IAR - hard faults with mbed-os-example-client & Mbed OS 5.9

JanneKiiskila, 2018-06-06:
### Description

```
git clone git@github.com:ARMmbed/mbed-os-example-client.git
git checkout eth_odin_fix
cp configs/wifi_odin_v4.json mbed_app.json
```
Put in your WiFi SSID/password to `mbed_app.json`.
Replace `security.h` with the downloaded `security.h` from Mbed Connector.

```
compile -m UBLOX_EVK_ODIN_W2 -t IAR
```
Flash to device, start it.

- should run (GCC ARM, CC ARM versions do run & pass)
- IAR version crashes.

### Issue request type
    [] Question
    [ ] Enhancement 
    [X] Bug 

More details in; https://github.com/ARMmbed/mbed-os-example-client/pull/405#issuecomment-394971576

```
14:57:21.199 | D1 <<< DutThread: ++ MbedOS Fault Handler ++
14:57:21.200 | D1 <<< DutThread: FaultType: HardFault
14:57:21.200 | D1 <<< DutThread: Context:
14:57:21.200 | D1 <<< DutThread: R0 : FFFFFFFF
14:57:21.201 | D1 <<< DutThread: R1 : 2002A33C
14:57:21.201 | D1 <<< DutThread: R2 : 00000000
14:57:21.202 | D1 <<< DutThread: R3 : 080E844C
14:57:21.203 | D1 <<< DutThread: R4 : 00000000
14:57:21.203 | D1 <<< DutThread: R5 : 080D0599
14:57:21.203 | D1 <<< DutThread: R6 : 00000000
14:57:21.204 | D1 <<< DutThread: R7 : 00000004
14:57:21.204 | D1 <<< DutThread: R8 : 00000000
14:57:21.204 | D1 <<< DutThread: R9 : 20008860
14:57:21.205 | D1 <<< DutThread: R10 : 00000000
14:57:21.205 | D1 <<< DutThread: R11 : 00000000
14:57:21.205 | D1 <<< DutThread: R12 : 080BFCA1
14:57:21.206 | D1 <<< DutThread: SP : 2000BEA0
14:57:21.206 | D1 <<< DutThread: LR : 080B6C1D
14:57:21.206 | D1 <<< DutThread: PC : 080D05A4
14:57:21.206 | D1 <<< DutThread: xPSR : 61000000
14:57:21.207 | D1 <<< DutThread: PSP : 2000BE80
14:57:21.207 | D1 <<< DutThread: MSP : 20006DE8
14:57:21.207 | D1 <<< DutThread: CPUID: 410FC241
14:57:21.207 | D1 <<< DutThread: HFSR : 80000000
14:57:21.207 | D1 <<< DutThread: MMFSR: 00000000
14:57:21.208 | D1 <<< DutThread: BFSR : 00000000
14:57:21.208 | D1 <<< DutThread: UFSR : 00000000
14:57:21.208 | D1 <<< DutThread: DFSR : 0000000A
14:57:21.208 | D1 <<< DutThread: AFSR : 00000000
14:57:21.209 | D1 <<< DutThread: SHCSR: 00000000
14:57:21.209 | D1 <<< DutThread: Mode : Thread
14:57:21.210 | D1 <<< DutThread: Priv : Privileged
14:57:21.210 | D1 <<< DutThread: Stack: PSP
14:57:21.210 | D1 <<< DutThread: Threads Info:
14:57:21.210 | D1 <<< DutThread: Current:
14:57:21.211 | D1 <<< DutThread: State: 0x00000002 Entry: 0x080BCD41 Stack Size: 0x00001000 Mem: 0x2000AF48 SP: 0x2000BED8
14:57:21.211 | D1 <<< DutThread: Next:
14:57:21.212 | D1 <<< DutThread: State: 0x00000002 Entry: 0x080BCD41 Stack Size: 0x00001000 Mem: 0x2000AF48 SP: 0x2000BED8
14:57:21.212 | D1 <<< DutThread: Wait:
14:57:21.212 | D1 <<< DutThread: State: 0x00000083 Entry: 0x08130139 Stack Size: 0x00000300 Mem: 0x20026BF0 SP: 0x20026E88
14:57:21.213 | D1 <<< DutThread: Delay:
14:57:21.303 | D1 <<< DutThread: State: 0x00000043 Entry: 0x080BCD41 Stack Size: 0x00000200 Mem: 0x200082A8 SP: 0x20008410
14:57:21.303 | D1 <<< DutThread: Idle:
14:57:21.304 | D1 <<< DutThread: State: 0x00000001 Entry: 0x080C20C1 Stack Size: 0x00000200 Mem: 0x20027C80 SP: 0x20027E30
14:57:21.304 | D1 <<< DutThread: -- MbedOS Fault Handler --
14:57:21.304 | D1 <<< DutThread: ++ MbedOS Error Info ++
14:57:21.304 | D1 <<< DutThread: Error Status: 0x80ff013d Code: 317 Entity: 255
14:57:21.305 | D1 <<< DutThread: Error Message: System encountered an unrecoverable fault excaption, halting system.
14:57:21.305 | D1 <<< DutThread: Location: 0x80d05a4
```

JanneKiiskila, 2018-06-06:
@ARMmbed/team-ublox 
@SeppoTakalo

SeppoTakalo, 2018-06-06:
CC: @KariHaapalehto please study.

asifrizwanubx, 2018-06-06:
Tried to reproduce with the driver source code but didn't get the hard fault. We will look it soon.

asifrizwanubx, 2018-06-07:
Internal u-blox ref: TE_OM2-197.

KariHaapalehto, 2018-06-15:
Running out of memory. I did say from the heap traces that in this case heap allocation did fail.

asifrizwanubx, 2018-06-25:
@mikaleppanen there is no change/update in drivers related to thread stack sizes.

asifrizwanubx, 2018-06-25:
But there are changes related to allocation, static memory allocation mostly changed.

screamerbg, 2018-06-27:
@mikaleppanen @asifrizwanubx Is the https://github.com/ARMmbed/mbed-os/pull/7289 PR fix this issue?

mikaleppanen, 2018-06-27:
Yes, tested mbed-os-example-client with the changes and error does now occur anymore.

asifrizwanubx, 2018-06-27:
yes, I have verifed it.

0xc0170, 2018-06-28:
Thanks for the verification !

0xc0170, 2018-06-28:
Not yet merged, too fast closing :)

JanneKiiskila, 2018-07-10:
When will this get merged? Known issue list with the Mbed OS boards is just getting longer and longer...

mikaleppanen, 2018-08-01:
Correction has been merged:

https://github.com/ARMmbed/mbed-os/pull/7289

closed 2018-08-02 by 0xc0170