ARMmbed/mbed-os #7138: NRF52: without softdevice, crash on Thread::sleep - 5.9-rc1

drahnr, 2018-06-06:
### Description

  - What target does this relate to? NRF52_DK / NRF52832
  - What toolchain (name + version) are you using? GCC
  - What tools (name + version - is it mbed-cli, online compiler or IDE) are you using? mbed-cli 1.7.2 - mbed-os-5.9rc1
  - What is the SHA of Mbed OS (git log -n1 --oneline)? 6817dc43f
  - Steps to reproduce. (Did you publish code or a test case that exhibits the problem?)

Use this `app_config.json`
```
{
	"config": {},
	"target_overrides": {
		"*": {
			"target.stack_size" : 8192,

			"nvstore.enabled": true,
			"nvstore.max_keys": 32,
			"nvstore.area_1_address": "0x000F00000",
			"nvstore.area_1_size": "16384",
			"nvstore.area_2_address": "0x000FF0000",
			"nvstore.area_2_size": "16384",
	
			"target.macros_add": [
				"SWI_DISABLE0",
				"NRF52_PAN_12",
				"NRF52_PAN_15",
				"NRF52_PAN_20",
				"NRF52_PAN_30",
				"NRF52_PAN_31",
				"NRF52_PAN_36",
				"NRF52_PAN_51",
				"NRF52_PAN_53",
				"NRF52_PAN_54",
				"NRF52_PAN_55",
				"NRF52_PAN_58",
				"NRF52_PAN_62",
				"NRF52_PAN_63",
				"NRF52_PAN_64",
				"CONFIG_NFCT_PINS_AS_GPIOS",
				"NDEBUG=1",
				"OS_TASKCNT=3",
				"OS_IDLESTKSIZE=32",
				"OS_STKSIZE=4096",
				"OS_TIMERS=3",
				"OS_FIFOSZ=4",
				"OS_MUTEXCNT=3",
				"MBED_HEAP_STATS_ENABLED=1",
				"MBED_STACK_STATS_ENABLED=1",
				"MBED_SLEEP_TRACING_ENABLED=1",
				"NRF52832_XXAA",
				"DEVICE_SLEEP",
				"MBED_FAULT_HANDLER_DISABLED=1"
			],

			"target.macros_remove": [
				"CONFIG_GPIO_AS_PINRESET",
				"BLE_STACK_SUPPORT_REQD",
				"SOFTDEVICE_PRESENT",
				"S132"
			],
			"target.extra_labels_add" : [
				"SOFTDEVICE_NONE"
			],
			"target.extra_labels_remove" : [
				"SOFTDEVICE_S132_FULL",
				"SOFTDEVICE_COMMON"
			],
			"target.features_remove": [
				"BLE"
			]
		},
		"MY_TARGET" : {
			"nello.hardware_name": "\"BREADBOARD\"",
			"log.backend": "2"
		}
	}
}
```

And this `custom_targets.json`

```
{
    "MY_TARGET": {
        "inherits": ["MCU_NRF52832"],
        "release_versions": ["5"],
        "device_name": "nRF52832_xxAA",
        "bootloader_supported": true,
        "MERGE_BOOTLOADER": false,
        "MERGE_SOFTDEVICE": false
    }
}
```

When I try to start debugging, it instantly flunks out:

```
++ MbedOS Error Info ++
Error Status: 0x80010130 Code: 304 Entity: 1
Error Message: pinmap not found for function
Location: 0x6b01
Error Value: 0xffffffff
Current Thread: Id: 0x20004a34 Entry: 0x808d StackSize: 0x1000 StackMem: 0x20004a80 SP: 0x20005980 
-- MbedOS Error Info --         0 0 D 0x20004a34 🕱  heap: current: 4, max: 256
```

where the las chunk is part of my custom `HardFault_Handler`

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

marcuschangarm, 2018-06-20:
Are you using the ADC in your test application? 

It looks like you are requesting and ADC pin that is outside the map: https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/TARGET_MCU_NRF52832/PeripheralPins.c#L55-L65

drahnr, 2018-06-22:
@marcuschangarm sorry for the long response time, you are correct, `p0` was attempted to be used as `AnalogIn` in this case. Is there a way the error message can be improved without too much overhead?

marcuschangarm, 2018-06-25:
I agree, the error message is misleading.

@bulislaw @0xc0170 

I think the current error message should be changed to something like: "Provided pin %d not found in pinmap." and then there should be a separate check on whether the PinMap pointer is `NULL`.

https://github.com/ARMmbed/mbed-os/blob/master/hal/mbed_pinmap_common.c#L80-L89

AGlass0fMilk, 2018-07-02:
On a (somewhat) related note, how does mbed handle the NFC-protect pins (see UICR section of manual)? 

I have had maddening issues when trying to use NFC pins as MISO inputs (it doesn't work by default 😅) and scratching my head for a few hours until I read through the manual in a bit more detail.

Internal NFC protection circuitry is enabled by default unless 0 is written to the NFCPINS register of the UICR (0x1000120C)

drahnr, 2018-07-02:
@AGlass0fMilk complete unrelated :) hint: `CONFIG_NFCT_PINS_AS_GPIOS`

0xc0170, 2018-07-02:
> I think the current error message should be changed to something like: "Provided pin %d not found in pinmap." and then there should be a separate check on whether the PinMap pointer is NULL.

@donatieng Can you review this (the error is located in mbed_pinmap_common code file)

donatieng, 2018-07-04:
Thanks, will investigate

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTPAN-231*

closed 2018-11-05 by ciarmcom