ARMmbed/mbed-os #7142: DeviceKey made Mbed TLS Configuration changes that haven't been upstreamed

sbutcher-arm, 2018-06-06:
### Description
PR  #6642 - "Device key implementation" made modifications to Mbed TLS configuration files (such as `features/mbedtls/inc/mbedtls/config.h`) but those modifications have not been upstreamed to the [Mbed TLS repository](https://github.com/ARMmbed/mbedtls), meaning when we import the next release of Mbed TLS, those changes will be lost. 

This needs immediate correction to ensure the changes aren't lost with the next Mbed TLS release.

### Issue request type

[ ] Question  
[ ] Enhancement  
[X ] Bug  

cc: @k-stachowiak

0xc0170, 2018-06-06:
@yossi2le

yossi2le, 2018-06-06:
@sbutcher-arm Hi,
I need just a short explanation of how to do it. What is the correct repository of the mbed-tls that i should fork from?

sbutcher-arm, 2018-06-19:
This issue was resolved by PR #7144 'Add configuration options to enable CMAC in mbedtls by default' and PR #7135 'Update Mbed TLS to version 2.10.0'. I think it can now be closed.

@yossi2le - Changes to Mbed TLS should be upstreamed to [Mbed TLS](http://github.com/ARMmbed/mbedtls/). However, the DeviceKey change didn't need to do that, just change the import script, which is what #7144 does.

yossi2le, 2018-06-19:
@sbutcher-arm 
Thanks for the info, everyday learning something new.