ARMmbed/mbed-os #7149: FLASH test for small targets

jeromecoutant, 2018-06-06:
### Description

Let's take an example with NUCLEO_F410RB:
Flash organization is :
- 4 sectors of 16 Kbytes
- 1 sector of 64 Kbytes

flashiap test flash size is more than 48K
and this test is using the 2 last sectors (program across sectors test).

=> Test should can not be supported...

Is there a way to add a check, in the test, for the available flash size ?

### Issue request type

[ ] Question  
[x] Enhancement  
[ ] Bug

dannybenor, 2018-06-10:
@jeromecoutant I think there is no point allocating a sector for data in this limited board. I prefer that the test will fail to indicate that.

davidsaada, 2018-06-10:
@jeromecoutant Not sure I understand the problem. The cross sector test doesn't allocate memory of sector sizes, but of program size (page size). Here the test allocates 2*page_size. Don't know how much it is in this board, but it should be much smaller than the sector size.

jeromecoutant, 2018-06-11:
Hi

Yes, tests-mbed_hal-flash is OK
But tests-mbed_drivers-flashiap is not :-(

        test case: 'FlashIAP - init' ................................................................. OK in 0.04 sec
        test case: 'FlashIAP - program' .............................................................. OK in 2.23 sec
        test case: 'FlashIAP - program across sectors' ............................................... ERROR in 0.00 sec
        test case: 'FlashIAP - program errors' ....................................................... SKIPPED in 0.00 sec

Here is NUCLEO_F410RB information:
page_size 1
flash_start 0x8000000
flash_size 131072

davidsaada, 2018-06-11:
Very strange. If page_size is 1, then it means that the test allocates 2 bytes for program data (as you can see [here](https://github.com/davidsaada/mbed-os/blob/master/TESTS/mbed_drivers/flashiap/main.cpp#L52)) and then another 2 bytes for reading the flashed data. Can you paste the output telling why the test fails?

jeromecoutant, 2018-06-11:
But programming is erasing all at line 102 ?

Test is failing with an hardfault:

[1528707204.19][CONN][RXD] ++ MbedOS Fault Handler ++
[1528707204.19][CONN][RXD]
[1528707204.20][CONN][RXD] FaultType: HardFault

davidsaada, 2018-06-11:
Erasing doesn't allocate the full sector size - it just issues an erase command to the flash. Here (line 102), the test erases the two sectors under test. You can see that the error is a hard fault and not an allocation problem. I'll try to investigate this problem - maybe there's a problem in the address/size calculation.

davidsaada, 2018-06-11:
Sorry for the misunderstanding. Programming doesn't erase all, but the last two sectors. Thought it was a heap problem. Yes, it is indeed a bug for small FLASH boards. Will fix.

closed 2018-08-21 by cmonr