ARMmbed/mbed-os #7153: All STM should have tickless mode enabled

janjongboom, 2018-06-06:
### Description

STM32 HAL should have low-power ticket enabled and have sleep enabled (it's the same file for all STM32 targets). Why is tickless mode not enabled by default for these devices?

### Issue request type

[X] Question  
[ ] Enhancement  
[ ] Bug

jeromecoutant, 2018-06-07:
Hi

> Why is tickless mode not enabled by default for these devices?

Maybe because ticker branch has been merged into master branch only 1 week ago...?

0xc0170, 2018-06-07:
@janjongboom Have you checked latest master, or 5.9 release candidate (its on the 5.9 branch) ?

jeromecoutant, 2018-06-07:
I have also understood from ARM that plan is to enable tickless automatically if sleep and lpticker is enabled ?
But maybe only for 5.10 ?

jeromecoutant, 2018-06-27:
For information, tests are still on going.

Waiting for #7328

screamerbg, 2018-09-13:
Bump @0xc0170

jeromecoutant, 2018-10-02:
Quick status:
Waiting for #8219 #8223 #8242 #8184 #8286 #7582 #8279 #7849

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTPART-5496*

atoy40, 2018-12-15:
@jeromecoutant any news about this one ? all issues you've mentionned are now closed or merged. IS TICKLESS ready to use with STM board ?
Thanks.

jeromecoutant, 2018-12-18:
Hi
I pushed PR #9134 to enable TICKLESS mainly for L0/L4/F7

jeromecoutant, 2019-10-11:
@MarceloSalazar @bulislaw 
I think general tickless work has been done.
Now each remaining issues or enhancement should be tracked by a specific issue.
Are you OK?
Thx

closed 2019-10-25 by 0xc0170