ARMmbed/mbed-os #7154: wait locks deepsleep, Thread::wait does not... What?

janjongboom, 2018-06-06:
### Description

In our docs (https://os.mbed.com/docs/v5.7/reference/sleep-manager.html) we state that when you call `wait` we automatically put the device in deep sleep. This is not true. For some reason `wait()` blocks deepsleep, but if you change it to `Thread::wait` it does not. Why?! This is super confusing. Why would there be two different behaviors when pausing a thread?

Then on non-RTOS mode wait actually can do deepsleep, so that makes it even more confusing.

All wait should behave equally and should not block deepsleep.

### Issue request type

[ ] Question  
[X] Enhancement  
[ ] Bug

c1728p9, 2018-06-06:
There was a PR for this change but it was rejected since it breaks backwards compatibility:
https://github.com/ARMmbed/mbed-os/pull/6969

kjbracey-arm, 2018-06-07:
I missed #6969 - I would have approved it with a mod to split long and short delay behaviour on the float version.

I don't think this long-running API glitch can be justified on backwards compatibility grounds - we do change bigger things in minor releases.

The most likely breakage would be people doing `wait(1)` to get 1 milllisecond. If we change that to just ticks, they could get as little as 0 milliseconds, which might break some hardware timing.

I wonder if following the 5.9 lpticker changes we could do a compromise solution which uses the lpticker to achieve better-than-rtos-tick precision without the lock?

kjbracey-arm, 2018-06-07:
> Then on non-RTOS mode wait actually can do deepsleep,

No it can't, it spins and uses 100% CPU.

klaas019, 2018-08-23:
Could someone update this example https://os.mbed.com/docs/v5.6/reference/sleep-manager.html . It is broken.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-23*

closed 2018-11-26 by ciarmcom