ARMmbed/mbed-os #7155: Error occurs when building exported mbed-os programs on e2studio

TomoYamanaka, 2018-06-07:
### Description

I faced the several build error when compiled the exported **mbed-os programs** on Renesas' IDE **e2studio**.
The build errors are a bit different by programs, but I frequently look messages such as "No such file or directory".

The procedure is as follows:
1. Export the program on Online Compiler
2. Build the exported program on e2studio

The error that is shown when building exported `mbed-os-example-blinky(mbed os 5.8.5)` on e2studio is as follows:
```
../mbed-os/features/storage/FEATURE_STORAGE/storage-volume-manager/source/storage_volume.cpp:25:59: fatal error: storage-volume-manager/storage_volume_manager.h: No such file or directory
 #include "storage-volume-manager/storage_volume_manager.h"
                                                           ^
../mbed-os/features/storage/FEATURE_STORAGE/storage-volume-manager/source/storage_volume_manager.cpp:25:59: fatal error: storage-volume-manager/storage_volume_manager.h: No such file or directory
 #include "storage-volume-manager/storage_volume_manager.h"
                                                           ^
../mbed-os/features/storage/FEATURE_STORAGE/flash-journal/flash-journal-strategy-sequential/flash_journal_crc.c:18:65: fatal error: flash-journal-strategy-sequential/flash_journal_crc.h: No such file or directory
 #include "flash-journal-strategy-sequential/flash_journal_crc.h"
                                                                 ^
make: *** [mbed-os/features/storage/FEATURE_STORAGE/storage-volume-manager/source/storage_volume_manager.o] Error 1
mbed-os/features/storage/FEATURE_STORAGE/storage-volume-manager/source/subdir.mk:21: recipe for target 'mbed-os/features/storage/FEATURE_STORAGE/storage-volume-manager/source/storage_volume_manager.o' failed
make: *** Waiting for unfinished jobs....
make: *** [mbed-os/features/storage/FEATURE_STORAGE/storage-volume-manager/source/storage_volume.o] Error 1
mbed-os/features/storage/FEATURE_STORAGE/storage-volume-manager/source/subdir.mk:21: recipe for target 'mbed-os/features/storage/FEATURE_STORAGE/storage-volume-manager/source/storage_volume.o' failed
make: *** [mbed-os/features/storage/FEATURE_STORAGE/flash-journal/flash-journal-strategy-sequential/flash_journal_crc.o] Error 1
mbed-os/features/storage/FEATURE_STORAGE/flash-journal/flash-journal-strategy-sequential/subdir.mk:24: recipe for target 'mbed-os/features/storage/FEATURE_STORAGE/flash-journal/flash-journal-strategy-sequential/flash_journal_crc.o' failed
../mbed-os/features/storage/FEATURE_STORAGE/flash-journal/flash-journal-strategy-sequential/strategy.c:18:65: fatal error: flash-journal-strategy-sequential/flash_journal_crc.h: No such file or directory
 #include "flash-journal-strategy-sequential/flash_journal_crc.h"
                                                                 ^
make: *** [mbed-os/features/storage/FEATURE_STORAGE/flash-journal/flash-journal-strategy-sequential/strategy.o] Error 1
mbed-os/features/storage/FEATURE_STORAGE/flash-journal/flash-journal-strategy-sequential/subdir.mk:24: recipe for target 'mbed-os/features/storage/FEATURE_STORAGE/flash-journal/flash-journal-strategy-sequential/strategy.o' failed

```

With respect to file of "No such file or directory", it seems that file was exported but not included headers in the include paths.

Anyway, I think that this should **not** be the expected behavior, could you confirm the above? 

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

toyowata, 2018-06-20:
@theotherjimmy 

Probably, it is related this?
https://github.com/ARMmbed/mbed-os/issues/7129

theotherjimmy, 2018-06-20:
@toyowata That's a good guess, but I was under the impression that GNU MCU Eclipse could handle the extra long paths without issues.

theotherjimmy, 2018-06-20:
Also, you'll notice that the error is different in this issue than in #7129

TomoYamanaka, 2018-06-25:
@toyowata @theotherjimmy 

I verified Export by following two patterns.
<br><br>
**By Online Compiler**
As shown the above, I faced the "No such file or directory" error when building the exported program.
Since the corresponding files(Following 5 passes) look like **unnecessary include**, I set them outside of build target on e2studio as a trial.
```
features/FEATURE_BLE
features/FEATURE_COMMON_PAL
features/FEATURE_UVISOR
features/nanostack
features/storage
```
As the result of it, the following build error was shown.
```
arm-none-eabi-g++ @"mbed-os-example-blinky.elf.in"
arm-none-eabi-g++: error: unrecognized command line option '--wrap=main'; did you mean '--warn-main'?
arm-none-eabi-g++: error: unrecognized command line option '--wrap=_malloc_r'
arm-none-eabi-g++: error: unrecognized command line option '--wrap=_free_r'
arm-none-eabi-g++: error: unrecognized command line option '--wrap=_realloc_r'; did you mean '--warn-realloc-lhs'?
arm-none-eabi-g++: error: unrecognized command line option '--wrap=_memalign_r'
arm-none-eabi-g++: error: unrecognized command line option '--wrap=_calloc_r'
arm-none-eabi-g++: error: unrecognized command line option '--wrap=exit'; did you mean '--graphite'?
arm-none-eabi-g++: error: unrecognized command line option '--wrap=atexit'; did you mean '--graphite'?
makefile:130: recipe for target 'mbed-os-example-blinky.elf' failed
make: *** [mbed-os-example-blinky.elf] Error 1
```

**By Mbed CLI**
I built the exported program by running the following command.
```
$ mbed import http://os.mbed.com/teams/mbed-os-examples/code/mbed-os-example-blinky/
$ cd mbed-os-example-blinky
$ mbed export -m RZ_A1H -i e2studio
Build the exported program on e2studio
```
As the result of it, the following build error was shown(same error mmesage with the above).
```
arm-none-eabi-g++ @"mbed-os-example-blinky.elf.in"
arm-none-eabi-g++: error: unrecognized command line option '--wrap=main'; did you mean '--warn-main'?
arm-none-eabi-g++: error: unrecognized command line option '--wrap=__malloc_r'
arm-none-eabi-g++: error: unrecognized command line option '--wrap=__free_r'
arm-none-eabi-g++: error: unrecognized command line option '--wrap=__realloc_r'
arm-none-eabi-g++: error: unrecognized command line option '--wrap=__memalign_r'
arm-none-eabi-g++: error: unrecognized command line option '--wrap=__calloc_r'
arm-none-eabi-g++: error: unrecognized command line option '--wrap=exit'; did you mean '--graphite'?
arm-none-eabi-g++: error: unrecognized command line option '--wrap=atexit'; did you mean '--graphite'?
makefile:185: recipe for target 'mbed-os-example-blinky.elf' failed
make: *** [mbed-os-example-blinky.elf] Error 1
```

To summarize the above, I think that check point is the below.<br>
- When exporting from Online Compiler, the **unnecessary include path** seems to set.<br>
- Whichever export method is used, the **--wrap** option related problem seems to be blocking the build on e2studio.<br>

theotherjimmy, 2018-06-25:
@TomoYamanaka That's not an "unnecessary include path", that's the include paths from `mbed compile`.

the `--wrap` options are required to link. The problem is that `g++` is being called instead of the linker. That's very odd.

TomoYamanaka, 2018-06-28:
@theotherjimmy 

I investigated several points.

**With respect to "No such file or directory" error :**
1. When building mbed-os program by Mbed LCI, this error is not shown.
```
$ mbed compile -m RZ_A1H -t GCC_ARM
```

2. Also when exporting mbed-os program by Mbed LCI, this error is not shown.
```
$ mbed import http://os.mbed.com/teams/mbed-os-examples/code/mbed-os-example-blinky/
$ cd mbed-os-example-blinky
$ mbed export -m RZ_A1H -i e2studio
Build the exported program on e2studio
```

3. Only when exporting mbed-os program by Online Compiler, this error is shown.
```
Select "Export program" and "Export Toolchain is e2studio" on Online compiler
Build the exported program on e2studio
```

It seems like there is something issue with export from Online, could you let me know your opinion on this ?<br><br>
**With respect to "--wrap" related error :**
I tried adding `-Wl` prefix manually in .cproject file as below.
```
<option id="ilg.gnuarmeclipse.managedbuild.cross.option.cpp.linker.flags.187528822" superClass="ilg.gnuarmeclipse.managedbuild.cross.option.cpp.linker.flags" valueType="stringList">
	<listOptionValue builtIn="false" value="-Wl,--wrap=main"/>
	<listOptionValue builtIn="false" value="-Wl,--wrap=_malloc_r"/>
	<listOptionValue builtIn="false" value="-Wl,--wrap=_free_r"/>
	<listOptionValue builtIn="false" value="-Wl,--wrap=_realloc_r"/>
	<listOptionValue builtIn="false" value="-Wl,--wrap=_memalign_r"/>
	<listOptionValue builtIn="false" value="-Wl,--wrap=_calloc_r"/>
	<listOptionValue builtIn="false" value="-Wl,--wrap=exit"/>
	<listOptionValue builtIn="false" value="-Wl,--wrap=atexit"/>
</option>
```

As the result of it, the build was successful.
Reference issue is here : https://github.com/ARMmbed/mbed-os/issues/4624

TomoYamanaka, 2018-07-04:
@theotherjimmy 
How is it going?

theotherjimmy, 2018-07-18:
Sorry about that @TomoYamanaka 

#7410 is not related to this. It may look like it because "exporter" but that's where the relevance stops.

-------------------------

Here's some pointers that may help you resolve the problem yourself:
 * https://github.com/ARMmbed/mbed-os/blob/master/tools/export/e2studio/.cproject.tmpl#L355-L357: The part of the template that deals with the linker flags
 * https://github.com/ARMmbed/mbed-os/blob/master/tools/export/e2studio/__init__.py#L33 The source of your jinja2 context. 
 * https://github.com/ARMmbed/mbed-os/blob/master/tools/export/gnuarmeclipse/__init__.py#L173-L174 Folder exclusion variable name (this is missing from the `.cproject` template, and it's absence possibly responsible for the extra includes)

TomoYamanaka, 2018-07-23:
Could you help me to resolve the following two points because this is a blocker for Renesas Mbed user's activity?
  - "No such file or directory" error (only when exporting on Online)
  - "--wrap" related error

theotherjimmy, 2018-07-23:
@TomoYamanaka You could extend the jinja template to include the `--wrap`. The "no such file or directory" errors sound like something else...

phyokyaw, 2018-07-24:
Thanks for the reply @theotherjimmy. @TomoYamanaka as for --warp problem since e2 studio template extends gnu arm eclipse template, I'll update the jinja template on GNU ARM exporter to fix both. As for not adding include paths (No such file or directory error) I am not sure why some include paths are added but not others. I can look into it the next week.

Thanks
Phyo

theotherjimmy, 2018-07-24:
@phyokyaw The missing include paths are related to features that are disabled. They're supposed to show up in the export (if I recall correctly), and be disabled. It looks to me like the e2studio is missing the second part.

theotherjimmy, 2018-07-24:
@phyokyaw The e2studio template does not extend the GNU ARM Eclipse template. It does use the same variables though.

TomoYamanaka, 2018-07-25:
@phyokyaw @theotherjimmy Thank you for your comments.
@phyokyaw I'll be looking forward to your update.

phyokyaw, 2018-08-03:
Hi @theotherjimmy,

Thank for your comment. I am looking into the issue regards to excluding source files to see why there are "No such file or directory" errors. When we export using mbed-cli `<entry excluding .../> `entries in the exported .cproject contains the right path. E.g. for FEATURE_STORAGE

`<sourceEntries>
   <entry excluding="...|./mbed-os/features/storage/FEATURE_STORAGE|... />
</sourceEntries>`

However in .cproject file exported from online compiler contains 

`<sourceEntries>
   <entry excluding="...|/filer/web_data/repo_dicts/github.com/armmbed/mbed-os/main/features/storage/FEATURE_STORAGE|..." />
</sourceEntries>`

Hence the compiler is not excluding the ignored files/directories. Looks like its referencing paths from online mapped repo rather than local path. Do you have view on how to make the excluded items reflect the correct path?

Thanks
Phyo

theotherjimmy, 2018-08-03:
@phyokyaw Thanks for looking into this. I'll take a look at why that's going wrong.

phyokyaw, 2018-08-08:
@TomoYamanaka @theotherjimmy 

Just for FYI, I also noticed that in 5.9 mbed-os/features/nanostack/sal-stack-nanostack/source/configs folder is missing for Web exporter. The problem has been reported for other exporters.

[https://os.mbed.com/questions/81597/Problems-when-exporting-to-uVision5-with/](https://os.mbed.com/questions/81597/Problems-when-exporting-to-uVision5-with/)

[https://os.mbed.com/forum/bugs-suggestions/topic/32790/?page=1#comment-58923](https://os.mbed.com/forum/bugs-suggestions/topic/32790/?page=1#comment-58923)

Once I manually copied the configs/ folder and resolved the excluded path I am able to build and debug from the exported project.

Thanks
Phyo

0xc0170, 2018-08-08:
This issue was reported here https://github.com/ARMmbed/mbed-os/issues/7615. It shall be fixed for 5.9.5

TomoYamanaka, 2018-08-09:
@phyokyaw 
Thank you for information. You updated about the excluded path problem, on the other hand, is there any update about the "--warp" problem?

phyokyaw, 2018-08-10:
@theotherjimmy Please review the PR if possible. This is to fix the --wrap error.

Thank you.
Phyo

TomoYamanaka, 2018-08-28:
@0xc0170  
Could you re-open this issue? There is still one problem that should be resolve. It seem that I can't reopen.

`"No such file or directory" error (only when exporting on Online Compiler)`

@theotherjimmy 
Is there anything update about the investigation result that @phyokyaw  mentioned(https://github.com/ARMmbed/mbed-os/issues/7155#issuecomment-410286102)?

TomoYamanaka, 2018-09-03:
@theotherjimmy 
How are you doing after that?
We will appreciate if you could inform us about the current status regarding the missing include paths.

TomoYamanaka, 2018-09-21:
@theotherjimmy 
Since this issue keeps to block Renesas Mbed user's activity, we would be grateful if you would resolve this quickly.

> @phyokyaw Thanks for looking into this. I'll take a look at why that's going wrong.

Would you please tell us update about the above?

0xc0170, 2018-09-21:
@theotherjimmy is out of office this week. @cmonr Who else could help?

theotherjimmy, 2018-10-02:
@TomoYamanaka The issue with `mbed_config.h` including filer paths is gone.

theotherjimmy, 2018-10-02:
And the `compute_exclusions` bug is still there.

theotherjimmy, 2018-10-02:
See #8303 for the `compute_exclusions` fix.

TomoYamanaka, 2018-10-04:
@theotherjimmy Thank you for your update. We'll check #8303.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-14*

closed 2018-10-19 by cmonr

TomoYamanaka, 2018-12-06:
@theotherjimmy 

I checked whether this issue could be solved by exporting `mbed-os-example-blinky(Mbed OS 5.10.4)` on Mbed Online Compiler, unfortunately the problem seems to occur yet. (the exported program by CLI was no problem.)
The build error on e2studio is here.
```
../mbed-os/features/storage/FEATURE_STORAGE/storage-volume-manager/source/storage_volume.cpp:25:59: fatal error: storage-volume-manager/storage_volume_manager.h: No such file or directory
 #include "storage-volume-manager/storage_volume_manager.h"
```

Also,  in the .cproject file exported from Online Compiler, I found that a part of excluding path is still mapped online path, not local path. (expected path is local path and this path by CLI is local path)
ex:  FEATURE_STORAGE
**by Mbed CLI**
`<sourceEntries> <entry excluding="...|./mbed-os/features/storage/FEATURE_STORAGE|... /> </sourceEntries>`
**by Online Compiler**
`<sourceEntries> <entry excluding="...|/filer/web_data/repo_dicts/github.com/armmbed/mbed-os/2fd0c5cfbd83fce62da6308f9d64c0ab64e1f0d6/features/storage/FEATURE_STORAGE|... /> </sourceEntries>`

<br>
Although the solution PR #8303 was already merged into Mbed OS 5.10.3, could you make sure whether this contents reflected to Online Exporter environment?

theotherjimmy, 2018-12-06:
@TomoYamanaka #8303 is not in the online compiler.

TomoYamanaka, 2018-12-07:
@theotherjimmy 

I'm in trouble with export from online.
My understanding is that PR #8303 solves to generate unusable project file from online compiler, and this solution should be applied to online environment not only mbed-os/tools. Also I know there is time rag regarding merge into online exporter. 
Are those correct?

cmonr, 2018-12-07:
I'm actually confused as well. @theotherjimmy This is at least in 5.10.3, correct? I'm missing the disconnect as well.

theotherjimmy, 2018-12-07:
I don't see what's so confusing. This is in 5.10.3. This is not in the online compiler.

TomoYamanaka, 2018-12-07:
@theotherjimmy 

Could you tell me why this is not in the online compiler? 
I think this should also be in online compiler because it affects project files generated from online.

theotherjimmy, 2018-12-10:
> Could you tell me why this is not in the online compiler?

Because that requires a tools-release that has not been made. I'll be making a tools release for 5.11.0, and updating the online compiler to use that in the future.

TomoYamanaka, 2018-12-11:
@theotherjimmy 

Thank you for your reply.

I understood that PR #8303 will be incorporated in next tools-release timing (in synchronization with mbed os 5.11.0).
I will confirm this issue again after next tools-release was published.

May I ask you something? Is the next tools-release that you mentioned different from "tools-release-5.11.0-rc1" ?

theotherjimmy, 2018-12-11:
> Is the next tools-release that you mentioned different from tools-release-5.11.0-rc1 ?

Yes. tools-release-5.11.0-rc1 is based on mbed-os-5.11.0-rc1, and the next tools release will be based on mbed-os-5.11.0.

TomoYamanaka, 2019-01-15:
@theotherjimmy 

Although [tools-release-5.11.0](https://github.com/ARMmbed/mbed-os/releases/tag/tools-release-5.11.0) was already released, when I build [mbed-os-example-blinky](https://os.mbed.com/teams/mbed-os-examples/code/mbed-os-example-blinky/) after exporting from Mbed Online Compiler to e2studio, I faced this issue yet.

Could you check to whether this tools-release took into Online Expoter?

theotherjimmy, 2019-01-15:
@TomoYamanaka The tools release for 5.11 is not yet in the online compiler.

TomoYamanaka, 2019-01-17:
@theotherjimmy 

When will it be ready? We will be grateful if you could let me know the scheduled date that is incorporated to Online Compiler.

TomoYamanaka, 2019-02-15:
@theotherjimmy 

Thank you for continued support, I confirmed this issue was finally solved. 
FYI, the program I tried by exporting is  `mbed-os-example-blinky` Rev.91.

PR #7758 (2018-08-27): Include user defined static lib files in the Linker object files paths
### Description

This is to fix linker not finding symbols declared in object files inside static library files. 

For this case throwing:-

> rm-none-eabi-g++: error: unrecognized command line option '--wrap=main'; did you mean '--warn-main'?

For this case to include libmbed.a which contains

mbed_alloc_wrappers.o:
         U __real__calloc_r
         U __real__free_r
         U __real__malloc_r
         U __real__memalign_r
         U __real__realloc_r
00000000 T __wrap__calloc_r
00000000 T __wrap__free_r
00000000 T __wr [... cut]
commit 87304f88 (2018-08-27): Updated to include user defined static lib files in the Linker object files path.
### Description

This is to fix linker not finding symbols declared in object files inside static library files.

For this case throwing:-

> rm-none-eabi-g++: error: unrecognized command line option '--wrap=main'; did you mean '--warn-main'?

For this case to include libmbed.a which contains

mbed_alloc_wrappers.o:
         U __real__calloc_r
         U __real__free_r
         U __real__malloc_r
         U __real__memalign_r
         U __real__realloc_r
00000000 T __wrap__calloc_r
00000000 T __wrap__free_r
00000 [... cut]
PR #8303 (2018-10-19): Tools: Use logical paths in ignored_dirs
### Description

Exporting to GNU ARM Eclipse, E2 Studio, and other exclude-based IDEs
currently generats unusable project files online. This is because the
list of directories ignored in the scan is inconsistant about what sort
of paths are used: logical paths, or phisical paths. This patch makes
all paths in ignored_dirs logical. This should fix the excluding tags
in these project files.

Resolves #7155
Tested on partnerbeta.
Includes minor formatting fixes that I found at the time as seperate
commits.

### Pull request type

    [x] Fix
    [ ] Refac [... cut]