ARMmbed/mbed-os #7168: Serious regression on RF52 idle power consumption

NeilMacMullen, 2018-06-07:
The current 5.9RC( and head of master) show very poor power consumption when executing

  while (1) Thread::wait(5000);

The attached picture shows two traces on the same RF52 based hardware platform.  Idle power has increased from ~180uA to 6.4mA suggesting that the processor is not idling at all.

[image]

NeilMacMullen, 2018-06-07:
Adding @MarceloSalazar and @marcuschangarm as interested parties.

jrobeson, 2018-06-07:
related? https://github.com/ARMmbed/mbed-os/issues/7154

kjbracey-arm, 2018-06-08:
> related? #7154

Don't think so. The lock discussed there has been present since 5.7.

Perhaps this is relevant: https://github.com/ARMmbed/mbed-os/pull/7073

0xc0170, 2018-06-08:
> Perhaps this is relevant: #7073

👍  @maciejbocianski Can you confirm?

@NeilMacMullen Thanks for reporting

maciejbocianski, 2018-06-08:
I think so, too.
`SLEEP` was disabled during HAL feature branches merging

0xc0170, 2018-06-08:
@NeilMacMullen This should be fixed in 5.9.1 (3 patches are waiting for CI). You can cherry-pick 7073 (it has some dependencies on another 1-2 PR opened) to fix this and report back that is not causing any regression.

NeilMacMullen, 2018-06-08:
I can confirm that https://github.com/ARMmbed/mbed-os/pull/7073/commits/f6d3058899636cfd22a8b6273f67e37fd55d24f9 does seem to fix the issue.

closed 2018-06-22 by cmonr