ARMmbed/mbed-os #7189: Thread with callback is not working

flipflop22, 2018-06-12:
### Description

my setup:
-UBLOX EVK ODIN W2
-GCC ARM Embedded Toolchain
-Mbed-CLI

I want to send SPI data continously paralel with my functions. So I create a thread.

`Thread spi_thread;`

However, when I want to run the function using callback.....

`  spi_thread.start(callback(receiveSPI,&spi));`

it gives me this error:

`[Error] main.cpp@146,46: no matching function for call to 'callback(void (&)(mbed::SPI), mbed::SPI*)'
[ERROR] .\main.cpp: In function 'int main()':
.\main.cpp:146:46: error: no matching function for call to 'callback(void (&)(mbed::SPI), mbed::SPI*)'
     spi_thread.start(callback(sendSPI,&spi));`

NOTE: I have Initialize an SPI pins and acquire them with number of bits, mode, and freq.
### Issue request type

[ ] Question  
[ ] Enhancement  
[x] Bug

pan-, 2018-06-12:
The signature of sendSPI is not correct as it takes the spi argument by value; try a pointer instead: 
`void sendSPI(SPI* spi)`.

flipflop22, 2018-06-13:
Thanks, it works

closed 2018-06-13 by flipflop22