ARMmbed/mbed-os #7190: Application Header - mbed compile warning

DBS06, 2018-06-12:
### Description
I added an application header to my mbed_app.json like described here [Adding an application header](https://os.mbed.com/docs/v5.8/tutorials/bootloader.html). So far so good, but when I do this I get from the command line with mbed compile the following error:
`[Warning] <command-line>@0,0: "HEADER_ADDR" redefined`

I found out that the warning is a result at calling the arm-none-eabi, i.e.:
`[DEBUG] Compile: arm-none-eabi-g++ -std=gnu++11 -fno-rtti -Wvla -c -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -fno-builtin -ffunction-sections -fdata-sections -funsigned-char -MMD -fno-delete-null-pointer-checks -fomit-frame-pointer -Os -DNDEBUG -g1 -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -DBOOTLOADER_ADDR=0x0 -DBOOTLOADER_SIZE=0xc800 -DHEADER_ADDR=0xc800 -DHEADER_SIZE=0x10 -DAPPLICATION_ADDR=0xc810 -DAPPLICATION_SIZE=0x3fff0 -DHEADER_ADDR=0x4c800 -DHEADER_SIZE=0x10 -DPOST_APPLICATION_ADDR=0x4c810 -DPOST_APPLICATION_SIZE=0xb37f0 -DDEVICE_SPI=1 -DMBED_BUILD_TIMESTAMP=1528795199.65 -D__MBED__=1 -DTARGET_1024K -D__FPU_PRESENT=1 -DDEVICE_PORTINOUT=1 -DTARGET_RTOS_M4_M7 -DTARGET_EFM32PG12 -DDEVICE_LOWPOWERTIMER=1 -DDEVICE_RTC=1 -DTOOLCHAIN_object -DDEVICE_SERIAL_ASYNCH=1 -D__CMSIS_RTOS -DTRANSACTION_QUEUE_SIZE_SPI=4 -DMBEDTLS_CONFIG_HW_SUPPORT -DTOOLCHAIN_GCC -DDEVICE_SPISLAVE=1 -DARM_MATH_CM4 -DDEVICE_I2C_ASYNCH=1 -DTARGET_LIKE_CORTEX_M4 -DTARGET_M4 -DTARGET_UVISOR_UNSUPPORTED -DDEVICE_SPI_ASYNCH=1 -DDEVICE_SERIAL=1 -DMBED_TICKLESS -DDEVICE_INTERRUPTIN=1 -DTARGET_CORTEX -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -DDEVICE_I2CSLAVE=1 -D__CORTEX_M4 -DDEVICE_STDIO_MESSAGES=1 -DTARGET_LIKE_MBED -DEFM32PG12B500F1024GL125 -DDEVICE_PORTIN=1 -DTARGET_RELEASE -DTARGET_EFM32 -DDEVICE_TRNG=1 -D__MBED_CMSIS_RTOS_CM -DDEVICE_SLEEP=1 -DTOOLCHAIN_GCC_ARM -DTARGET_EFM32PG12_STK3402 -DTARGET_Silicon_Labs -DTARGET_SL_CRYPTO -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DTARGET_EFM32PG12B500F1024GL125 -DDEVICE_FLASH=1 -DTARGET_CORTEX_M @BUILD_RELEASE\bootloader-app\EFM32PG12_STK3402\.includes_439041110852595a2173d174c3f79d80.txt -include BUILD_RELEASE\bootloader-app\EFM32PG12_STK3402\mbed_config.h -MD -MF BUILD_RELEASE\bootloader-app\EFM32PG12_STK3402\mbed-os\drivers\RawSerial.d -o BUILD_RELEASE\bootloader-app\EFM32PG12_STK3402\mbed-os\drivers\RawSerial.o .\mbed-os\drivers\RawSerial.cpp`

If you look at the command you will recognize that `-DHEADER_SIZE` and `-DHEADER_ADDR` is defined twice.

I think the problem is a result from this mbed compile output:
> Using regions in this build:
  Region bootloader size 0xc800, offset 0x0
  Region header size 0x10, offset 0xc800
  Region application size 0x3fff0, offset 0xc810
  Region header size 0x10, offset 0x4c800
  Region post_application size 0xb37f0, offset 0x4c810

You can see that the `Region header` is there twice and I think that this is false.

### Issue request type
[ ] Question  
[ ] Enhancement  
[X] Bug

DBS06, 2018-06-12:
@ciarmcom what does 
> ARM Internal Ref: MBOTRIAGE-665
mean?

theotherjimmy, 2018-06-12:
That our internal bug tracker has mirrored this issue.

theotherjimmy, 2018-06-12:
@DBS06 Could you test #7196?

0xc0170, 2018-06-12:
> That our internal bug tracker has mirrored this issue.

I'll send PR adding this detail to the workflow document in our documentation.

DBS06, 2018-06-13:
@theotherjimmy nice done!

PR #7196 (2018-06-15): Tools: Remove second header when every bl config param is used
### Description

Resolves #7190

The issue is simple: the managed bootloader mode will accept both 
`target.bootloader_img` and `target.restrict_size` at the same time.
When combined with `target.header_format` though, you end up with
two application headers, one before the application and one after.

This PR removes the second one; the after-application header.

### Pull request type

    [x] Fix
    [ ] Refactor
    [ ] New target
    [ ] Feature
    [ ] Breaking change
closed 2018-06-13 by DBS06