ARMmbed/mbed-os #7193: STM32L031K6 requires comprehensive pin function definition

Tony-YI, 2018-06-12:
### Description
Currently I am using the ***STML031K6 MCU*** on a customer PCB. In my previous program, I was using the mbed2 revision 158 and I could define a serial port on pins ***PA_9*** and ***PA_10***. After I updated my mbed library to revision 162 or later, I can no longer define serial port using ***PA_9*** and ***PA_10***. I looked into the source code and found out that ***PA_9*** in ***PinMap_UART_TX[]*** and ***PA_10*** in ***PinMap_UART_RX[]*** of ***PeripeheralPins.c*** were commented. I believe this is the reason why I could not define serial port using PA_9 and PA_10.

I could use the mbed-dev to redefine the ***PeripeheralPins.c*** of course. However, it would be more flexible if the official ***PeripeheralPins.c*** defines all functions of all pins instead of commenting some of the pin functions. Using warnings to inform the users may be more appropriate.

[ ] Question  
[X] Enhancement  
[ ] Bug

jeromecoutant, 2018-06-12:
Hi

Yes, PeripheralPins.c file has been modified according to NUCLEO_L031K6 HW.
And PA_9 and 10 can't be used for UART as they use the same UART2 as default virtual COM.

Note that PinMap_UART_TX are defined as WEAK, so you can also only redefine it in your application.

bcostm, 2018-06-12:
Or you can create a new target for your PCB and change the PeripheralPins.c

Tony-YI, 2018-06-13:
@jeromecoutant 

Appreciate if you could let me know how to redefine the PinMap_UART_TX[] and PinMap_UART_RX[] in the application. I am using the online compiler and mbed2.

I have tried adding the following code into my main.cpp but it did not work:

```
#include "mbed.h"
#include "pinmap.h"
const PinMap PinMap_UART_TX[]{
    {PA_2_ALT0, LPUART_1,STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF6_LPUART1)},
    {PA_9,      UART_2,  STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF4_USART2)},
    {NC, NC, 0}
};
    
const PinMap PinMap_UART_RX[]{
    {PA_3_ALT0,LPUART_1,STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF6_LPUART1)},
    {PA_10,    UART_2,  STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF4_USART2)},
    {NC, NC, 0}
};

RawSerial serial(PA_9, PA_10);

int main()
{
    // do somthing
}
```

Tony-YI, 2018-06-13:
@jeromecoutant 

Appreciate if you could let me know how to redefine the PinMap_UART_TX[] and PinMap_UART_TX[] in the application. I am using the online compiler and mbed2.

I have tried the following code and it did not work:

```
#include "mbed.h"

#include "pinmap.h"
const PinMap PinMap_UART_TX[] = {
    {PA_2_ALT0, LPUART_1,STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF6_LPUART1)},
    {PA_9,      UART_2,  STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF4_USART2)},
    {NC, NC, 0}
};
    
const PinMap PinMap_UART_RX[] = {
    {PA_3_ALT0,LPUART_1,STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF6_LPUART1)},
    {PA_10,    UART_2,  STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF4_USART2)},
    {NC, NC, 0}
};

DigitalOut led(PB_1);
RawSerial lp_uart1(PA_2_ALT0, PA_3_ALT0);
RawSerial usart2(PA_9, PA_10);

int main()
{
    while(1)
    {
        led = !led;
        wait(1);
    }
}
``` 

The compiler returns two warnings:
```
Warning: Variable "PinMap_UART_TX" was declared but never referenced in "main.cpp", Line: 7, Col: 15
Warning: Variable "PinMap_UART_RX" was declared but never referenced in "main.cpp", Line: 13, Col: 15
```

Thanks in advance.

0xc0170, 2018-06-13:
> Note that PinMap_UART_TX are defined as WEAK, so you can also only redefine it in your application.

The code snippet above should work?

@Tony-YI What's mbed 2 version that you use?

Tony-YI, 2018-06-13:
@0xc0170 

mbed2 Revision 166 (Release version 161).

The code above did not work. I suppose it is because I am using the online compiler.

0xc0170, 2018-06-13:
That explains it.

@jeromecoutant This is a bug. To have this available in the mbed 2 , peripheral pins should be compiled as separate object files , see this taken from build_api:

```

        # A number of compiled files need to be copied as objects as opposed to
        # way the linker search for symbols in archives. These are:
        #   - mbed_retarget.o: to make sure that the C standard lib symbols get
        #                 overridden
        #   - mbed_board.o: mbed_die is weak
        #   - mbed_overrides.o: this contains platform overrides of various
        #                       weak SDK functions
        #   - mbed_main.o: this contains main redirection
        separate_names, separate_objects = ['mbed_retarget.o', 'mbed_board.o',
                                            'mbed_overrides.o', 'mbed_main.o', 'mbed_sdk_boot.o'], []

```

bcostm, 2018-06-14:
> peripheral pins should be compiled as separate object files

How can we achieve this ? Simply by adding `PeripheralPins.o ` in this list ?

0xc0170, 2018-07-13:
@theotherjimmy  https://github.com/ARMmbed/mbed-os/pull/7353 - was this already propagated to the compiler ? If not, when it will be?

theotherjimmy, 2018-07-13:
@0xc0170 The online compiled does not run that function.

0xc0170, 2018-07-18:
> @0xc0170 The online compiled does not run that function.

👍  this is part of the release process. This should be now resolved (can be confirmed on master via building mbed library for this target)

closed 2018-07-18 by 0xc0170