ARMmbed/mbed-os #7198: Missing feature in BLE API, Listening to Notifications to BLE Characteristic changes.

yogeshk19, 2018-06-12:
### Description
Although the BLE API provides a mechanism to listen to Characteristic Notifications, using the method below:
```
//subscribing to notifications.
ble.gattClient().onHVX(&BleMaster::OnSensorDataChanged);
```
However this doesn't trigger notifications and requires quite a bit of additional code as mentioned in the BLE issue [#192](https://github.com/ARMmbed/ble/issues/191) . This feature needs to be provided out of box in the mbed BLE API. Also the latest version of mbed-os this work around doesn't work, returns an error BLE_ERROR_INVALID_STATE.

Target: Tested in NRF52_DEV KIT.

### Issue request type
[ ] Question  
[X ] Enhancement  
[X] Bug  

Thanks,
Yogesh

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTPAN-206*

yogeshk19, 2019-03-03:
@Pan , @donatieng , Are there  any updates to this issue in the new BLE Cordio API. Is it more simplified in Cordio API to register for notification for characteristic value updates?

Thanks,
Yogesh

yogeshk19, 2019-06-28:
@pan- Is this being handled differently BLE Cordio?

40Grit, 2019-06-29:
@aglass0fmilk

pan-, 2019-07-01:
@yogeshk19 The underlying stack doesn't matter; discovering characteristics, services and descriptors then writing descriptors continue to be the mandatory step of notification subscription.

AGlass0fMilk, 2019-10-29:
I think this issue should be closed -- it is not a bug. The BLE stack works as it should (in this case). The application (or possibly a framework) is required to do the following to subscribe to notifications/indications from a connected GATT Server:

1.) Discover services on the server -- something like [`ble.gattClient().launchServiceDiscovery()`](
https://github.com/ARMmbed/mbed-os/blob/f1392d6b6bf8b6829aed6fd4b2392b7927cd63bc/features/FEATURE_BLE/ble/GattClient.h#L231-L287)

2.) From the list of discovered services, identify the one you're interested in. Then launch characteristic discovery -- Again, `ble.gattClient().launchServiceDiscovery()` with some different parameters. Steps 1 and 2 may be combined.

3.) Once desired characteristics are found, you must then discover the _descriptors_ for each of the characteristics you want to subscribe to -- see [`DiscoveredCharacteristic::discoverDescriptors()`](https://github.com/ARMmbed/mbed-os/blob/f1392d6b6bf8b6829aed6fd4b2392b7927cd63bc/features/FEATURE_BLE/ble/DiscoveredCharacteristic.h#L351-L376)

4.) Once descriptor discovery is done, you will need to identify which one is the [characteristic's CCCD](https://www.oreilly.com/library/view/getting-started-with/9781491900550/ch04.html#gatt_cccd) based on it's UUID and initiate a write request to the descriptor to set the notification/indication flags as appropriate. Something like `ble.gattClient().write()` with the descriptor's attribute handle ([see `DiscoveredCharacteristicDescriptor::getAttributeHandle()`](https://github.com/ARMmbed/mbed-os/blob/f1392d6b6bf8b6829aed6fd4b2392b7927cd63bc/features/FEATURE_BLE/ble/DiscoveredCharacteristicDescriptor.h#L127-L138)) as the `attributeHandle` parameter. I suggest using [`GATT_OP_WRITE_REQ`](https://github.com/ARMmbed/mbed-os/blob/f1392d6b6bf8b6829aed6fd4b2392b7927cd63bc/features/FEATURE_BLE/ble/GattClient.h#L147-L154) as the operation type so you can get a callback and check if the write succeeded/failed (and retry as appropriate, or elevate the link security/pair with the GattServer).

5.) Once the CCCD is written to enable notifications/indications, you will then receive callbacks in your `onHVX` handler.

yogeshk19, 2019-10-31:
@AGlass0fMilk I am exactly doing what you have described and currently seems to work.

yogeshk19, 2020-05-21:
Closing this issue, as this doesn't need a fix.

closed 2020-05-21 by yogeshk19