ARMmbed/mbed-os #7200: How do I remove the bootloader from my build?

RobMeades, 2018-06-12:
### Description
What is the proper way to remove the bootloader from a target?  I'm using `UBLOX_EVK_NINA_B1` and have run out of flash so I need to remove the bootloader, which is taking up 128 kbytes of my 512 kbytes of flash.

I've tried adding `"target.bootloader_supported": false` to my `mbed_app.json` file but that causes compilation to stop with the error `Bootloader not supported on this target`. What is the right way to get all my flash back please?

### Issue request type

[x] Question  
[ ] Enhancement  
[ ] Bug

RobMeades, 2018-06-12:
Hmmm, I see that 140 kbytes is actually reserved for the Nordic SoftDevice, which I guess is what's giving me a BLE interface.  I guess there's no way of reducing that and still having BLE...?

0xc0170, 2018-06-13:
Regarding the bootloader. The error is because there are regions defined in the config - so your target has no bootloader support (set to false) but application still defines some attributes for the bootloader. What application are you building ?

> Hmmm, I see that 140 kbytes is actually reserved for the Nordic SoftDevice, which I guess is what's giving me a BLE interface. I guess there's no way of reducing that and still having BLE...?

@ARMmbed/mbed-os-pan

RobMeades, 2018-06-13:
Yes, I've now found the `mbed_lib.json` down in the Nordic target directories, hadn't realised Nordic were using the bootloader mechanism to load the BLE protocol code.  Having read their documentation, it seems as though I'll just have to live with it.  I'm building my own application, which has grown quite large, I'll just have to live with the fact that I can't do a debug build (a release build fits).  Or I'll switch to ARM rather than GCC_ARM.  Anyway, thanks for your help, I have a way forward now.

closed 2018-06-13 by RobMeades