ARMmbed/mbed-os #7203: ST NUCLEO F411RE flashing doesn't work twice in a row on Linux

yogpan01, 2018-06-13:
### Description
 - Type: Bug
 - Priority: Critical

#### Info
On Linux environment (Using Ubuntu 16.04), when flashing binary twice on ST NUCLEO F411RE, it gives error "no space on disk" and only way to recover is unplug and plug USB cable again.
This prevents this board to be deployed in our Continuous Integration.
    - Target: ST NUCLEO F411RE 
    - Toolchain; GCC_ARM , ARMCC
    - DAPLINK: version 0221

[ ] Question  
[ ] Enhancement  
[X] Bug

yogpan01, 2018-06-13:
@MarceloSalazar can you please take this one?

jeromecoutant, 2018-06-13:
Hi

Did you check you have the latest FW : http://www.st.com/en/development-tools/stsw-link007.html

@schstm

yogpan01, 2018-06-13:
@VeliMattiLahtela can you try this firmware on NUCLEO F411RE board?

jeromecoutant, 2018-06-20:
Hi

Any update ? Di you try V2J31M21 FW version ?

Thx

screamerbg, 2018-09-20:
@yogpan01 Bump. Have you tried the latest STLink firmware?

yogpan01, 2018-09-20:
@screamerbg This seems to work now as we haven't seen issue in our CI. Happy to close this one now!

closed 2018-09-20 by yogpan01