ARMmbed/mbed-os #7204: TCPsocket and UDPsocket Speed

flipflop22, 2018-06-13:
### Description

My setup:
-EVK-ODIN-W2
-GCC ARM Embedded Toolchain
-MBED-CLI
I want to send a data with 20 bytes size with the speed of 1 MHz. I have always fails to achieve this speed with TCP or UDP. The fastest I can get is with UDP 1kHz (in average). Is there a way to optimize this speed???

here to reproduce: 
I create a UDPsocket
`UDPSocket udpsocket`
the run a while loop to send
`do{
udpsocket.sendto("IP Address", port, buffer, strlen(buffer)} while(true);
`
### Issue request type

[X] Question  
[ ] Enhancement  
[ ] Bug

ohagendorf, 2018-06-13:
When you calculate the transfered bits, you would see, that your transfer speed is physically not possible:
- user data 160 bits
- udp frame 64 bits
- ip frame min. 160 bits
- ethernet frame 336 bits (with inter frame gaps, ...)
sum: 720 bits

With a 100Mbit ethernet you can transfer theoretically your data with 138kHz with a 100% ethernet utilisation. But these 100% are praktically not possible. So the possible transfer speed is much lesser, maybe round about 60..70 kHz maximum. This calculation is only the view at the bits per second. The software side needs some additional time ...

flipflop22, 2018-06-14:
> - udp frame 64 bits

UDP frame 64 bits here is the payload? Or also include overheads?
If it is the payload: Does this means that the function below expecting buffer of 64 bits? Means that if I pass more than 64 bits the next data will be send at the next UDP frame?
`udpsocket.sendto("IP Address", port, buffer, strlen(buffer))`

> So the possible transfer speed is much lesser, maybe round about 60..70 kHz maximum.

how big is the buffer I should send to achieve this speed?

flipflop22, 2018-06-14:
How about TCP?
What is the typical maximum speed and the (char) buffer size so that the TCP connection keeps remain (It wont disconnected)?

The data rate I need is 500 kHz with size of 160 bits, and it should be constant rate...

ohagendorf, 2018-06-14:
With the 60..70 kHz I mean the maximum on the physical side. Using low level programmig without any HAL, maybe you can come into this speed range. But not with the common high level programming.

TCP needs even more bits for its header. You can find die frame format discriptions e.g. in wikipedia.

When you use WIFI, I think you have a SPI connection to the WIFI module - the SPI speed may be the limiting factor in your configuration.

SeppoTakalo, 2018-06-14:
@flipflop22 
`socket.sendto("DNS address", port, data, size);` means one DNS query per packet send.

Try at least querying the address first and then use:
`socket.sendto(SocketAddress, data, size); `

And as @ohagendorf was already commenting: 1 Mhz packet rate is utopistic

kjbracey-arm, 2018-06-14:
> 1 Mhz packet rate is utopistic

Are we sure this is what is actually being asked for? Could @flipflop22 actually mean 1Mbit/s?

Anyway, on the other hand 1000 packets/s seems a bit too low. I'd be a bit nervous that something is inadvertently delaying or pacing by millisecond timer ticks. I'd like to see a number a bit closer to 10000 packets/s.

I suspect the overhead per packet is somewhat constant, and you would get much higher total throughput by sending much bigger packets.

> How about TCP? What is the typical maximum speed and the (char) buffer size so that the TCP connection keeps remain (It wont disconnected)?

TCP is flow controlled - you just keep writing, and by default the write will block until the data has gone into the buffer. If the link is slow, your program will slow down. It won't disconnect as a result of you writing.  It will automatically combine your data into larger packets to improve throughput.

If actually sending data via the internet, you must use TCP or something with equivalent congestion control that slows down when the network is busy. You can't just fire data across the internet at some constant speed - that's a sure way to get your connection blocked as being the source of an attack.

Possibly relevant: https://blog.cloudflare.com/how-to-receive-a-million-packets/

ohagendorf, 2018-06-14:
I looked into the documentation and the used board has a WIFI interface with a max data rate of 20Mbit/s.
That means theoretically 27kHz data rate only considering the physical point of view.

flipflop22, 2018-06-15:
> When you use WIFI, I think you have a SPI connection to the WIFI module - the SPI speed may be the limiting factor in your configuration.

@kjbracey-arm Considering SPI baud rate 10 Mbps. If I set it to be 16 bits, I can send total of 20 bytes within at least 20 microseconds which is (50 kHz). Ok now my expectation drop to 50 kHz.

> I looked into the documentation and the used board has a WIFI interface with a max data rate of 20Mbit/s

@ohagendorf To which docs you refer to? The 802.11ag speed is supposed to be 54Mbps.

> ethernet frame 336 bits

@ohagendorf the WIFI header is 192 bits rite? CMIIW

NOTE: ANYWAY I think I find a rootcause, it happens because I use MBED OS 3 Minar for my receiver, looks like Minar is so slow. I can't use MBED OS 5 because the ublox odin-w2 target's WiFi stacks doesn't support Access Point curently.

kjbracey-arm, 2018-06-15:
> Considering SPI baud rate 10 Mbps. If I set it to be 16 bits, I can send total of 20 bytes within at least 20 microseconds which is (50 kHz). Ok now my expectation drop to 50 kHz. However right now it is only 1 kHz.

Your packets have 20 byte UDP payload, 28 bytes of UDP/IP header, and 14 bytes of Ethernet header, assuming you're sending 802.3 frames to a MAC on the SPI. (That's not allowing for any command overhead to the device)

So 62 total bytes per packet = 496 bit, which if SPI is 10Mbit/s gives you theoretical limit of 20 kpackets/second.

With small transfers like that, you may not be invoking DMA for the SPI, and the CPU may actually be fully occupied during the transfer, depending on how the Odin/STM SPI HAL driver works. Which means it will occasionally need to stop SPI transfers to do some processing.

I know the STM SPI driver has had performance problems in the past - small transfers are far from ideal, and STM has a significant per-packet overhead, which has bitten us with 802.15.4. There were improvements here, but wouldn't be in mbed OS 3: https://github.com/ARMmbed/mbed-os/issues/4246

Unless you have a very specific reason, you really should be trying to avoid sending such small packets. Make your packets 50 times bigger and you should get a huge bytes-per-second throughput boost.

kjbracey-arm, 2018-06-15:
Actually, it's worse than I thought - until fairly recently mbed didn't have a block-transfer API at all for SPI HAL, so regardless of what you do, you would be sending 1 word at a time over SPI. The STM boards were achieving something like 12 us per SPI transfer like that, looking at #4246. If in 16-bit SPI mode, as you suggested, that would limit you to 1.3 Mbit/s on the SPI bus, giving a theoretical 2,700 packets/second.

Given that, 1000 packets/second actual now sounds about right, for a system with old STM SPI HAL. Current mbed OS 5 should do a fair bit better.

kjbracey-arm, 2018-06-15:
@ARMmbed/team-ublox  - want to comment on this?

> I can't use MBED OS 5 because the ublox odin-w2 target's WiFi stacks doesn't support Access Point curently.

flipflop22, 2018-06-19:
> 12 us per SPI transfer

Wew mine is 24 us per SPI transfer (16 bit, 20MHz clock)

flipflop22, 2018-06-19:
> ANYWAY I think I find a rootcause, it happens because I use MBED OS 3 Minar for my receiver, looks like Minar is so slow. I can't use MBED OS 5 because the ublox odin-w2 target's WiFi stacks doesn't support Access Point curently.

I was wrong, the problem is on the transmitter that use MBED OS 5 UDPSocket Stack. Looks like the UDP HAL is only able to send in 1 kHz.... This is very slow

> socket.sendto(SocketAddress, data, size);

@SeppoTakalo this does not improve UDP speed

Here is my loop.
`do{`
`       send_response = udpsocket.sendto(remote_address, buffer, strlen(buffer));`
`	   spi.write(0xAAAA);`
`	   wait_us(10);`
`    } while (send_response >=0);`

Then I monitor the spi data with oscilloscope, it exactly 1 kHz.

ohagendorf, 2018-06-19:
When you look into the documentation of your board, you would see that the wifi speed is limited to 20 Mbit/s nevertheless the speed of the specification of the wifi radio.
Now it depends on the wifi chip, which data has to be transfered. Maybe only the user data, that means just your 20 bytes per packet. Or maybe the complete raw frame which is much more.

SeppoTakalo, 2018-06-19:
I would like to know what is the application where you expect 1 kHz send rate to be working?

It is true, that our embedded devices are not optimized for speed, but I still feel its unrealistic to even assume this is working through the Internet.

I don't think there exist any protocols that spam with this are without any rate limitting and acknowledgements from receiver end.

kjbracey-arm, 2018-06-19:
> Wew mine is 2 us per SPI transfer (16 bit, 20MHz clock)

Until recently, this was not remotely achievable because the SPI HAL was transferring 1 SPI word at a time (using `spi.write(int)`), and it took 10us for the STM HAL to set up for that 1 word. You can see oscilloscope traces showing this in https://github.com/ARMmbed/mbed-os/issues/4246 .

With current mbed OS 5, it will be managing proper burst speed, but you won't be getting that in mbed OS 3.

flipflop22, 2018-06-20:
>I would like to know what is the application where you expect 1 kHz send rate to be working?

for power electronics control. 

> Until recently, this was not remotely achievable because the SPI HAL was transferring 1 SPI word at a time (using spi.write(int)), and it took 10us for the STM HAL to set up for that 1 word. You can see oscilloscope traces showing this in #4246 .

I use ODIN-W2. In MBED OS 3 for ODIN W2 I found it is 2us/16 bits data on Oscilloscope. Yet it is still not enough for my requirements.

SeppoTakalo, 2018-06-20:
When your applications sample rate is starting to be in the same range as audio signals, you should start studying how audio is send over internet.
https://en.wikipedia.org/wiki/Audio_over_IP

Audio codecs easily add delay of hundreds of milliseconds and buffering is always used.

Buffering, re-routing and variable delays are common in Internet. If constant bitrate, constant delay network is required, in not sure IP is the best medium for your use.

For example, on audio applications where low latency and constant delay is required, IP is not used, but dedicated Ethernet network may be used instead: https://en.wikipedia.org/wiki/Audio_over_Ethernet
Various applications exist, but some operate directly on Ethernet frame, skipping the IP frame, and IP stack.

Also, note that none of those standards operate on WiFi. Wifi cannot guarantee constant delay as there are interference, collisions and retransmissions on the path.

SeppoTakalo, 2018-06-20:
Also, easiest way to get buffering would be to use TCP stream.

Once you are connected, you can easily do
```
while(1) {
    len = measure(buffer);
    socket.send(buffer, len);
}
```
and leave the TCP/IP stack pack as much data into packets as it likes. This way you achieve best guaranteed performance. Sending faster would not help, if the receiver is not as fast. In TCP, the receiver is sending acknowledgements allowing stacks to find the optimal speed of the stream.

But, when congestion occurs and stack runs out of buffers, the `socket.send()` starts blocking.

flipflop22, 2018-06-20:
What is the recommended buffer size that I can send.
In UDP try to send data over 500 data buffer (char). However in the rec_from function I only get 270 data.
Can this size of buffer be set?

SeppoTakalo, 2018-06-20:
In UDP, each sendto() call sends out one packet.
So your packet must fit into one UDP payload.
If its Ethernet, then 1500 is the MTU. IP header is 20 bytes, UDP header is 8 bytes. Then most probably 1472 bytes can go through.
But when you start to send out packets through internet, RFCs on IPv4 only guarantees that MTU 576 can be assumed to pass on all devices. Therefore safe payload sizes should be less than 550.

For incoming, each `recvfrom()` call returns a payload of one packet. Therefore the size is exactly as much there was data in the packet. And for packet sizes, same apply as sending. But we have seen devices that cannot get full Ethernet frames.
So 270 bytes you get from `recvfrom()` call is not the maximum, it was the payload sent from remote end.

SeppoTakalo, 2018-06-21:
> The remote end MTU I set to be 500. However I still get 270 bytes at the receiver.

I don't understand this. MTU is a option of the network interface. At least in Mbed OS, you cannot set that. It other operating systems you most probably can.

But if the UDP works, as it is supposed to, you get what you send. So how much data did you send with single `socket.sendto()` call? And was it Mbed OS 5 device that was sending?

> Is 6LoWPAN performance better than WiFi for 20 bytes payload?
No. 6LoWPAN works over IEEE 802.15.4 networks which are low power and lossy. Usually something like 250 kbit/s on a shared medium.

[image]

flipflop22, 2018-06-21:
> I don't understand this. MTU is a option of the network interface. At least in Mbed OS, you cannot set that. It other operating systems you most probably can.

Sorry I mean is the buffer to send and to receive

> And was it Mbed OS 5 device that was sending?

MBED OS 5 is the one to transmit, MBED OS 3 is the one to receive.

SeppoTakalo, 2018-06-21:
I cannot guarantee whether there are bugs in Mbed OS 3 socket APIs, I can guarantee only Mbed OS 5, which I'm maintaining and testing.

sendto(500) should lead to recvfrom() returning packet sized 500 bytes. If something else, it is a bug.

flipflop22, 2018-06-22:
> and leave the TCP/IP stack pack as much data into packets as it likes. 

I have been trying to use TCP. After it is connected, then it sends 550 bytes of data. However the second time it sends data there is always been error -3005. I realize that the TCP connection was disconnected.

Is there a way to check the status of TCP connection in mbed OS 5.9?

flipflop22, 2018-06-22:
> and leave the TCP/IP stack pack as much data into packets as it likes. 

Also, it can be a problem since my data package is only 20 Bytes. If it stacks the data as it likes, difficult to identify particular data. For example, I try to send a stream of 540 bytes (27 packages) . It will receive the first 500 then 36 then 4 bytes. I only send one stream. if I send multiple stream it will be difficult to identify as some packs of a stream stacked together with the next stream.

asifrizwanubx, 2018-06-25:
> @ARMmbed/team-ublox - want to comment on this?

    I can't use MBED OS 5 because the ublox odin-w2 target's WiFi stacks doesn't support Access Point curently.

WiFi AP feature will be target in the next ODIN-W2 Mbed OS 5 releases.

flipflop22, 2018-06-26:
in TCP the reliable payload is 536 bytes.
I implement with sigio and the speed is slow at around 200ms between successful sending or in otherwords the throughput around 21.4 Kbps.
@asifrizwanubx the datasheet of ublox shows the WiFi throughput of 20 Mbps. Can you tell how to get this speed?

SeppoTakalo, 2018-06-26:
In UDP the reliable payload is 536, unless Path MTU discovery is used, or Ethernet frame can be assumed all the way.

In TCP there is no such thing as the TCP stack is taking care of segmenting the transfer.

flipflop22, 2018-06-26:
> In TCP there is no such thing as the TCP stack is taking care of segmenting the transfer.

Rite. What I mean is the maximum size of the segment I found to be 536 (WiFi Station: MBED OS 5, WiFi AP: MBED OS 3). So if I send more than 536 TCP stack will segment the data.

flipflop22, 2018-06-27:
> In UDP the reliable payload is 536, unless Path MTU discovery is used, or Ethernet frame can be assumed all the way.

I see at the configuration there is this parameter
```
Name: lwip.pbuf-pool-bufsize
    Description: Size of pbufs in pool. If set to null, lwIP will base the size on the TCP MSS, which is 536 unless overridden by the target
    Defined by: library:lwip
    No value set
```

Is this the one to set the buffer size?

flipflop22, 2018-06-28:
@SeppoTakalo If I increase the PBUF_SIZE_BUFSIZE of LwIP, do I have to increase the MEM_SIZE?

flipflop22, 2018-07-02:
I saw the TCP_MSS for Realtek and RZ_A1XX targets are 1460. Can I confirm that this is working?

flipflop22, 2018-07-25:
@asifrizwanubx Any update about Access Point at Mbed OS 5.
The minar scheduler in Mbed OS 3 seems not working that well with my application.

asifrizwanubx, 2018-07-31:
@flipflop22 

https://github.com/ARMmbed/mbed-os/pull/7579/

flipflop22, 2018-08-20:
Is it ready to use?

closed 2018-10-02 by adbridge