ARMmbed/mbed-os #7213: PPPCellularInterface::connect calls nwk_registration with a hard coded value

cedrickkukela-cd, 2018-06-13:
### Description

While attempting to bring on a new target enabled target mbed, I noticed that in PPPCellularInterface::connect the value passed to nwk_registration is hard coded. Since this value is used to switch between at+creg and at+cgreg, I effectively locks you into using at+cgreg for registration. This may not be a problem on cat 1 modems but on cat m modems such a Ublox r410, at+cgreg will always return a 0,4. Only creg and cereg will return the proper value. I was wondering if there was a fix for this issue in place or should i extend the interface to support nwk_type. 

### Issue request type

[ ] Question  
[x] Enhancement  
[ ] Bug

AriParkkila, 2018-06-14:
Please use [new cellular](https://github.com/ARMmbed/mbed-os/tree/master/features/cellular) because PPPCellularInterface is to be deprecated.

You can then define modem to be GSM/LTE in `framework/targets` folder in `has_registration()` function as `return (reg_type == C_REG || reg_type == C_EREG);`

cedrickkukela-cd, 2018-06-18:
Thanks one final question. When i attempt to run the cell test, I receive the following error. 

Build successes:
  * MTS_DRAGONFLY_L471QG::GCC_ARM::MBED-BUILD
mbedgt: greentea test automation tool ver. 1.2.5
mbedgt: test specification file 'C:\Users\Cedrick\cellTest\BUILD\tests\MTS_DRAGONFLY_L471QG\GCC_ARM\test_spec.json' (specified with --test-spec option)
mbedgt: using 'C:\Users\Cedrick\cellTest\BUILD\tests\MTS_DRAGONFLY_L471QG\GCC_ARM\test_spec.json' from current directory!
mbedgt: detecting connected mbed-enabled devices...
mbedgt: detected 1 device
	+----------------------+-------------------------+-------------+-------------+--------------------------+
	| platform_name        | platform_name_unique    | serial_port | mount_point | target_id                |
	+----------------------+-------------------------+-------------+-------------+--------------------------+
	| MTS_DRAGONFLY_L471QG | MTS_DRAGONFLY_L471QG[0] | COM70       | E:          | 031002211623607E2967F253 |
	+----------------------+-------------------------+-------------+-------------+--------------------------+
mbedgt: processing target 'MTS_DRAGONFLY_L471QG' toolchain 'GCC_ARM' compatible platforms... (note: switch set to --parallel 1)
	+----------------------+-------------------------+-------------+-------------+--------------------------+
	| platform_name        | platform_name_unique    | serial_port | mount_point | target_id                |
	+----------------------+-------------------------+-------------+-------------+--------------------------+
	| MTS_DRAGONFLY_L471QG | MTS_DRAGONFLY_L471QG[0] | COM70:9600  | E:          | 031002211623607E2967F253 |
	+----------------------+-------------------------+-------------+-------------+--------------------------+
mbedgt: test case filter (specified with -n option)
mbedgt: running 0 tests for platform 'MTS_DRAGONFLY_L471QG' and toolchain 'GCC_ARM'
	use 1 instance of execution threads for testing
mbedgt: all tests finished!
mbedgt: shuffle seed: 0.0358614844
mbedgt: no platform/target matching tests were found!
mbedgt: completed in 2.64 sec
mbedgt: exited with code -10
[mbed] Working path "C:\Users\Cedrick\cellTest" (program)
[mbed] Exec "C:\Python27\python.exe -u C:\Users\Cedrick\cellTest\mbed-os\tools\test.py -t GCC_ARM -m MTS_DRAGONFLY_L471QG --source C:\Users\Cedrick\cellTest --build C:\Users\Cedrick\cellTest\BUILD\tests\MTS_DRAGONFLY_L471QG\GCC_ARM --test-spec C:\Users\Cedrick\cellTest\BUILD\tests\MTS_DRAGONFLY_L471QG\GCC_ARM\test_spec.json -n features-cellular-tests-* -v" in C:\Users\Cedrick\cellTest
[mbed] Exec "mbedgt --test-spec C:\Users\Cedrick\cellTest\BUILD\tests\MTS_DRAGONFLY_L471QG\GCC_ARM\test_spec.json -n features-cellular-tests-* -V" in C:\Users\Cedrick\cellTest

The rest to the test suite runs just fine. Do cellular targets need to be define in any other locations?

maclobdell, 2018-06-20:
@cedrickkukela-cd  - can you confirm the test name you are trying to run?

There is a blog post about running network socket tests.  Be sure to pass in a test config file which specifies the interface you are using?  Does this help?  https://os.mbed.com/blog/entry/Verify-a-Network-Driver-Greentea/

For details, see Testing section of https://os.mbed.com/docs/v5.8/reference/contributing-connectivity.html

AriParkkila, 2018-07-27:
@cedrickkukela-cd you need to define [CELLULAR_DEVICE](https://github.com/ARMmbed/mbed-os/blob/master/features/cellular/framework/common/CellularTargets.h) to select correct cellular module drivers.

cedrickkukela-cd, 2018-07-27:
Already done in current pull request. 

The only remaining issue is the use of atd vs vs at+cgdata in at_cellularnetwork. There should be a way to  select with command is used for initiating a ppp connection as some modules may not support at+cgdata( ublox r410).

AriParkkila, 2018-08-02:
Please see my comment about `at+cgdata` in https://github.com/ARMmbed/mbed-os/pull/7304#discussion_r206033500

cedrickkukela-cd, 2018-08-03:
sorry about the delay 
I will push something up tomorrow

closed 2018-10-02 by adbridge