ARMmbed/mbed-os #7223: mbed Client example : Current time Timestamp issue

prvthkr, 2018-06-15:
### Description

**Target**
K64F

**Toolchain:**
GCC_ARM

**Toolchain version:**
6.3.1 20170215 (release) [ARM/embedded-6-branch revision 245512]

**mbed-cli version:** 
1.5.1

**Mbed OS:**
Latest master package

### Issue request type

[X] Question  
[ ] Enhancement  
[ ] Bug  

**Question:**
I want to add time resource which will display current time in [mbed connector](https://connector.mbed.com/).
Below code ([mbed-os-env](https://github.com/ARMmbed/mbed-os/blob/master/features/unsupported/tests/mbed/env/test_env.cpp#L83)) is displaying the build time.
```
unsigned int testenv_randseed()
{
    unsigned int seed = 0;
#ifdef MBED_BUILD_TIMESTAMP
    long long_seed = static_cast<long>(MBED_BUILD_TIMESTAMP);
    seed = long_seed & 0xFFFFFFFF;
#endif /* MBED_BUILD_TIMESTAMP */
    return seed;
}
```
How to read server time and display on the `/endpoints/{endpoint-name}/{resource-path}`?

yogpan01, 2018-06-19:
Hi,
You need to pass this value through some M2MResource and read that value on the mbed connector.

Something like this would work 
```
M2MObject *time_object = M2MInterfaceFactory::create_object("20000");
M2MObjectInstance* time_inst = time_object->create_object_instance();
M2MResource* time_res = time_inst->create_dynamic_resource("1", "Timestamp",
            M2MResourceInstance::STRING, false);
        // read 
        time_res->set_operation(M2MBase::GET_ALLOWED);
        // set timestamp
        time_res->set_value((const uint8_t*)MBED_BUILD_TIMESTAMP, sizeof(MBED_BUILD_TIMESTAMP));
```

yogpan01, 2018-06-28:
Hi,
Hope this has resolved your issue, if not please create a new issue with steps. For now, we are closing this issue as resolved.

Regards,
Yogesh

deepikabhavnani, 2018-07-18:
> For now, we are closing this issue as resolved.

Closing now

closed 2018-07-18 by deepikabhavnani