ARMmbed/mbed-os #7227: STM Nucleo ethernet driver hardware checksum insertion corrupts fragmented UDP packets

mikaleppanen, 2018-06-15:
### Description

STM Nucleo Ethernet driver hardware checksum insertion corrupts fragmented UDP packets that are sent by the driver to ethernet.

This can be repeated by running greentea LWIP UDP echo test with Nucleo boards and by setting the MTU size smaller than tested UDP packet size, e.g. to 800 bytes, which causes LWIP to fragment packets.

https://github.com/ARMmbed/mbed-os/blob/master/features/netsocket/emac-drivers/TARGET_STM_EMAC/stm32xx_emac.cpp

```
uint32_t STM32_EMAC::get_mtu_size() const
{
    return 800; //STM_ETH_MTU_SIZE;
}
```
Then by running UDP tests:

> mbed test --compile --run -m NUCLEO_F767ZI -t GCC_ARM -v -n tests-netsocket-udp

If checksum mode is set from hardware to software on the "stm32xx_emac.cpp" fragmented UDP packets are not corrupted:
```
    EthHandle.Init.ChecksumMode = ETH_CHECKSUM_BY_HARDWARE;
```
to:
```
    EthHandle.Init.ChecksumMode = ETH_CHECKSUM_BY_SOFTWARE;
```

Here are Wireshark logs from passing/failing cases:

[ethernet_mtu800_checksum_pass_and_fail.zip](https://github.com/ARMmbed/mbed-os/files/2105792/ethernet_mtu800_checksum_pass_and_fail.zip)

On failing log file, message number 37 contains 0x3B 0x90 in place of passing case 0x57 0x54
and wireshark indicates that UDP checksum is invalid.

ethernet_mtu800_hw_checksum_fail.pcap      message 37:

> 02f0   39 4f 4f 5a 32 49 3b 42 56 51 43 32 55 4f 4f 3d
0300   44 44 53 4d 47 3f 3b 90 56 4a 4d 3c 3e 42 41 3b  <----- 3b 90
0310   4d 4f 52 30 4b 33 3e 31 3f 48 5a 58 3d 56 46 54

ethernet_mtu800_no_hw_checksum_pass.pcap message 37:

> 02e0   34 30 50 53 35 59 45 4c 3e 43 58 33 35 34 53 50
02f0   39 4f 4f 5a 32 49 3b 42 56 51 43 32 55 4f 4f 3d
0300   44 44 53 4d 47 3f 57 54 56 4a 4d 3c 3e 42 41 3b  <------ 57 54
0310   4d 4f 52 30 4b 33 3e 31 3f 48 5a 58 3d 56 46 54

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

0xc0170, 2018-06-15:
@ARMmbed/team-st-mcd Please review

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTPART-5528*

jeromecoutant, 2019-01-29:
Please close this issue
Thx

Wufus, 2022-02-03:
> @jeromecoutant Did #7229 fix the issue?

Yes, but "it is not bug, it is feature".
In fact, you can always use hardware_mode when you're not using IP packet fragmentation.