ARMmbed/mbed-os #7235: DISCO-F413ZH - Idle thread stack overflow in debug build w/ stats on Mbed OS 5.9

janjongboom, 2018-06-18:
### Description

Mbed OS 5.9.0 (rev 03196b244ee00fc748cd4dbfa83b26f3fc0f376b) on the DISCO-F413ZH, running blinky. I've defined `MBED_TICKLESS`, `MBED_CPU_STATS_ENABLED` and `MBED_HEAP_STATS_ENABLED`. Build this with debug profile and run. After ~10 seconds a stack overflow triggers on the idle thread:

```
++ MbedOS Error Info ++
Error Status: 0x80020125 Code: 293 Module: 2
Error Message: CMSIS-RTOS error: Stack overflow
Location: 0x8002559
Error Value: 0x1
Current Thread: Id: 0x20001180 Entry: 0x8003e69 StackSize: 0x200 StackMem: 0x20001210 SP: 0x2004fed8 
-- MbedOS Error Info --
```

(verified with nm that this is the idle thread)

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

janjongboom, 2018-06-18:
Disabled the stats, and it still overflows. But now a bit later (~15 seconds).

0xc0170, 2018-06-18:
@ARMmbed/team-st-mcd Please review

LMESTM, 2018-06-18:
@0xc0170 @janjongboom 
reproduced as well on DISCO_L475VG_IOT01A. 
Is this ST specific ?

LMESTM, 2018-06-18:
@janjongboom 
I've posted #7241 to hopefully solve the issue. This fixed it for DISCO_L475VG_IOT01A
Would you check with your F413 DISCO board as well ?

LMESTM, 2018-06-19:
Just noticed a similar issue reported on another target: #7222

janjongboom, 2018-06-21:
@cmonr This cannot be closed. The issue is still persistent with mbed-os-example-wifi, see the PR.

LMESTM, 2018-06-21:
@cmonr @janjongboom @0xc0170 please feel free to re-open. But as far as I could debug, this is not anymore in ST driver, and this is also reproduced on other devices (#7222), so please consider assigning someone from mbed-os core to have a look

0xc0170, 2018-06-21:
> @cmonr This cannot be closed. The issue is still persistent with mbed-os-example-wifi, see the PR.

Github closed it automatically, reopening this

adustm, 2018-07-10:
Hello,
Any news on this ?

deepikabhavnani, 2018-07-30:
@janjongboom 
I tried reproducing this issue with Mbed OS 5.9.0/master on the DISCO-F413ZH with MBED_TICKLESS, MBED_CPU_STATS_ENABLED and MBED_HEAP_STATS_ENABLED flags and did not get stack overflow with blinky.

Please help me reproduce the issue, do I need to change anything else?

janjongboom, 2018-08-02:
@deepakvenugopal See my comments earlier. Please try this with a large application like [mbed-os-example-wifi](https://github.com/armmbed/mbed-os-example-wifi).

deepikabhavnani, 2018-08-02:
mbed-os-example-wifi - with added macros in json file + no code change  + mbed-os-5.9.4 runs without any issue.
`
"macros": ["MBED_THREAD_STATS_ENABLED", "MBED_CPU_STATS_ENABLED", "MBED_HEAP_STATS_ENABLED", "MBED_TICKLESS"],
`

Do I need to add stats as well to code?

deepikabhavnani, 2018-08-23:
Confirmed in other thread https://github.com/ARMmbed/mbed-os/issues/7222, issue does not exists

PR #7241 (2018-06-20): STM32: Reduce HAL_deepsleep stack usage
### Description

There are cases where a call hal_deepsleep would overflow the idle task
stack, especially in developper or debug profile.

In order to avoid this case, we split ForceClockOutofDeepSleep
into two separate functions the two structure RCC_ClkInitStruct and
RCC_OscInitStruct are not allocated at the same time.

Fixes #7235 

### Pull request type
    [x] Fix
    [ ] Refactor
    [ ] New target
    [ ] Feature
    [ ] Breaking change
closed 2018-08-23 by deepikabhavnani