ARMmbed/mbed-os #7236: mbed-os documentation: expand all sections + download in pdf

bcostm, 2018-06-18:
### Description

The mbed-os documentation here https://os.mbed.com/docs/latest/introduction/index.html is very nice but it is not easy to read it. It would be nice to have these 2 features:

- be able to expand all the sections in order to see all the title at once
- be able to donwload the entire documentation in a pdf file

### Issue request type

[ ] Question  
[X] Enhancement  
[ ] Bug

0xc0170, 2018-06-18:
@AnotherButler Shall this be logged to https://github.com/ARMmbed/mbed-os-5-docs/issues ?

AnotherButler, 2018-06-18:
@0xc0170 Yes, it should. Should I create a new issue there?

0xc0170, 2018-06-19:
Yes please do and close this one (or we do as soon as the new one is created)

bcostm, 2018-06-19:
OK thanks, I will post in the correct place next time...

0xc0170, 2018-06-21:
Moved to https://github.com/ARMmbed/mbed-os-5-docs/issues/604.

I'll close this one.

closed 2018-06-21 by 0xc0170