ARMmbed/mbed-os #7246: Ublox ODIN W2 does not provide WiFiInterface::get_default_instance()

SeppoTakalo, 2018-06-18:
### Description
In Mbed OS 5.9, targets are able to provide default network interface of each type.

Ublox Odin W2 however, does not provide `WiFiInterface *WiFiInterface::get_default_instance()` making it impossible to select default network interface to be a WiFi.

### Issue request type
[ ] Question  
[ ] Enhancement  
[X] Bug

0xc0170, 2018-06-19:
Please review

@ARMmbed/team-ublox 
@ARMmbed/team-realtek

screamerbg, 2018-09-26:
With Mbed OS 5.10.0 out, can we close this @SeppoTakalo @0xc0170 ?

SeppoTakalo, 2018-09-26:
Yes. Fixed in #7259

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTPART-5622*

AdamZhang0124, 2019-01-09:
Hi, since this issue is related to our target, I would like to ask if this issue can be closed. We are trying to minimize the open issues that related to our target. 
Thanks.

SeppoTakalo, 2019-01-09:
This has already been fixed in Mbed OS 5.9.2.

Can be closed.

closed 2019-01-09 by SeppoTakalo

AdamZhang0124, 2019-01-09:
Noted with thanks.