ARMmbed/mbed-os #7253: IPV6 compiler warnings

teetak01, 2018-06-19:
### Description
Mbed OS 5.9.1
IAR ANSI C/C++ Compiler V7.80.1.28/LNX for ARM

```
08:52:04 [Warning] ipv6.c@115,10: [Pe546]: transfer of control bypasses initialization of:
08:52:04 [Warning] ipv6.c@121,18: [Pe546]: transfer of control bypasses initialization of:
08:52:04 [Warning] ipv6.c@153,18: [Pe546]: transfer of control bypasses initialization of:
08:52:04 [Warning] ipv6.c@168,14: [Pe546]: transfer of control bypasses initialization of:
08:52:04 [Warning] ipv6.c@178,14: [Pe546]: transfer of control bypasses initialization of:
08:52:04 [Warning] ipv6.c@785,10: [Pe546]: transfer of control bypasses initialization of:
08:52:04 [Warning] ipv6.c@1056,10: [Pe546]: transfer of control bypasses initialization of:
08:52:04 [Warning] ipv6.c@1061,10: [Pe546]: transfer of control bypasses initialization of:
08:52:04 [Warning] ipv6.c@1066,10: [Pe546]: transfer of control bypasses initialization of:
08:52:04 [Warning] ipv6.c@1071,10: [Pe546]: transfer of control bypasses initialization of:
08:52:04 [Warning] ipv6.c@1081,10: [Pe546]: transfer of control bypasses initialization of:
08:52:04 [Warning] ipv6.c@1094,10: [Pe546]: transfer of control bypasses initialization of:
08:52:04 [Warning] ipv6.c@1176,10: [Pe546]: transfer of control bypasses initialization of:
08:52:04 [Warning] ipv6.c@1232,14: [Pe546]: transfer of control bypasses initialization of:
08:52:04 [Warning] ipv6.c@1238,14: [Pe546]: transfer of control bypasses initialization of:
08:52:04 [Warning] ipv6.c@1246,14: [Pe546]: transfer of control bypasses initialization of:
08:52:04 [Warning] ipv6.c@1255,0: [Pe546]: transfer of control bypasses initialization of:
```
[ ] Question  
[ ] Enhancement  
[X] Bug

SeppoTakalo, 2018-06-19:
This is from Nanostack, where following kind of syntax is common:
```
static buffer_t *icmpv6_redirect_handler(buffer_t *buf, protocol_interface_info_entry_t *cur)
{
    ...
    if (addr_is_ipv6_multicast(dest)) {
        goto drop;
    }
    const uint8_t *tllao = icmpv6_find_option_in_buffer(buf, 36, ICMPV6_OPT_TGT_LL_ADDR, 0);
    if (tllao) {
        ...
    }
    ...
    ipv6_destination_redirect(tgt, buf->src_sa.address, dest, buf->interface->id, tgt_ll.addr_type, tgt_ll.address);
    return buffer_free(buf);

drop:
    tr_warn("Redirect drop");
    return buffer_free(buf);
}
```

So code actually does not do anything wrong, but compilers tend to warn about skipping initializer. This time its the `*llao` variable.

This is for Mesh team to fix.

kjbracey-arm, 2018-06-19:
Or for the tools team to disable the pointless warning.

kjbracey-arm, 2018-06-19:
I guess it's `--diag_suppress=Pe546` applied to C.

SeppoTakalo, 2018-06-19:
If we suppress the warning, then we need to rely on proper tool to check that uninitialised variables are not referenced. For example Coverity.

SeppoTakalo, 2018-06-19:
I'm fine to suppress this warning, as it matched what our other tools do.

kjbracey-arm, 2018-06-19:
> If we suppress the warning, then we need to rely on proper tool to check that uninitialised variables are not referenced. For example Coverity.

We have to rely on that anyway - the warning only triggers for this:

       goto label;
       int foo = 2;
    label:

It doesn't trigger for

       int foo;
       goto label;
       foo = 2;
    label:

The warning only makes sense as a "not C++ compatible" warning. C++ enforces the rule to stop you side-stepping constructors.

PR #7261 (2018-06-20): IAR: Suppress C "bypasses initialization" warning

### Description

By default IAR generates "transfer of control bypasses initialization"
warnings for C code - it's a legal construct that frequently occurs when
doing Linux-style "goto error". Many occurrences in Nanostack.

Suppress the warning for C only, to align with GCC and ARMCC. Have to
take care not to put it in the "common" section, as this would suppress
it for C++, where it actually is illegal.

Resolves #7253 and #7254.

### Pull request type

    [ ] Fix
    [X] Refactor
    [ ] New target
    [ ] Feature
    [ ] Breaking change
closed 2018-06-20 by cmonr