ARMmbed/mbed-os #7254: RPL control and pana client warnings

teetak01, 2018-06-19:
### Description
Mbed OS 5.9.1
IAR ANSI C/C++ Compiler V7.80.1.28/LNX for ARM

```
08:52:12 [Warning] rpl_control.c@822,10: [Pe546]: transfer of control bypasses initialization of:
08:52:12 [Warning] rpl_control.c@828,14: [Pe546]: transfer of control bypasses initialization of:
08:52:12 [Warning] rpl_control.c@837,14: [Pe546]: transfer of control bypasses initialization of:
08:52:12 [Warning] rpl_control.c@841,10: [Pe546]: transfer of control bypasses initialization of:
08:52:12 [Warning] rpl_control.c@854,10: [Pe546]: transfer of control bypasses initialization of:
08:52:12 [Warning] rpl_control.c@861,14: [Pe546]: transfer of control bypasses initialization of:
08:52:12 [Warning] rpl_control.c@871,10: [Pe546]: transfer of control bypasses initialization of:
08:52:12 [Warning] rpl_control.c@880,10: [Pe546]: transfer of control bypasses initialization of:
08:52:12 [Warning] rpl_control.c@885,14: [Pe546]: transfer of control bypasses initialization of:
08:52:12 [Warning] rpl_control.c@893,0: [Pe546]: transfer of control bypasses initialization of:
08:52:12 [Warning] pana_client.c@180,10: [Pe546]: transfer of control bypasses initialization of:
08:52:12 [Warning] pana_client.c@284,10: [Pe546]: transfer of control bypasses initialization of:
08:52:12 [Warning] pana_client.c@288,0: [Pe546]: transfer of control bypasses initialization of:
08:52:13 [Warning] rpl_upward.c@1075,14: [Pe546]: transfer of control bypasses initialization of:
08:52:13 [Warning] rpl_upward.c@1104,14: [Pe546]: transfer of control bypasses initialization of:
08:52:13 [Warning] rpl_upward.c@1116,0: [Pe546]: transfer of control bypasses initialization of:
```
[ ] Question  
[ ] Enhancement  
[X] Bug

kjbracey-arm, 2018-06-19:
Like #7253, best fixed by suppressing in IAR config.

adbridge, 2018-06-20:
@kjbracey-arm can this be closed now then ?

teetak01, 2018-06-20:
Verified. Thanks @kjbracey-arm

closed 2018-06-20 by teetak01